Koenkk/zigbee-OTA #6: Please add Osram FW

guptamp, 2020-06-06:
Hello,

the firmware for Osram devices can be found here:
https://update.ledvance.com/firmware-overview?submit=all
I'm afraid to upload them myself for copyright issues.
Can you just point the config to this website?

Specifically i'm after the fw for 
OUTDOOR LANTERN W RGBW OSRAM | 14.03.2019 | 0x110c | 0x0068 | V0.20 Build 510

Is there an easy way to point zigbee2mqtt to an alternative host to look for updates?
I'm using the docker version.

Many thanks for your help!

Koenkk, 2020-06-07:
I think this is already covered with https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/ota/ledvance.js

guptamp, 2020-06-07:
Do I need to do anything to enable this?
Update check always says instantly there is no update available.

Koenkk, 2020-06-07:
Could you enable the debug logging while checking for an update and post the log here?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

guptamp, 2020-06-08:
Hm,

I saw now that it successfully queried the website (I also just noticed that i got updated to z2M 1.13.1 this morning).
What I noticed that could cause this is that the original FW on the light (from 2015) is reported as "V1.05.10" but the new one from the update server (from 2019) is "0.20.510"... So the versioning scheme doesn't seem to match at all?! Not sure how to resolve this.
Could there be a manual override of some sort like "forceupdate"?

```
zigbee2mqtt:info  2020-06-08 10:04:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"No update available for 'Garden_Light1'","meta":{"status":"not_available","device":"Garden_Light1"}}',
zigbee2mqtt:info  2020-06-08 10:04:24: MQTT publish: topic 'zigbee2mqtt/Garden_Light1', payload '{"update_available":false,"last_seen":1591603464311,"state":"OFF","brightness":0,"color_temp":14085,"color":{"x":0,"y":0}}',
zigbee2mqtt:info  2020-06-08 10:04:24: No update available for 'Garden_Light1',
zigbee2mqtt:debug 2020-06-08 10:04:24: Update available for '0x7cb03eaa00ae4ac6': NO,
zigbee2mqtt:debug 2020-06-08 10:04:24: Is new image available for '0x7cb03eaa00ae4ac6', current '{"fieldControl":0,"manufacturerCode":4364,"imageType":104,"fileVersion":16909584}', latest meta '{"fileVersion":16909584,"fileSize":142968,"url":"https://api.update.ledvance.com/v1/zigbee/firmwares/download?company=4364&product=104&version=0.20.510"}',
zigbee2mqtt:debug 2020-06-08 10:04:24: Got OTA request '{"fieldControl":0,"manufacturerCode":4364,"imageType":104,"fileVersion":16909584}',
zigbee2mqtt:debug 2020-06-08 10:04:24: Received Zigbee message from 'Garden_Light1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4364,"imageType":104,"fileVersion":16909584}' from endpoint 3 with groupID 0,
zigbee2mqtt:debug 2020-06-08 10:04:23: Using endpoint '3',
zigbee2mqtt:info  2020-06-08 10:04:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Checking if update available for 'Garden_Light1'","meta":{"status":"checking_if_available","device":"Garden_Light1"}}',
zigbee2mqtt:debug 2020-06-08 10:04:23: Check if update available for '0x7cb03eaa00ae4ac6' (Outdoor Lantern W RGBW OSRAM),
zigbee2mqtt:info  2020-06-08 10:04:23: Checking if update available for 'Garden_Light1',
zigbee2mqtt:debug 2020-06-08 10:04:23: Received MQTT message on 'zigbee2mqtt/bridge/ota_update/check' with data 'Garden_Light1',
zigbee2mqtt:info  2020-06-08 10:04:03: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.13.1","commit":"2cb234c","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}',
zigbee2mqtt:info  2020-06-08 10:04:03: Switching log level to 'debug'
```

Koenkk, 2020-06-08:
According to this log your bulb is already up-to-date. Note that the versioning is done based on the `fileVersion` (not version numbers). The current one of your bulb is `16909584` which is the same as on https://api.update.ledvance.com/v1/zigbee/firmwares/download?company=4364&product=104&version=0.20.510

guptamp, 2020-06-08:
Ok, thank you.
Maybe this would be useful to document so others like will not wonder why OTA is not working :)
 Guess I have to send back my lights now :( (was hoping the new fw will fix the switch off issue)

Thanks for your help!

closed 2020-06-08 by guptamp