Koenkk/zigbee-OTA #16: Where can i find the link between 'supported devices' and their firmware?

gvdhoven, 2020-12-22:
I have the following devices:
* https://www.zigbee2mqtt.io/devices/8718699688882.html
* https://www.zigbee2mqtt.io/devices/LWG004.html
* https://www.zigbee2mqtt.io/devices/9290019758.html
* https://www.zigbee2mqtt.io/devices/929002240401.html
* https://www.zigbee2mqtt.io/devices/9290012607.html
* https://www.zigbee2mqtt.io/devices/324131092621.html
* and the Hue Tap dimmer switch.

Every device is listed as 'ota updatable', however [the OTA JSON ](https://github.com/Koenkk/zigbee-OTA/blob/master/index.json) does not give any hints to which file is related to which device code. Reason for asking is that i have a Deconz setup (and therefore can't use the OTA update functionality from Zigbee2MQTT can i?)

gvdhoven, 2020-12-22:
An alternative would be that i stop my Deconz docker container and start a zigbee2mqtt docker container with my Deconz II USB stick, since i see it is supported. However, i don't know if that will screw up my zigbee network? Would that work @Koenkk ?

Koenkk, 2020-12-22:
There is no 1-1 mapping between a device and firmware. In practice devices share the same firmware or in theory a device could use multiple firmware versions (e.g. different revisions of a product). The device will query the gateway (Deconz/Z2M/ZHA) with a `manufacturerCode` and `imageType`. With this combination you can determine what firmware is needed for this device. 

> However, i don't know if that will screw up my zigbee network? 

I don't know but, but since you need to re-pair the devices in order for Zigbee2MQTT to gather required information (populate the data/database.db file) I can imagine you will get into strange situations.

I will close this since it's not an issue but a questions.

closed 2020-12-22 by Koenkk