Koenkk/zigbee-OTA #17: Trying OTA on TERNCY-SD01 , got error message.

edison72, 2021-01-06:
I just bought TERNCY-SD01 last week , i seen new OTA update function for TERNCY-SD01 , but already try many times , got same error message , also changing new battery , i use web interface in home assistant.

INFO: Updating '0x000d6f001673d0c2' to latest firmware
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating '0x000d6f001673d0c2' to latest firmware","meta":{"device":"0x000d6f001673d0c2","status":"update_in_progress"},"type":"ota_update"}'
INFO: MQTT publish: topic 'zigbee2mqtt/0x000d6f001673d0c2', payload '{"battery":84,"direction":"clockwise","number":8,"update":{"state":"available"},"update_available":true}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of '0x000d6f001673d0c2' failed (Device didn't respond to OTA request)","meta":{"device":"0x000d6f001673d0c2","status":"update_failed"},"type":"ota_update"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x000d6f001673d0c2"},"error":"Update of '0x000d6f001673d0c2' failed (Device didn't respond to OTA request)","status":"error","transaction":"xo6yc-2"}'
ERROR: Update of '0x000d6f001673d0c2' failed (Device didn't respond to OTA request)

[image]

Koenkk, 2021-01-06:
You probably need to wakeup the device right before executing the request (by e.g. rotating it)

edison72, 2021-01-06:
I have try push it or rotate it before update firmware , but got same error message.

Koenkk, 2021-01-07:
@tube0013 since you added the OTA file, how did you update it?

tube0013, 2021-01-07:
I use ZHA, I sniffed the fw ota from the Terncy hub doing the update. I had 2 other ZHA users with the device confirm successful firmware updates with it before sharing here. I Wanted to contribute it here so others could use it.

ilarrain, 2021-01-07:
I was one of the contacted users, and can confirm that the update was applied correctly though ZHA (a bit slow, but I think that's inherent to the Zigbee protocol).

edison72, 2021-01-07:
I don't have Terncy hub , my hardware and software list :
1.Home Assistant OS 5.9
2.Home Assistant Core 2020.12.7 
3.Mosquitto broker 5.1
4..Zigbee2mqtt 1.17.0
5.cc2538+cc2592+esp32 Coordinator link by wifi

[image]

closed 2021-02-16 by github-actions[bot]