Koenkk/zigbee-OTA #22: Eurotronic Technology Spirit Zigbee OTA provider source

Hedda, 2021-02-01:
FYI, Eurotronic recently started publishing official OTA firmware image releases on GitHub:

https://github.com/EUROTRONIC-Technology/Spirit-ZigBee/releases

They have also posted a guide on how to OTA update using deCONZ software from Dresden Elektronik:

https://github.com/EUROTRONIC-Technology/Spirit-ZigBee/blob/main/README.md

PS: Apparently upgrading is recommended because issues like these https://github.com/search?q=%22spirit+zigbee%22&state=open&type=Issues

Koenkk, 2021-02-01:
This firmware has already been added (https://github.com/Koenkk/zigbee-OTA/commit/53d2f6af9c93b384b1f4d98d6f6312f7312858ea)

closed 2021-02-01 by Koenkk

BeistleA, 2021-03-08:
Hey Koenkk,

sorry to bother you in this case. I got TVR´s with 20191014 FW and get the message that the FW on the device is newer than the one available.  Is there any possibility to downgrade them cause the new FW ist the faulty one.

Thanks in avantage
Andy

h-tro, 2021-03-15:
Hi Koenkk,

I am using a **new** Eurotronic SPZB0001 with firmware 20191014 and  Zigbee2mqtt Current version: 1.18.1-2

[image]

**and I get the following error:**

_Zigbee2MQTT:error 2021-03-15 15:09:54: Failed to check if update available for 'EurotronicSP-01' (No image available for imageType '65535')_

Can you explain that ?

Thanks
HTro

hargathor, 2021-03-22:
Same issue for me, "the firmware is newer than the latest firmware online". The faulty one beeing the new firmware is it possible to force a downgrade of the device ?

IronButterfly, 2021-12-29:
Same story. Is it possible to force firmware downgrade for "SPZB0001"?