Koenkk/zigbee-OTA #35: Eurotronic Technology Spirit Zigbee OTA Error

h-tro, 2021-03-15:
Hi Koenkk,

I am using a **new** Eurotronic SPZB0001 with firmware 20191014 and Zigbee2mqtt Current version: 1.18.1-2

[image]

**and I get the following error:**

_Zigbee2MQTT:error 2021-03-15 15:09:54: Failed to check if update available for 'EurotronicSP-01' (No image available for imageType '65535')_

Can you explain that ?

Thanks
HTro

Koenkk, 2021-03-16:
It means we have no image for this specific device available, we only have `4364` (https://github.com/Koenkk/zigbee-OTA/blob/0014500fc341c056572ab0694ce63883e082d23c/index.json#L788)

h-tro, 2021-03-17:
Boost, OFF and Window Open function do not work with firmware firmware 20191014 and a firmware update is not possible.
Do not buy Eurotronic Technology Spirit Zigbee model SPZB0001

sorryusernameisalreadytaken, 2021-04-02:
> Boost, OFF and Window Open function do not work with firmware firmware 20191014 and a firmware update is not possible.
> Do not buy Eurotronic Technology Spirit Zigbee model SPZB0001

You should write them a email to info@eurotronic.org
Ask for a json url with all ota files and explain a little bit why they should support 3rd party zigbee ota updates (security and fullfill zigbee standards).

Hedda, 2021-05-03:
It looks like @EUROTRONIC-Technology might be active on GitHub so could try pinging @Witriol and @magicdude4eva

https://github.com/EUROTRONIC-Technology/

https://github.com/EUROTRONIC-Technology/Spirit-ZigBee

closed 2021-06-11 by github-actions[bot]