Koenkk/zigbee-OTA #45: Is there a way to add OTA firmware updates for own custom device?

grafalex82, 2021-06-10:
Hi Koen & All,

I am developing my own device, and I would like to build an OTA Update feature for it. From the description I understand how to add the firmware to index.json. But the documentation does not describe how get this firmware to a working z2m instance. 

Could you please shed some light on it?

Koenkk, 2021-06-10:
This is not really supported, but can be achieved by hosting the root as this git repo as http server (use e.g. the builtin python server). Add `ota: ota.zigbeeOTA,` to your definition and change the url of https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/ota/zigbeeOTA.js#L1 to your own one.

grafalex82, 2021-06-14:
Thank you very much for the reply. 

Would it be possible to 
1) expose base images URL to Z2M configuration file, so that there is no need to patch z2m sources
1.5) perhaps this option should come in addition to the base URL, so that users can add their own firmwares in addition to those that come with Z2M OTA repository
2) Allow firmware files located on local drive in addition to files downloaded from web, so that it is possible to provide custom device firmware files without having to host a http server for them.

What do you think?

Koenkk, 2021-06-14:
1. Yes, could you create a PR for this? You can use the `ikea_ota_use_test_url` as an example
2. Possible, but would require a lot of work.

grafalex82, 2021-06-16:
Unfortunately I have absolutely no experience in JS :( This will take some (a lot) time for me to get through this. I would really appreciate if you could boost this somehow.

closed 2021-07-24 by github-actions[bot]

qvntra, 2021-10-31:
I'm about to do this. We're adding OTA client to a Zigbee Router device and will attempt to update through the route that @Koenkk mentioned. I will give a write-up on how to implement OTA end-to-end so that we have it for the future.

grafalex82, 2021-10-31:
I am also going to dig in the same direction.