Koenkk/zigbee-OTA #55: Unable to add Osram/Ledvance firmware

redahb, 2021-10-31:
Zigbee2MQTT tells me some of my OSRAM smart plugs have no update (while on v1.05.10) while ones I updated with the ightify bridge a couple of years ago are on version 1.05.11. Zigbee2MQTT also tells me the .11 is newer than available online.

There are a bunch of firmware files located at http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/, including the 1.05.11 firmware for the smart plug (http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/DEV/ZLL_Plug01_OnOff_MK_0x01020511_encrypt.bin). So I tried to add the files to a repo, but keep getting this error message:

`(node:18412) UnhandledPromiseRejectionWarning: AssertionError [ERR_ASSERTION]: Not an OTA file
    at Object.parseImage (/root/zigbee-OTA/lib/ota.js:42:5)
    at main (/root/zigbee-OTA/scripts/add.js:63:24)
    at <anonymous>
`

Is this because they are encrypted? I noticed the files in this repo also have 'encrypted' in the filename, so I'm assuming they can be added.

Koenkk, 2021-11-01:
@DJFliX you previously added OSRAM firmware files in https://github.com/Koenkk/zigbee-OTA/pull/11, could you help with this?

DJFliX, 2021-11-03:
So I had a quick look. All files that I sourced for #11 came from https://update.ledvance.com/firmware-overview and I added the files to the repo that didn't throw weird errors. The files that are in that S3 bucket contain all kinds of different types of firmware, both deployment and production firmware which is awesome! Chances are that all files that I added are also in this s3 bucket. 

Unfortunately I'm not using zigbee2mqtt any more (I converted to SmartThings for all my non-Hue zigbee stuff) and I sold the USB Zigbee module I was running it with. So I don't have a way to test the newer firmwares for Osram/Lightify hardware that I still have. 

My best guess would be
* Compare md5 checksum of firmware in repo and firmware files with the same name in the S3 bucket (so you know files in this folder are potentially usable)
* For matching files, check if there are newer versions of firmware for that target device in the same directory (because they're in the same folder)
* Place the newer files next to the older files in the repo and cross your fingers

Sorry, this is the best I can do without having hardware to rest 🙃

redahb, 2021-11-03:
I think the files are really encrypted, but I have no idea how to un-encrypt them. Example:

When on https://update.ledvance.com/firmware-overview?sortOrder=created&submit=all, download the OTA for A19 RGBW (sorry, I can't get a direct link) and it will give you a file named 'A19_RGBW_IMG0019_00102428-encrypted.ota'. Viewed with a hex editor, the file is mostly garbage but the first part has 'EBL RGBW' visible.

Then, on the other server, there's this file: http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/firmware/release/Cloud/NAFTA/LDVreleaseFW/A19_RGBW_IMG0019_00102428-encrypted_encrypt.bin. Same filesize, same device (A19 RGBW), same firmware version. But the file is all garbage, there are no similarities.

This is the case for various other firmware files as well, and there are certain newer versions and I have no clue why Osram/Ledvance didn't put them on the 'ota' server. Especially because their own Lightify gateway does supply the newer firmware files from this other server.

DJFliX, 2021-11-03:
If they files are really encrypted you are probably out of luck. You can
try binwalk on both files to see if there is some padding or known
packaging algorithm used but without a rooted bridge (assuming the bridge
actually does the decoding) you probably won't get very far.

It is very possible that at some point Osram/Ledvance decided that they
need to "protect their IP" and that they wouldn't provide firmware publicly
without additional encryption measures. Unfortunately I'm not very
knowledgeable on the Osram/Lightify ecosystem.

closed 2021-12-12 by github-actions[bot]