Koenkk/zigbee-OTA #60: Gledopto GL-C-006P

Sineos, 2021-11-15:
I have received a new firmware for Gledopto GL-C-006P. Would be grateful if anyone can check if it is valid and add.
[GL-C-006P_V10276801OTAVA.zip](https://github.com/Koenkk/zigbee-OTA/files/7536414/GL-C-006P_V10276801OTAVA.zip)

Sineos, 2021-11-15:
This is strange. Following the Readme, I tried to add it to my fork: https://github.com/Sineos/zigbee-OTA/commit/04a0cf23486a70a5f9bfc419013a17e7c13d42b3#

It replaces 

```
images/Gledopto/GL-D-005P_V11076801_OTAV12_20211108_60%.ota

```

with

```
images/Gledopto/GL-C-006P_V10276801OTAVA.ota

```

I received the OTA image from Gledopto officials via mail.

Koenkk, 2021-11-16:
could you check with gledopto if GL-C-006P and GL-D-005P use the same firmware?

Sineos, 2021-11-17:
I received following feedback from Gledopto:

> Just upgrade GL-C-006P, and ignore GL-D-005P. This OTA file only work for GL-C-006P.
> If you encounter difficulties during the upgrade process, please contact us.

Sineos, 2021-11-17:
It seems that these are indeed different firmwares and the replacement should not take place:

GL-D-005P_V11076801_OTAV12_20211108_60%.ota
[image]

GL-C-006P_V10276801OTAVA.ota
[image]

Before asking Gledopto for a new firmware file, I bricked one of my GL-C-006P with the attempt to upgrade the firmware. Maybe it was trying to install the GL-D-005P firmware? Or are there any safe-guards against installing the wrong FW?

Koenkk, 2021-11-17:
> Or are there any safe-guards against installing the wrong FW?

I zigbee safe-guards are implemented on the device side (looks like gledopto didn't do this correctly)

closed 2021-11-17 by Koenkk

Sineos, 2021-11-19:
Hmpf, I bricked my second GL-C-006P with the attempt to update it.
Is there a way to manually flash via serial interface or something?

Koenkk, 2021-11-19:
Does this mean this fw is not correct and should be reverted?

Sineos, 2021-11-19:
> Does this mean this fw is not correct and should be reverted?

To be honest, I have no clue. This is all I got log-wise:
`npm[72827]: #033[31mZigbee2MQTT:error#033[39m 2021-11-18 08:05:39: Update of 'LEDKitchenEG' failed (Timeout: device did not request any image blocks)`

After that the device was dead. Unfortunately I do not know if the correct OTA file was requested, respectively delivered, etc.
Gledopto explicitly advised to use only Conbee for OTA updates, on the other hand there are reports of successful Gledopto FW updates 🤷

Koenkk, 2021-11-19:
Reverted it just to be sure.

Sineos, 2021-11-19:
Do you know if there is a way to flash manually, e.g. via a serial interface? I would love to test more but each failed attempt is quite costly.

Koenkk, 2021-11-19:
I don't know that, you could try contacting gledopto on how to manually flash these devices.

Sineos, 2021-11-20:
I got a new GL-C-006P and it has exactly the same firmware string as we have been trying to add:

[image]

Judging from this the firmware should be valid

The board itself has debug / testing / serial pins exposed:

[image]

The zigbee chip is a Silicon Labs MG21

Going to ask if Gledopto is willing to help to flash the module manually. Maybe we can find out what went wrong

TheJulianJES, 2021-11-23:
I'm not sure if this is related at all, but at least for the ``GL-C-007P`` there are multiple hardware versions (requiring different images). I think rev. 2 send ``hw_version=2`` (or similar) when requesting an OTA.
It's possible the same thing is the issue with newer/older GL-C-006P controllers?

Sineos, 2021-11-23:
I do not know. The controller PCB revision is given as: PCB-05F8-PB v1.0

Unfortunately Gledopto seems not willing to support and provide an instruction how to potentially flash such a controller manually. 

This simply means two things for me:
- I'm not risking another controller as I already killed two of them
- I'm going to stay clear of Gledopto in the future

andrewostroumov, 2021-12-08:
@Sineos you could try to follow this

Sineos, 2021-12-08:
@andrewostroumov Thanks. I do have a ST Link v2 JTAG and maybe I could simply try to flash the the OTA file. Although it kind of seems that the JTAG pins are not on the header shown above.

@TheJulianJES 
> I'm not sure if this is related at all, but at least for the `GL-C-007P`, there are multiple hardware versions (requiring different images). I think rev. 2 sends `hw_version=2` (or similar) when requesting an OTA. It's possible the same thing is the issue with newer/older GL-C-006P controllers? (I updated multiple GLEDOPTO controllers with custom ZHA/zigpy versions.)

Well, just out of interest I opened one of my new GL-C-006P controller

[image]

I'm not an expert but this somehow looks different ...

andrewostroumov, 2021-12-08:
The first link in google give me this

https://developer.tuya.com/en/docs/iot/zs3l?id=K97r37j19f496

Seems like under the cover this is the same EFR32MG21 and could be flashed the same way

In any way please report any luck with flashing ota file

andrewostroumov, 2021-12-08:
@Sineos try this scheme to connect j-link v9

[image]

andrewostroumov, 2021-12-08:
Is anybody know difference compiled firmware and OTA file? And maybe any converters? I wanna flash my 006 controllers with newest firmware with hope of startup behaviour could be configures

PR #64 (2021-11-17): Add GL-C-006P firmware
Version: 10276801
Date: 2021-07-19

Fixes: #60