Koenkk/zigbee-OTA #67: New Hue Firmwares

Bascht74, 2021-11-29:
fds.dc1.philips.com/firmware/ZGB_100B_0103/1124096769/100B_0103_LivingColors-Hue-Target_0012_91.1.sbl-ota --> Must be for Bloom LLC011

fds.dc1.philips.com/firmware/ZGB_100B_0119/33565954/100B-0119-02002D02-Switch-EFR32MG22.zigbee --> Must be for New Dimmer RWL022

Koenkk, 2021-11-30:
Please make a pull request, instructions on how to add can be found here: https://github.com/Koenkk/zigbee-OTA/blob/master/README.md#adding-new-and-updating-existing-ota-files

Bascht74, 2021-11-30:
@Koenkk 
I will try, but not sure if I am capable off... (not a programmer)
If not, is there somebody I can ask for help with that ?
(I updated the devices via Hue, but there might be other users who could use OTA in Z2M then)

Koenkk, 2021-11-30:
added them, thanks!

closed 2021-11-30 by Koenkk