Koenkk/zigbee-OTA #103: Flash Hive TRV with Danfoss Ally firmware 1.18

robertalexa, 2022-04-02:
Hi @Koenkk 

I have hit a brick wall and I would appreciate your guidance if you have the time.

As you might have heard before, Hive TRV is just a Danfoss Ally hardware but with a Hive firmware on it (I think exactly the same for Popp TRV but i do not have experience with it). 

And i hate it :D I hate Hive... but that is offtopic. My aim is to flash the device with the official Danfoss firmware, which should hopefully leave me with native danfoss devices with just a different logo on the plastic.

Here are the steps I have done, in the order I did them:
1. as the device extends danfoss, it extends the ota functionality. But of course there is no firmware for it in z-ota repo. This gives me `Failed to check if update available for 'Rob's Office TRV' (No image available for imageType '66')`. So i understand that imageType 66 is what i need
2. Forked z-ota, copied the danfoss entry but used `66` as imageType. Also added `force: true` because Hive's version is higher than Danfoss
```
    {
        "fileVersion": 274,
        "fileSize": 379806,
        "manufacturerCode": 4678,
        "imageType": 66,
        "sha512": "873effaa113e1d07643fbf25f6e26ee4b23f8deb7d006cce7fc490fd8ad7a5754de4c352b6f3d5e172022a34ea2699f5d1d8fda90f4348b25d5feaa169e38c85",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota",
        "path": "images/Danfoss/1246-0100-01180118.0002_(F5BFD8B8).ota",
        "force": true
    },
```
3. Commented out the following assertion: This of course is because the device and index are 66 but the file itself is Danfoss' and is 256
```
assert(image.header.imageType === current.imageType, 'Image type mismatch');
```
4. Run OTA and it fails:
```
Zigbee2MQTT:debug 2022-04-02 21:22:51: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"0x842e14fffe203c4c","transaction":"z5442-1"}'
Zigbee2MQTT:info  2022-04-02 21:22:51: Updating '0x842e14fffe203c4c' to latest firmware
Zigbee2MQTT:debug 2022-04-02 21:22:57: Received Zigbee message from '0x842e14fffe203c4c', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200402","swBuildId":"02.33.0008 02.33"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-04-02 21:22:57: Updating to latest '0x842e14fffe203c4c' (TRV001)
Zigbee2MQTT:debug 2022-04-02 21:22:57: Using endpoint '1'
Zigbee2MQTT:debug 2022-04-02 21:23:01: Received Zigbee message from '0x842e14fffe203c4c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":545,"imageType":66,"manufacturerCode":4678}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-04-02 21:23:01: Device '0x842e14fffe203c4c' requested OTA
Zigbee2MQTT:debug 2022-04-02 21:23:01: Got OTA request '{"fieldControl":0,"manufacturerCode":4678,"imageType":66,"fileVersion":545}'
Zigbee2MQTT:debug 2022-04-02 21:23:01: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2022-04-02 21:23:01: getNewImage for '0x842e14fffe203c4c', meta {"fileVersion":274,"fileSize":379806,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota","sha512":"873effaa113e1d07643fbf25f6e26ee4b23f8deb7d006cce7fc490fd8ad7a5754de4c352b6f3d5e172022a34ea2699f5d1d8fda90f4348b25d5feaa169e38c85","force":true}
Zigbee2MQTT:debug 2022-04-02 21:23:01: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota
Zigbee2MQTT:debug 2022-04-02 21:23:01: OTA update checksum validation succeeded for '0x842e14fffe203c4c'
Zigbee2MQTT:debug 2022-04-02 21:23:01: getNewImage for '0x842e14fffe203c4c', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4678,"imageType":256,"fileVersion":274,"zigbeeStackVersion":2,"otaHeaderString":"29-11-2021 V.0112.0112\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":379806}
Zigbee2MQTT:debug 2022-04-02 21:23:01: Got new image for '0x842e14fffe203c4c'
Zigbee2MQTT:debug 2022-04-02 21:23:01: Starting upgrade
Zigbee2MQTT:debug 2022-04-02 21:23:02: Received Zigbee message from '0x842e14fffe203c4c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":545,"imageType":66,"manufacturerCode":4678}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-04-02 21:23:02: Device '0x842e14fffe203c4c' requested OTA
Zigbee2MQTT:debug 2022-04-02 21:25:29: Received Zigbee message from '0x842e14fffe203c4c', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2172}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-04-02 21:25:29: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":39,"external_measured_room_sensor":-8000,"keypad_lockout":"unlock","linkquality":115,"load_estimate":5,"local_temperature":21.72,"mounted_mode_active":false,"system_mode":"heat","update":{"state":"available"}}'
Zigbee2MQTT:debug 2022-04-02 21:25:31: Update of '0x842e14fffe203c4c' failed (Error: Timeout: device did not request any image blocks)
Zigbee2MQTT:info  2022-04-02 21:25:31: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":39,"external_measured_room_sensor":-8000,"keypad_lockout":"unlock","linkquality":115,"load_estimate":5,"local_temperature":21.72,"mounted_mode_active":false,"system_mode":"heat","update":{"state":"available"}}'
Zigbee2MQTT:info  2022-04-02 21:25:31: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x842e14fffe203c4c"},"error":"Update of '0x842e14fffe203c4c' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"z5442-1"}'
Zigbee2MQTT:error 2022-04-02 21:25:31: Update of '0x842e14fffe203c4c' failed (Timeout: device did not request any image blocks)
Zigbee2MQTT:debug 2022-04-02 21:25:31: Error: Timeout: device did not request any image blocks
    at /home/rob/Projects/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:360:28
    at runNextTicks (node:internal/process/task_queues:59:5)
    at listOnTimeout (node:internal/timers:526:9)
    at processTimers (node:internal/timers:500:7)
```

Notes:
- device still functions fine after this failed OTA
- pressed the button during OTA, to make sure the device is awake. The device reports the check command but OTA still fails
- i do not intend to do a PR for this even if I am successful, instead I plan on making a post, maybe even put it on z2m.io under the devices notes. This is because it would be irreversible, most likely the device will never be able to work with Hive again, and of course this could brick the device (like i might with this exercise)

I would appreciate your thoughts on this Koen, i love these TRVs, but I hate Hive and the fact I can't get the most recent features, which would only be software related. And i would rather not need to a) live with it or b) sell them just to buy the Danfoss ones.... i would basically lose money on the exact same device just a different firmware.

Thanks in advance

Koenkk, 2022-04-03:
Could you provide the herdsman debug log when running z2m with `DEBUG=zigbee-herdsman:controller:device*`?

robertalexa, 2022-04-03:
Hi Koen,

See below from starting z2m to stopping it, with OTA attempt:

```
rob@rob-ThinkPad-T590:~/Projects/zigbee2mqtt$ npm start DEBUG=zigbee-herdsman:controller:device*
 
> zigbee2mqtt@1.25.0 start
> node index.js "DEBUG=zigbee-herdsman:controller:device*"

Zigbee2MQTT:debug 2022-04-03 10:12:36: Loaded state from file /home/rob/Projects/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-04-03 10:12:36: Logging to console and directory: '/home/rob/Projects/zigbee2mqtt/data/log/2022-04-03.10-12-36' filename: log.txt
Zigbee2MQTT:debug 2022-04-03 10:12:36: Removing old log directory '/home/rob/Projects/zigbee2mqtt/data/log/2022-01-21.13-18-41'
Zigbee2MQTT:info  2022-04-03 10:12:36: Starting Zigbee2MQTT version 1.25.0 (commit #6f1460e4)
Zigbee2MQTT:info  2022-04-03 10:12:36: Starting zigbee-herdsman (0.14.20)
Zigbee2MQTT:debug 2022-04-03 10:12:36: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/rob/Projects/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/home/rob/Projects/zigbee2mqtt/data/database.db.backup","databasePath":"/home/rob/Projects/zigbee2mqtt/data/database.db","network":{"channelList":[20],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2022-04-03 10:12:39: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-04-03 10:12:39: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201128,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:debug 2022-04-03 10:12:39: Zigbee network parameters: {"channel":20,"extendedPanID":"0x00124b0018e32e48","panID":6754}
Zigbee2MQTT:info  2022-04-03 10:12:39: Currently 1 devices are joined:
Zigbee2MQTT:info  2022-04-03 10:12:39: 0x842e14fffe203c4c (0x842e14fffe203c4c): UK7004240 - Hive Radiator valve based on Danfoss Ally (014G2461) (EndDevice)
Zigbee2MQTT:info  2022-04-03 10:12:39: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-04-03 10:12:39: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:debug 2022-04-03 10:12:39: Using MQTT anonymous login
Zigbee2MQTT:info  2022-04-03 10:12:39: Connected to MQTT server
Zigbee2MQTT:info  2022-04-03 10:12:39: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:debug 2022-04-03 10:12:39: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"6f1460e4","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":20,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_legacy_entity_attributes":false,"last_seen":"disable","legacy_api":false,"legacy_availability_payload":true,"log_directory":"/home/rob/Projects/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"output":"json","pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"blocklist":[],"device_options":{"legacy":false},"devices":{"0x842e14fffe203c4c":{"friendly_name":"0x842e14fffe203c4c"}},"external_converters":[],"frontend":{"host":"0.0.0.0","port":8080},"groups":{},"homeassistant":false,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://localhost"},"ota":{"disable_automatic_update_check":false,"update_check_interval":1440},"passlist":[],"permit_join":false,"serial":{"disable_led":false,"port":"/dev/ttyACM0"}},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore debounce","type":"array"},"filtered_attributes":{"description":"Filter attributes from publish payload.","examples":["temperature","battery","action"],"items":{"type":"string"},"title":"Filtered publish attributes","type":"array"},"filtered_optimistic":{"description":"Filter attributes from optimistic publish payload when calling /set. (This has no effect if optimistic is set to false).","examples":["color_mode","color_temp","color"],"items":{"type":"string"},"title":"Filtered optimistic attributes","type":"array"},"friendly_name":{"description":"Used in the MQTT topic of a device. By default this is the device ID","readOnly":true,"title":"Friendly name","type":"string"},"icon":{"description":"The user-defined device icon for the frontend. It can be a link to an image (not a path to a file) or base64 encoded data URL like: image/svg+xml;base64,PHN2ZyB3aW....R0aD","title":"Icon","type":"string"},"optimistic":{"default":true,"description":"Publish optimistic state after set","title":"Optimistic","type":"boolean"},"qos":{"description":"QoS level for MQTT messages of this device","title":"QoS","type":"number"},"retain":{"descript [... cut ...]
^CZigbee2MQTT:debug 2022-04-03 10:16:22: Saving state to file /home/rob/Projects/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-04-03 10:16:22: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2022-04-03 10:16:22: Disconnecting from MQTT server
Zigbee2MQTT:info  2022-04-03 10:16:22: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-04-03 10:16:22: Stopped zigbee-herdsman
Zigbee2MQTT:info  2022-04-03 10:16:22: Stopped Zigbee2MQTT

rob@rob-ThinkPad-T590:~/Projects/zigbee2mqtt$ 
```

Koenkk, 2022-04-03:
`npm start DEBUG=zigbee-herdsman:controller:device*` should be `DEBUG=zigbee-herdsman:controller:device* npm start`

robertalexa, 2022-04-03:
Doh...

Not a massive difference, there are a couple of lines from the debugged related to the check in command, but that is it

```javascript
    ~/Pr/zigbee2mqtt    master !1  DEBUG=zigbee-herdsman:controller:device* npm start

> zigbee2mqtt@1.25.0 start
> node index.js

Zigbee2MQTT:debug 2022-04-03 13:43:17: Loaded state from file /home/rob/Projects/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-04-03 13:43:17: Logging to console and directory: '/home/rob/Projects/zigbee2mqtt/data/log/2022-04-03.13-43-17' filename: log.txt
Zigbee2MQTT:debug 2022-04-03 13:43:17: Removing old log directory '/home/rob/Projects/zigbee2mqtt/data/log/2022-01-22.18-52-34'
Zigbee2MQTT:info  2022-04-03 13:43:17: Starting Zigbee2MQTT version 1.25.0 (commit #6f1460e4)
Zigbee2MQTT:info  2022-04-03 13:43:17: Starting zigbee-herdsman (0.14.20)
Zigbee2MQTT:debug 2022-04-03 13:43:17: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/rob/Projects/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/home/rob/Projects/zigbee2mqtt/data/database.db.backup","databasePath":"/home/rob/Projects/zigbee2mqtt/data/database.db","network":{"channelList":[20],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2022-04-03 13:43:20: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-04-03 13:43:20: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201128,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:debug 2022-04-03 13:43:20: Zigbee network parameters: {"channel":20,"extendedPanID":"0x00124b0018e32e48","panID":6754}
Zigbee2MQTT:info  2022-04-03 13:43:20: Currently 1 devices are joined:
Zigbee2MQTT:info  2022-04-03 13:43:20: 0x842e14fffe203c4c (0x842e14fffe203c4c): UK7004240 - Hive Radiator valve based on Danfoss Ally (014G2461) (EndDevice)
Zigbee2MQTT:info  2022-04-03 13:43:20: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-04-03 13:43:20: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:debug 2022-04-03 13:43:20: Using MQTT anonymous login
Zigbee2MQTT:info  2022-04-03 13:43:20: Connected to MQTT server
Zigbee2MQTT:info  2022-04-03 13:43:20: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:debug 2022-04-03 13:43:20: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"6f1460e4","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":20,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_legacy_entity_attributes":false,"last_seen":"disable","legacy_api":false,"legacy_availability_payload":true,"log_directory":"/home/rob/Projects/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"output":"json","pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"blocklist":[],"device_options":{"legacy":false},"devices":{"0x842e14fffe203c4c":{"friendly_name":"0x842e14fffe203c4c"}},"external_converters":[],"frontend":{"host":"0.0.0.0","port":8080},"groups":{},"homeassistant":false,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://localhost"},"ota":{"disable_automatic_update_check":false,"update_check_interval":1440},"passlist":[],"permit_join":false,"serial":{"disable_led":false,"port":"/dev/ttyACM0"}},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore debounce","type":"array"},"filtered_attributes":{"description":"Filter attributes from publish payload.","examples":["temperature","battery","action"],"items":{"type":"string"},"title":"Filtered publish attributes","type":"array"},"filtered_optimistic":{"description":"Filter attributes from optimistic publish payload when calling /set. (This has no effect if optimistic is set to false).","examples":["color_mode","color_temp","color"],"items":{"type":"string"},"title":"Filtered optimistic attributes","type":"array"},"friendly_name":{"description":"Used in the MQTT topic of a device. By default this is the device ID","readOnly":true,"title":"Friendly name","type":"string"},"icon":{"description":"The user-defined device icon for the frontend. It can be a link to an image (not a path to a file) or base64 encoded data URL like: image/svg+xml;base64,PHN2ZyB3aW....R0aD","title":"Icon","type":"string"},"optimistic":{"default":true,"description":"Publish optimistic state after set","title":"Optimistic","type":"boolean"},"qos":{"description":"QoS level for MQTT messages of this device","title":"QoS","type":"number"}," [... cut ...]
    at /home/rob/Projects/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:360:28
^CZigbee2MQTT:debug 2022-04-03 13:47:24: Saving state to file /home/rob/Projects/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-04-03 13:47:24: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2022-04-03 13:47:24: Disconnecting from MQTT server
Zigbee2MQTT:info  2022-04-03 13:47:24: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-04-03 13:47:25: Stopped zigbee-herdsman
Zigbee2MQTT:info  2022-04-03 13:47:25: Stopped Zigbee2MQTT
```

Koenkk, 2022-04-03:
Ah sorry, it should be `DEBUG=zigbee-herdsman:controller:endpoint* npm start` (I want to see what commands are send to the device)

robertalexa, 2022-04-03:
Done.

```bash
Zigbee2MQTT:debug 2022-04-03 22:06:39: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"0x842e14fffe203c4c","transaction":"baneb-1"}'
Zigbee2MQTT:info  2022-04-03 22:06:39: Updating '0x842e14fffe203c4c' to latest firmware
  zigbee-herdsman:controller:endpoint Read 0x842e14fffe203c4c/1 genBasic(["dateCode","swBuildId"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +5s
Zigbee2MQTT:debug 2022-04-03 22:06:43: Received Zigbee message from '0x842e14fffe203c4c', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200402","swBuildId":"02.33.0008 02.33"}' from endpoint 1 with groupID 0
  zigbee-herdsman:controller:endpoint DefaultResponse 0x842e14fffe203c4c/1 0(1, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +4s
Zigbee2MQTT:debug 2022-04-03 22:06:43: Updating to latest '0x842e14fffe203c4c' (TRV001)
Zigbee2MQTT:debug 2022-04-03 22:06:43: Using endpoint '1'
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +8ms
Zigbee2MQTT:debug 2022-04-03 22:06:47: Received Zigbee message from '0x842e14fffe203c4c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":545,"imageType":66,"manufacturerCode":4678}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-04-03 22:06:47: Device '0x842e14fffe203c4c' requested OTA
Zigbee2MQTT:debug 2022-04-03 22:06:47: Got OTA request '{"fieldControl":0,"manufacturerCode":4678,"imageType":66,"fileVersion":545}'
Zigbee2MQTT:debug 2022-04-03 22:06:47: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2022-04-03 22:06:47: getNewImage for '0x842e14fffe203c4c', meta {"fileVersion":274,"fileSize":379806,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota","sha512":"873effaa113e1d07643fbf25f6e26ee4b23f8deb7d006cce7fc490fd8ad7a5754de4c352b6f3d5e172022a34ea2699f5d1d8fda90f4348b25d5feaa169e38c85","force":true}
Zigbee2MQTT:debug 2022-04-03 22:06:47: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota
Zigbee2MQTT:debug 2022-04-03 22:06:48: OTA update checksum validation succeeded for '0x842e14fffe203c4c'
Zigbee2MQTT:debug 2022-04-03 22:06:48: getNewImage for '0x842e14fffe203c4c', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4678,"imageType":256,"fileVersion":274,"zigbeeStackVersion":2,"otaHeaderString":"29-11-2021 V.0112.0112\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":379806}
Zigbee2MQTT:debug 2022-04-03 22:06:48: Got new image for '0x842e14fffe203c4c'
Zigbee2MQTT:debug 2022-04-03 22:06:48: Starting upgrade
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +5s
Zigbee2MQTT:debug 2022-04-03 22:06:48: Received Zigbee message from '0x842e14fffe203c4c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":545,"imageType":66,"manufacturerCode":4678}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-04-03 22:06:48: Device '0x842e14fffe203c4c' requested OTA
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.queryNextImageResponse({"status":0,"manufacturerCode":4678,"imageType":256,"fileVersion":274,"imageSize":379806}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +444ms
Zigbee2MQTT:debug 2022-04-03 22:08:30: Received Zigbee message from '0x842e14fffe203c4c', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2211}' from endpoint 1 with groupID 0
  zigbee-herdsman:controller:endpoint DefaultResponse 0x842e14fffe203c4c/1 513(10, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +2m
Zigbee2MQTT:debug 2022-04-03 22:08:30: Received Zigbee message from '0x842e14fffe203c4c', type 'attributeReport', cluster 'hvacThermostat', data '{"danfossLoadEstimate":3}' from endpoint 1 with groupID 0
[... cut ...] nternal":"quarantine"}'
^CZigbee2MQTT:debug 2022-04-03 22:09:33: Saving state to file /home/rob/Projects/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-04-03 22:09:33: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2022-04-03 22:09:33: Disconnecting from MQTT server
Zigbee2MQTT:info  2022-04-03 22:09:33: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-04-03 22:09:33: Stopped zigbee-herdsman
Zigbee2MQTT:info  2022-04-03 22:09:33: Stopped Zigbee2MQTT
```

Koenkk, 2022-04-04:
I think you have to hack the image type to 66 here https://github.com/Koenkk/zigbee-herdsman-converters/blob/dda3ab3d584f46dc6d48a91aba0bedf36386d30a/lib/ota/common.js#L174

nahl-robert-alexa, 2022-04-04:
After your suggested change i get:
```bash
Zigbee2MQTT:debug 2022-04-04 22:13:35: Updating to latest '0x842e14fffe203c4c' (TRV001)
Zigbee2MQTT:debug 2022-04-04 22:13:35: Using endpoint '1'
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +4ms
Zigbee2MQTT:debug 2022-04-04 22:13:38: Received Zigbee message from '0x842e14fffe203c4c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":545,"imageType":66,"manufacturerCode":4678}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-04-04 22:13:38: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":39,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649106818942,"linkquality":178,"load_estimate":-5,"local_temperature":22.35,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19.5,"pi_heating_demand":1,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"state":"available"},"window_open_internal":"quarantine"}'
Zigbee2MQTT:debug 2022-04-04 22:13:38: Device '0x842e14fffe203c4c' requested OTA
Zigbee2MQTT:debug 2022-04-04 22:13:38: Got OTA request '{"fieldControl":0,"manufacturerCode":4678,"imageType":66,"fileVersion":545}'
Zigbee2MQTT:debug 2022-04-04 22:13:39: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2022-04-04 22:13:39: getNewImage for '0x842e14fffe203c4c', meta {"fileVersion":274,"fileSize":379806,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota","sha512":"873effaa113e1d07643fbf25f6e26ee4b23f8deb7d006cce7fc490fd8ad7a5754de4c352b6f3d5e172022a34ea2699f5d1d8fda90f4348b25d5feaa169e38c85","force":true}
Zigbee2MQTT:debug 2022-04-04 22:13:39: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota
Zigbee2MQTT:debug 2022-04-04 22:13:39: OTA update checksum validation succeeded for '0x842e14fffe203c4c'
Zigbee2MQTT:debug 2022-04-04 22:13:39: getNewImage for '0x842e14fffe203c4c', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4678,"imageType":256,"fileVersion":274,"zigbeeStackVersion":2,"otaHeaderString":"29-11-2021 V.0112.0112\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":379806}
Zigbee2MQTT:debug 2022-04-04 22:13:39: Got new image for '0x842e14fffe203c4c'
Zigbee2MQTT:debug 2022-04-04 22:13:39: Starting upgrade
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +5s
Zigbee2MQTT:debug 2022-04-04 22:13:40: Received Zigbee message from '0x842e14fffe203c4c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":545,"imageType":66,"manufacturerCode":4678}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-04-04 22:13:40: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":39,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649106820397,"linkquality":178,"load_estimate":-5,"local_temperature":22.35,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19.5,"pi_heating_demand":1,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"state":"available"},"window_open_internal":"quarantine"}'
Zigbee2MQTT:debug 2022-04-04 22:13:40: Device '0x842e14fffe203c4c' requested OTA
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.queryNextImageResponse({"status":0,"manufacturerCode":4678,"imageType":66,"fileVersion":274,"imageSize":379806}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +449ms
Zigbee2MQTT:info  2022-04-04 22:13:41: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":39,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649106821400,"linkquality":178,"load_estimate":-5,"local_temperature":22.35,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19.5,"pi_heating_demand":1,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"state":"available"},"window_open_internal":"quarantine"}'
Zigbee2MQTT:debug 2022-04-04 22:13:41: Got upgrade end request for '0x842e14fffe203c4c': {"status":150,"manufacturerCode":4678,"imageType":66,"fileVersion":545}
[... cut ...]
Zigbee2MQTT:info  2022-04-04 22:13:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x842e14fffe203c4c"},"error":"Update of '0x842e14fffe203c4c' failed (Update failed with reason: 'invalid image')","status":"error","transaction":"5ople-1"}'
Zigbee2MQTT:error 2022-04-04 22:13:41: Update of '0x842e14fffe203c4c' failed (Update failed with reason: 'invalid image')
Zigbee2MQTT:debug 2022-04-04 22:13:41: Error: Update failed with reason: 'invalid image'
    at /home/rob/Projects/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:411:24
```

robertalexa, 2022-04-04:
After your suggested change i get:
```bash
Zigbee2MQTT:debug 2022-04-04 22:13:35: Updating to latest '0x842e14fffe203c4c' (TRV001)
Zigbee2MQTT:debug 2022-04-04 22:13:35: Using endpoint '1'
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +4ms
Zigbee2MQTT:debug 2022-04-04 22:13:38: Received Zigbee message from '0x842e14fffe203c4c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":545,"imageType":66,"manufacturerCode":4678}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-04-04 22:13:38: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":39,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649106818942,"linkquality":178,"load_estimate":-5,"local_temperature":22.35,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19.5,"pi_heating_demand":1,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"state":"available"},"window_open_internal":"quarantine"}'
Zigbee2MQTT:debug 2022-04-04 22:13:38: Device '0x842e14fffe203c4c' requested OTA
Zigbee2MQTT:debug 2022-04-04 22:13:38: Got OTA request '{"fieldControl":0,"manufacturerCode":4678,"imageType":66,"fileVersion":545}'
Zigbee2MQTT:debug 2022-04-04 22:13:39: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2022-04-04 22:13:39: getNewImage for '0x842e14fffe203c4c', meta {"fileVersion":274,"fileSize":379806,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota","sha512":"873effaa113e1d07643fbf25f6e26ee4b23f8deb7d006cce7fc490fd8ad7a5754de4c352b6f3d5e172022a34ea2699f5d1d8fda90f4348b25d5feaa169e38c85","force":true}
Zigbee2MQTT:debug 2022-04-04 22:13:39: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danfoss/1246-0100-01180118.0002_%28F5BFD8B8%29.ota
Zigbee2MQTT:debug 2022-04-04 22:13:39: OTA update checksum validation succeeded for '0x842e14fffe203c4c'
Zigbee2MQTT:debug 2022-04-04 22:13:39: getNewImage for '0x842e14fffe203c4c', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4678,"imageType":256,"fileVersion":274,"zigbeeStackVersion":2,"otaHeaderString":"29-11-2021 V.0112.0112\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":379806}
Zigbee2MQTT:debug 2022-04-04 22:13:39: Got new image for '0x842e14fffe203c4c'
Zigbee2MQTT:debug 2022-04-04 22:13:39: Starting upgrade
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +5s
Zigbee2MQTT:debug 2022-04-04 22:13:40: Received Zigbee message from '0x842e14fffe203c4c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":545,"imageType":66,"manufacturerCode":4678}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-04-04 22:13:40: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":39,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649106820397,"linkquality":178,"load_estimate":-5,"local_temperature":22.35,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19.5,"pi_heating_demand":1,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"state":"available"},"window_open_internal":"quarantine"}'
Zigbee2MQTT:debug 2022-04-04 22:13:40: Device '0x842e14fffe203c4c' requested OTA
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.queryNextImageResponse({"status":0,"manufacturerCode":4678,"imageType":66,"fileVersion":274,"imageSize":379806}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +449ms
Zigbee2MQTT:info  2022-04-04 22:13:41: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":39,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649106821400,"linkquality":178,"load_estimate":-5,"local_temperature":22.35,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19.5,"pi_heating_demand":1,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"state":"available"},"window_open_internal":"quarantine"}'
Zigbee2MQTT:debug 2022-04-04 22:13:41: Got upgrade end request for '0x842e14fffe203c4c': {"status":150,"manufacturerCode":4678,"imageType":66,"fileVersion":545}
[... cut ...]
Zigbee2MQTT:info  2022-04-04 22:13:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x842e14fffe203c4c"},"error":"Update of '0x842e14fffe203c4c' failed (Update failed with reason: 'invalid image')","status":"error","transaction":"5ople-1"}'
Zigbee2MQTT:error 2022-04-04 22:13:41: Update of '0x842e14fffe203c4c' failed (Update failed with reason: 'invalid image')
Zigbee2MQTT:debug 2022-04-04 22:13:41: Error: Update failed with reason: 'invalid image'
    at /home/rob/Projects/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:411:24
```

PS: sorry for the previous reply, did it with my work account accidentaly

Koenkk, 2022-04-05:
Could you set the manufacturer code to 256 and fileversion to 546?

robertalexa, 2022-04-05:
Are you sure manuf code 256 is what you meant to say? Both Danfoss and Hive use the same manufacturer code in zigbee commands, so that would be the same already?

robertalexa, 2022-04-05:
So @Koenkk 

2:30h later, the update failed :) The device is still operational, even after unpairing, re-pairing, and is still recognised as a Hive TRV.

See below the log including the final part of the updating showing 99% progress
```
Zigbee2MQTT:info  2022-04-05 19:45:15: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":36,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649184315420,"linkquality":170,"load_estimate":-10,"local_temperature":22.22,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19,"pi_heating_demand":0,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"progress":99.67,"remaining":26,"state":"updating"},"window_open_internal":"closed"}'
  zigbee-herdsman:controller:endpoint CommandResponse 0x842e14fffe203c4c/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4678,"imageType":66,"fileVersion":546,"fileOffset":379800,"dataSize":6,"data":{"type":"Buffer","data":[129,15,148,51,91,98]}}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +1s
Zigbee2MQTT:info  2022-04-05 19:45:16: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":36,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649184316442,"linkquality":170,"load_estimate":-10,"local_temperature":22.22,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19,"pi_heating_demand":0,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"progress":99.67,"remaining":26,"state":"updating"},"window_open_internal":"closed"}'
Zigbee2MQTT:debug 2022-04-05 19:45:16: Got upgrade end request for '0x842e14fffe203c4c': {"status":150,"manufacturerCode":4678,"imageType":66,"fileVersion":546}
Zigbee2MQTT:debug 2022-04-05 19:45:16: Update failed with reason: 'invalid image'
Zigbee2MQTT:debug 2022-04-05 19:45:16: Update of '0x842e14fffe203c4c' failed (Error: Update failed with reason: 'invalid image')
Zigbee2MQTT:info  2022-04-05 19:45:16: MQTT publish: topic 'zigbee2mqtt/0x842e14fffe203c4c', payload '{"battery":36,"external_measured_room_sensor":-8000,"heat_available":true,"heat_required":false,"keypad_lockout":"unlock","last_seen":1649184316442,"linkquality":170,"load_estimate":-10,"local_temperature":22.22,"mounted_mode_active":false,"mounted_mode_control":false,"occupied_heating_setpoint":19,"pi_heating_demand":0,"running_state":"idle","system_mode":"heat","thermostat_vertical_orientation":true,"update":{"state":"available"},"window_open_internal":"closed"}'
Zigbee2MQTT:info  2022-04-05 19:45:16: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x842e14fffe203c4c"},"error":"Update of '0x842e14fffe203c4c' failed (Update failed with reason: 'invalid image')","status":"error","transaction":"n25ht-2"}'
Zigbee2MQTT:error 2022-04-05 19:45:16: Update of '0x842e14fffe203c4c' failed (Update failed with reason: 'invalid image')
Zigbee2MQTT:debug 2022-04-05 19:45:16: Error: Update failed with reason: 'invalid image'
    at /home/rob/Projects/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:413:24
```

Koenkk, 2022-04-05:
`Update failed with reason: 'invalid image'` looks to me that the device refused the image.

robertalexa, 2022-04-05:
I was getting the same error before setting fileversion to 546.

So i take it that we have reached the z2m's software assertions/validations and we now hit a brick wall because of the device itself?

In my mind i would have expected the device to be "braindead" when it comes to this and just accept whatever, but maybe that isn't the case with OTA vs serial flashing? The fact the OTA process went on for over 2 hours made me thing that it would work...

Koenkk, 2022-04-06:
@robertalexa once the firmware has been transferred to the device (this is what happens during the two hours)  the device executes some validation before flashing it, this is what seems to fail.

closed 2022-04-06 by robertalexa