Koenkk/zigbee-OTA #104: Image for TuYa TV02 (2.0.4) not parseable

fettlaus, 2022-04-03:
Today I intercepted a request upon updating a TuYa TRV TV02 via Tuya Smart Life. The linked binary however sems to not be parseable by ```add.js```. Either I took a completely wrong turn somewhere or the binary is not compatible with the script.
```
PS D:\Workspaces\zigbee-OTA> node scripts/add.js https://images.tuyaeu.com/smart/firmware/upgrade/ay156144560900410bUk/164155152229641863dfc.bin
node:internal/errors:464
    ErrorCaptureStackTrace(err);
    ^

RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:371:5)
    at boundsError (node:internal/buffer:84:11)
    at Buffer.readUInt16LE (node:internal/buffer:243:5)
    at Object.parseImage (D:\Workspaces\zigbee-OTA\lib\ota.js:17:34)
    at main (D:\Workspaces\zigbee-OTA\scripts\add.js:65:24)
    at processTicksAndRejections (node:internal/process/task_queues:96:5) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}
```
Intercepted request:
```yaml
{
    "result": [
        {
            "controlType": 1,
            "currentVersion": "1.0.5",
            "devType": 0,
            "lastUpgradeTime": 0,
            "timeout": 60,
            "type": 3,
            "typeDesc": "ZigBee Module",
            "upgradeStatus": 0
        },
        {
            "canUpgrade": true,
            "controlType": 1,
            "currentVersion": "2.0.0",
            "desc": "In order to cooperate with the upgrade of the Tuya platform, \nthe upgrade will be carried out before the device is used.\nNotes:\n1. Please do not delete or unbind the device  in the APP when upgrading or
upgrade failed, otherwise, the device will be unusable.\n2. Ensure good network when upgrading.\n3. Please do not power off the device when upgrading.\n4. Upgrade failed, the device displays \"---\", please follow the app prompt t☼upgrade again.\n5. If there are multiple temperature control valves, pls upgrade in turn. The device will display \"UP\" when upgrading\n6. After the upgrade, the device will  restart and restore the connection.",
            "devType": 0,
            "diffOta": false,
            "fileSize": "49032",
            "firmwareDeployTime": 1645523068,
            "lastUpgradeTime": 0,
            "md5": "5a416bf477da3330aae8b9f56a8c90e1",
            "timeout": 60,
            "type": 9,
            "typeDesc": "MCU Module",
            "upgradeStatus": 1,
            "upgradeType": 0,
            "upgradingDesc": "Please keep the power of the device connected during the upgrade process, please be patient.",
            "url": "https://images.tuyaeu.com/smart/firmware/upgrade/ay156144560900410bUk/164155152229641863dfc.bin",
            "version": "2.0.4"
        }
    ],
    "status": "ok",
    "success": true,
    "t": 1648976003782
}
```

jowlo, 2022-04-08:
I had a quick look at the linked file and `lib/ota.js` and this file does not contain the OTA Update file header `0xbeeef11e` (see [the Zipbee cluster library spec, section 11.4.2.1](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf). The script thus fails to parse the data format.

The file looks to be some sort of firmware, as it contains a few strings like `In Hard Fault Handler` and `0123456789abcdef`. I have not much experience with zigbee at all and no idea if tuya is doing something out of the spec here, maybe their gateway adds a header to the actual payload?

vladi1234, 2022-04-09:
Hello @fettlaus,
what's up with the new "version": "2.0.4"? Are there any improvements?

Impact123, 2022-04-09:
The important one is that periodically sending it `local_temperature_calibration` is not needed anymore. It sends data on its own now. I didn't see a changelog nor did I notice any other differences so far.

vladi1234, 2022-04-09:
Upgraded TV02-Zigbee with new version 2.0.4 sends through z2m every ~30 min?
Or through TuyaHUB?

Impact123, 2022-04-09:
Through Zigbee2MQTT. I only used the tuya gateway to upgrade and intercept the traffic before I found this already existing issue while trying to contribute my findings.

vladi1234, 2022-04-09:
I have already made adjustments and am now trying to create a pull request.

pabigot, 2022-04-10:
On current master f51ed0445422b49f098eef3404eaa9e3119b2230 including #106 and #107 when running:

    tirzah[122]$ node scripts/updateall.js 

I get this:
<details>
<summary>Click to expand</summary>

```
Updated existing entry ({"fileVersion":87,"fileSize":234174,"manufacturerCode":4098,"imageType":5634,"sha512":"1113486a67403d922fc4a462b7477e0a0986fa8862a89b74ab8c5eac7b4ecb31c30f1da9ae1bbc1d6d50ae1fed66d38c37a1834ab71fd384c0555002783aba13","url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Tuya/1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin","path":"images/Tuya/1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin"})

node:internal/process/promises:246
          triggerUncaughtException(err, true /* fromPromise */);
          ^

RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:363:5)
    at boundsError (node:internal/buffer:84:11)
    at Buffer.readUInt16LE (node:internal/buffer:243:5)
    at Object.parseImage (/mnt/devel/external/HomeAssistant/zigbee-OTA/lib/ota.js:17:34)
    at main (/mnt/devel/external/HomeAssistant/zigbee-OTA/scripts/add.js:65:24)
    at Object.<anonymous> (/mnt/devel/external/HomeAssistant/zigbee-OTA/scripts/add.js:136:8)
    at Module._compile (node:internal/modules/cjs/loader:1095:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1124:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:816:12) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}
node:child_process:826
    err = new Error(msg);
          ^

Error: Command failed: node ./scripts/add.js "images/Tuya/TV02-Zigbee_MCU_Module_2.0.4_164155152229641863dfc.bin" ""
node:internal/process/promises:246
          triggerUncaughtException(err, true /* fromPromise */);
          ^

RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:363:5)
    at boundsError (node:internal/buffer:84:11)
    at Buffer.readUInt16LE (node:internal/buffer:243:5)
    at Object.parseImage (/mnt/devel/external/HomeAssistant/zigbee-OTA/lib/ota.js:17:34)
    at main (/mnt/devel/external/HomeAssistant/zigbee-OTA/scripts/add.js:65:24)
    at Object.<anonymous> (/mnt/devel/external/HomeAssistant/zigbee-OTA/scripts/add.js:136:8)
    at Module._compile (node:internal/modules/cjs/loader:1095:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1124:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:816:12) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}

    at checkExecSyncError (node:child_process:826:11)
    at Object.execSync (node:child_process:900:15)
    at /mnt/devel/external/HomeAssistant/zigbee-OTA/scripts/updateall.js:8:38
    at Array.forEach (<anonymous>)
    at main (/mnt/devel/external/HomeAssistant/zigbee-OTA/scripts/updateall.js:7:15)
    at Object.<anonymous> (/mnt/devel/external/HomeAssistant/zigbee-OTA/scripts/updateall.js:15:8)
    at Module._compile (node:internal/modules/cjs/loader:1095:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1124:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:816:12) {
  status: 1,
  signal: null,
  output: [
    null,
    Buffer(0) [Uint8Array] [],
    Buffer(933) [Uint8Array] [
      110, 111, 100, 101,  58, 105, 110, 116, 101, 114, 110,  97,
      108,  47, 112, 114, 111,  99, 101, 115, 115,  47, 112, 114,
      111, 109, 105, 115, 101, 115,  58,  50,  52,  54,  10,  32,
       32,  32,  32,  32,  32,  32,  32,  32,  32, 116, 114, 105,
      103, 103, 101, 114,  85, 110,  99,  97, 117, 103, 104, 116,
       69, 120,  99, 101, 112, 116, 105, 111, 110,  40, 101, 114,
      114,  44,  32, 116, 114, 117, 101,  32,  47,  42,  32, 102,
      114, 111, 109,  80, 114, 111, 109, 105, 115, 101,  32,  42,
       47,  41,  59,  10,
      ... 833 more items
    ]
  ],
  pid: 3846075,
  stdout: Buffer(0) [Uint8Array] [],
  stderr: Buffer(933) [Uint8Array] [
    110, 111, 100, 101,  58, 105, 110, 116, 101, 114, 110,  97,
    108,  47, 112, 114, 111,  99, 101, 115, 115,  47, 112, 114,
    111, 109, 105, 115, 101, 115,  58,  50,  52,  54,  10,  32,
     32,  32,  32,  32,  32,  32,  32,  32,  32, 116, 114, 105,
    103, 103, 101, 114,  85, 110,  99,  97, 117, 103, 104, 116,
     69, 120,  99, 101, 112, 116, 105, 111, 110,  40, 101, 114,
    114,  44,  32, 116, 114, 117, 101,  32,  47,  42,  32, 102,
    114, 111, 109,  80, 114, 111, 109, 105, 115, 101,  32,  42,
     47,  41,  59,  10,
    ... 833 more items
  ]
}
```
</details>

Removing the entry from `index.json` allowed me to run the script.

vladi1234, 2022-04-10:
Hello @fettlaus,

Do you have the same `"manufacturerName":  "_TZE200_hue3yfsn"` ?

vladi1234, 2022-04-15:
Hello @fettlaus,
Despite all this, I have the same error.

mrmuro, 2022-04-19:
Hello for me it is the same, the update still does not work.

jowlo, 2022-04-19:
Like I said in my comment above, the update file is not parseable as it does not contain the ZigBee OTA header (as other TuYa firmware does).

Nothing changed in that regard and PR #106 did not change any code, just manually added the metadata.

sebastianschauenburg, 2022-05-06:
Also getting the same error when trying to update the firmware of this device: TS0601 / _TZE200_hue3yfsn / [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa) / [TV02-Zigbee](https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee)
Am running the latest docker image (latest push was 5 days ago).

Is there anything I can do / you need, to help get this move along?

fettlaus, 2022-05-06:
After researching a bit further it seems that this file is really just the binary for upgrading the mcu. This seems to be done by utilizing special TUYA-commands and not the usual zigbee ota. See also [here](https://developer.tuya.com/en/docs/iot-device-dev/tuya-zigbee-universal-docking-access-standard?id=K9ik6zvofpzql) (TUYA_MCU_OTA_NOTIFY etc.).

I guess unless someone implements this in z2m we're out of luck.

Also this is probably the reason why #106 isn't working. It's just not the right way to upgrade using this file.

MattWestb, 2022-05-09:
@fettlaus More then likely is one BIN file for the MCU. 
All TS0601 / tuya MCU devices is having the same firmware (more correct 2 Zigbee modules and 2 versions for sleeping and no sleeping devices) in the Zigbee module and can being updated with Zigbee OTA files is tuya have making it right.

The MCU firmware updating need implanting of the tuya MCU firmware updating code or flashing it with one J-Link flasher if its one plane bin-file that can being flashed standard way.

owangen, 2022-05-30:
Trying to add firmware update for Connecte Thermostat and have same problem I think. Getting ERR_BUFFER_OUT_OF_BOUNDS when using 
`node scripts/add.js https://images.tuyaeu.com/smart/firmware/upgrade/bay1628040814785F7DM/1641969211de5c0fe5900.bin`

Intercepted response:
`{
    "result": [
        {
            "controlType": 1,
            "currentVersion": "1.0.3",
            "devType": 0,
            "lastUpgradeTime": 0,
            "timeout": 60,
            "type": 3,
            "typeDesc": "ZigBee Module",
            "upgradeStatus": 0
        },
        {
            "canUpgrade": true,
            "controlType": 1,
            "currentVersion": "1.0.6",
            "desc": "1. Repair zigbee3. 0 upload multiple of temperature value\n2. Fixed January and February week errors",
            "devType": 0,
            "diffOta": false,
            "fileSize": "38724",
            "firmwareDeployTime": 1643530974,
            "lastUpgradeTime": 0,
            "md5": "39b58db8ae44adcdffa10f508f8c60d0",
            "timeout": 100,
            "type": 9,
            "typeDesc": "MCU Module",
            "upgradeStatus": 1,
            "upgradeType": 0,
            "upgradingDesc": "Please keep the power of the device connected during the upgrade process, please be patient.",
            "url": "https://images.tuyaeu.com/smart/firmware/upgrade/bay1628040814785F7DM/1641969211de5c0fe5900.bin",
            "version": "1.0.7"
        }
    ],
    "status": "ok",
    "success": true,
    "t": 1653861857250
}
`

MattWestb, 2022-05-30:
You is trying updating the Zigbee module with the firmware for the tuya MCU that looks being one bad thing to do.
tuya MCU us using DP commands for updating the MCU that is not implanted in the open system for the moment.

owangen, 2022-05-30:
Yes I understood that this is no normal OTA file and found this issue. I'll keep an eye if someone with the right knowledge implements support for Tuya MCU update in z2m. @MattWestb do you know if it's proposed as a new feature to z2m? I could not find any.

MattWestb, 2022-05-31:
I have not seen any request in Z2M, deCON or ZHA but normally some creative person is getting one device that need it and is start cocking and its being made if all is going well :-))
By the way great that you have getting the URL to the firmware so it can being used in the future then all pieces is in place and tuya MCU is working !!

TheJulianJES, 2022-07-01:
Still an issue. Per the above comments, the OTA file doesn't seem valid at all. I'd consider removing it for now.

pabigot, 2022-08-03:
PR #146 removes this firmware.

pabigot, 2022-08-04:
@Koenkk Thanks for merging #146; I forgot to add the magic text there to automatically close this issue.

closed 2022-08-04 by Koenkk

hidayahhtaaufik, 2022-09-07:
still got this issue

Tntdruid, 2022-11-02:
I got the issue too

william-aqn, 2022-11-04:
I got the issue too - (No image available for imageType '5634')

tracetechnical, 2023-10-25:
*bump* Having this issue too

lubomirp, 2023-11-05:
probably still no way how to update TV02-Zigbee without TuYa hub, right?