Koenkk/zigbee-OTA #114: ERR_BUFFER_OUT_OF_BOUNDS

mdeweerd, 2022-05-13:
I've already applied this update and I wanted to add it to the list for others:

```shell
node scripts/add.js https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v5.0/ZLinky_router_v5.0.ota
```

However I get:
```raw
node:internal/errors:466
    ErrorCaptureStackTrace(err);
    ^

RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:377:5)
    at boundsError (node:internal/buffer:84:11)
    at Buffer.readUInt16LE (node:internal/buffer:243:5)
    at Object.parseImage (D:\mdeweerd\workspace\zigbee-OTA\lib\ota.js:17:34)
    at main (D:\mdeweerd\workspace\zigbee-OTA\scripts\add.js:65:24)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}
```

I tried `node scripts/add.js http://fds.dc1.philips.com/firmware/ZGB_100B_010D/1107323831/Sensor-ATmega_6.1.1.27575_0012.sbl-ota` which executed without any issue.

mdeweerd, 2022-05-13:
I get the same message with, but I do not know if zip files are supported:
```shell
node scripts\add.js https://files.danfoss.com/download/Heating/Ally/Danfoss%20Ally%20Radiator%20Thermostat%201.18%20OTA%20file%2001122021.zip
```

Also, when performing:
```shell
wget https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v5.0/ZLinky_router_v5.0.ota
node scripts/add.js ZLinky_router_v5.0.ota
```

I get

```raw
Added new entry ({"fileVersion":5,"fileSize":240478,"manufacturerCode":4151,"imageType":1,
"sha512":"692417e8863827f120cda063850aee65b7b6eb58109d419ff58dced1d7c9251617312368bda0f848
8aa26aeb7a8e3642339c7648af24cbff66b870401c9113da","url":"https://github.com/Koenkk/zigbee-
OTA/raw/master/images/Eurotronic/ZLinky_router_v5.0.ota","path":"images/Eurotronic/ZLinky_
router_v5.0.ota"})
```

Which proves that it works, but the image is put under "Eurotronic" while the manufacturercode is actually "Jennic", so the component manufacturer's id, not Eurotronic.    I am opening #115 for that.

mdeweerd, 2022-05-15:
Note:
This also occurs when performing an updateall.js

```raw

node:internal/errors:466
    ErrorCaptureStackTrace(err);
    ^

RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside
buffer bounds
    at new NodeError (node:internal/errors:377:5)
    at boundsError (node:internal/buffer:84:11)
    at Buffer.readUInt16LE (node:internal/buffer:243:5)
    at Object.parseImage (D:\mdeweerd\workspace\zigbee-OTA\lib\ota.js:17
:34)
    at main (D:\mdeweerd\workspace\zigbee-OTA\scripts\add.js:65:24)
    at Object.<anonymous> (D:\mdeweerd\workspace\zigbee-OTA\scripts\add.
js:136:8)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Module._load (node:internal/modules/cjs/loader:827:12) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}

Node.js v18.1.0
node:internal/errors:845
  const err = new Error(message);
              ^

Error: Command failed: node ./scripts/add.js "images/Tuya/TV02-Zigbee_MC
U_Module_2.0.4_164155152229641863dfc.bin" ""
node:internal/errors:466
    ErrorCaptureStackTrace(err);
    ^

RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside
buffer bounds
    at new NodeError (node:internal/errors:377:5)
    at boundsError (node:internal/buffer:84:11)
    at Buffer.readUInt16LE (node:internal/buffer:243:5)
    at Object.parseImage (D:\mdeweerd\workspace\zigbee-OTA\lib\ota.js:17
:34)
    at main (D:\mdeweerd\workspace\zigbee-OTA\scripts\add.js:65:24)
    at Object.<anonymous> (D:\mdeweerd\workspace\zigbee-OTA\scripts\add.
js:136:8)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Module._load (node:internal/modules/cjs/loader:827:12) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}
```

Koenkk, 2022-05-16:
This error means the firmware doesn't use the standard ZCL format or it is encrypted.

mdeweerd, 2022-05-16:
In https://github.com/Koenkk/zigbee-OTA/issues/114#issuecomment-1125819387 above I showed that the following works:

```shell
wget https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v5.0/ZLinky_router_v5.0.ota
node scripts/add.js ZLinky_router_v5.0.ota
```

While adding the link like this does not work:
```script
node scripts/add.js https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v5.0/ZLinky_router_v5.0.ota
```

If the issue would be the file format, then adding the downloaded file from the same link should not work either.

Koenkk, 2022-05-16:
I see now, script was not handling redirect which caused the issue. Fixed that now.

closed 2022-05-16 by Koenkk

mdeweerd, 2022-05-16:
Ok, we "cross-posted".