Koenkk/zigbee-OTA #130: GLEDOPTO GL-C-008P

Krolik1988, 2022-06-17:
Hello, after z2m-edge update From:
```
Zigbee2MQTT version    1.18.3-dev commit: 3d00338
Coordinator type    zStack12
Coordinator revision    20201128
Frontend version    0.4.16
```
To:
```
Zigbee2MQTT version    1.25.2-dev commit: [820c20a](https://github.com/Koenkk/zigbee2mqtt/commit/820c20a)
Coordinator type    zStack12
Coordinator revision    20201128
Frontend version    0.6.101
```

start to receive notifications:
"warning 2022-06-17 12:08:33Firmware on '0x84fd27fffec35f33' is newer than latest firmware online."

```
Zigbee Model    GL-C-008P
Zigbee Manufacturer    GLEDOPTO
Description    Zigbee LED Controller RGB+CCT (pro)
Firmware build date    20210714
Firmware version    10976802
```

If I can provide it to you, please give me instructions how to. 

Have a good day.

Ionic97, 2022-07-08:
Similar issue on my side. I think ever since I received my device, it had the firmware V_1_8_T1 with the build date 20210527 on it. "Firmware on '0xcc86ecfffe1e777b' is newer than latest firmware online."
The firmware versions in the repo seem to be older though, so this message is legit in my case. Confusing is the naming scheme. Is there any newer version known?

iculookn, 2022-07-27:
Received this from Gledopto. Have not tested it

[GL-C-008P_V11076801_OTAV12_211020%.zip](https://github.com/Koenkk/zigbee-OTA/files/9197435/GL-C-008P_V11076801_OTAV12_211020.zip)

MartB, 2022-08-16:
This update keeps popping up but it seems to break some controllers.

See https://github.com/Koenkk/zigbee-OTA/commit/b6c8127ccd70d05ff4a2e75a9196478f7cb68c3c and https://github.com/Koenkk/zigbee-OTA/pull/125

closed 2022-09-23 by github-actions[bot]