Koenkk/zigbee-OTA #144: Tuya Smart Plug (20A) TS011F Image

V4n1X, 2022-08-02:
Here is a image I got that you can add:

Model: _TZ3000_cehuw1lw

https://images.tuyaeu.com/smart/firmware/upgrade/20220602/1654157434-oem_zg_tl8258_plug_OTA_1.0.13.bin

Should fix the random power off issue and enhance connectivity.

V4n1X, 2022-08-02:
Updated existing entry ({"fileVersion":77,"fileSize":307682,"manufacturerCode":4417,"imageType":54179,"sha512":"97ea8413e8ab662f2bebf3b013b3030754cbc2c0744b91b14741fff17a2d8d99e7088aa22a3b6e43b3595ccd65433f96d32516fc057fddcc13a816c921af8c97","url":"https://images.tuyaeu.com/smart/firmware/upgrade/20220602/1654157434-oem_zg_tl8258_plug_OTA_1.0.13.bin"})

closed 2022-08-02 by V4n1X

cloudbr34k84, 2022-09-27:
> Here is a image I got that you can add:
> 
> Model: _TZ3000_cehuw1lw
> 
> https://images.tuyaeu.com/smart/firmware/upgrade/20220602/1654157434-oem_zg_tl8258_plug_OTA_1.0.13.bin
> 
> Should fix the random power off issue and enhance connectivity.

Hey, how do you apply this as i have the TS011F
by _TZ3000_typdpbpg and maybe this might fix the issue if it just turning off??

SgtBatten, 2022-12-31:
Did you make this @V4n1X 

How did you "get" it?