Koenkk/zigbee-OTA #168: lumi.plug not completing update

mzanetti, 2022-10-19:
I've got a lumi.plug (not mmeu01, maeu01 or the likes, just "lumi.plug") which claims there is a firmware available and starts updating, however, upon finishing it responds with a status code "aborted" in the updateEndRequest call. Interestingly it also reports an imageType of "0" in the updateEndRequest which seems wrong, as for all previous calls the imageType is 257.

After testing a lot with it, my current understanding is, that it is the wrong firmware for this device and there must be more to it than comparing manufacturerCode, imageType and (in case of lumi) model type. It could also be that my device's fimware is too old and it can't do the big jump (like hue for instance has the minVersion and maxVersion properties for this) but would need some firmware version in between.

The version currently reported by the device is 0.0.0.22. The available version is 0.0.0.92. The selected update image file is https://github.com/Koenkk/zigbee-OTA/blob/master/images/Xiaomi/lumi.plug_20211224_v92.ota (manufacturer 4447, imageType 257, modelName lumi.plug).

All seems correct to me up until the epdateEndRequest coming from the device....

Here's my lumi's node/endpoint/cluster descriptor:

```
 D | Zigbee: --> ZigbeeNode(0x4677, 00:15:8D:00:02:93:0F:EF, LUMI (0x115f), lumi.plug, Router, RxOn:true)
 D | Zigbee:   ZigbeeNodeEndpoint(0x01, Zigbee::ZigbeeProfileHomeAutomation, Zigbee::HomeAutomationDeviceSmartPlug)
 D | Zigbee:   Manufacturer: "LUMI"
 D | Zigbee:   Model "lumi.plug"
 D | Zigbee:     Input clusters ( 9 )
 D | Zigbee:      - ZigbeeCluster(0x0000, Basic, Server)
 D | Zigbee:        -  ZigbeeClusterAttribute(0x0001, ZigbeeDataType(Unsigned 8-bit integer, 0x01))
 D | Zigbee:        -  ZigbeeClusterAttribute(0x0004, ZigbeeDataType(Character string, LUMI))
 D | Zigbee:        -  ZigbeeClusterAttribute(0x0005, ZigbeeDataType(Character string, lumi.plug))
 D | Zigbee:        -  ZigbeeClusterAttribute(0xff01, ZigbeeDataType(Character string, d))
 D | Zigbee:      - ZigbeeCluster(0x0001, PowerConfiguration, Server)
 D | Zigbee:      - ZigbeeCluster(0x0010, BinaryOutput, Server)
 D | Zigbee:      - ZigbeeCluster(0x0002, DeviceTemperature, Server)
 D | Zigbee:      - ZigbeeCluster(0x0003, Identify, Server)
 D | Zigbee:      - ZigbeeCluster(0x0004, Groups, Server)
 D | Zigbee:      - ZigbeeCluster(0x0005, Scenes, Server)
 D | Zigbee:      - ZigbeeCluster(0x0006, OnOff, Server)
 D | Zigbee:        -  ZigbeeClusterAttribute(0x0000, ZigbeeDataType(Bool, 0x00))
 D | Zigbee:        -  ZigbeeClusterAttribute(0xf000, ZigbeeDataType(Unsigned 32-bit integer, 0x0e 0x00 0x00 0x07))
 D | Zigbee:      - ZigbeeCluster(0x000a, Time, Server)
 D | Zigbee:     Output clusters ( 3 )
 D | Zigbee:      - ZigbeeCluster(0x0000, Basic, Client)
 D | Zigbee:      - ZigbeeCluster(0x000a, Time, Client)
 D | Zigbee:      - ZigbeeCluster(0x0019, OtaUpgrade, Client)
 D | Zigbee:   ZigbeeNodeEndpoint(0x02, Zigbee::ZigbeeProfileHomeAutomation, Zigbee::HomeAutomationDeviceMainPowerOutlet)
 D | Zigbee:     Input clusters ( 3 )
 D | Zigbee:      - ZigbeeCluster(0x000c, AnalogInput, Server)
 D | Zigbee:        -  ZigbeeClusterAttribute(0x0055, ZigbeeDataType(Single precision, 0x00 0x00 0x00 0x00))
 D | Zigbee:      - ZigbeeCluster(0x0001, PowerConfiguration, Server)
 D | Zigbee:        -  ZigbeeClusterAttribute(0x0021, ZigbeeDataType(Unsigned 8-bit integer, 0x00))
 D | Zigbee:      - ZigbeeCluster(0x0400, IlluminanceMeasurement, Server)
 D | Zigbee:        -  ZigbeeClusterAttribute(0x0000, ZigbeeDataType(Unsigned 16-bit integer, 0x47 0x4b))
 D | Zigbee:     Output clusters ( 2 )
 D | Zigbee:      - ZigbeeCluster(0x000c, AnalogInput, Client)
 D | Zigbee:      - ZigbeeCluster(0x0004, Groups, Client)
 D | Zigbee:   ZigbeeNodeEndpoint(0x03, Zigbee::ZigbeeProfileHomeAutomation, Zigbee::HomeAutomationDeviceMeterInterface)
 D | Zigbee:     Input clusters ( 1 )
 D | Zigbee:      - ZigbeeCluster(0x000c, AnalogInput, Server)
 D | Zigbee:        -  ZigbeeClusterAttribute(0x0055, ZigbeeDataType(Single precision, 0x5d 0x4e 0x43 0x3c))
 D | Zigbee:     Output clusters ( 1 )
 D | Zigbee:      - ZigbeeCluster(0x000c, AnalogInput, Client)
 D | Zigbee:   ZigbeeNodeEndpoint(0x64, Zigbee::ZigbeeProfileHomeAutomation, Zigbee::HomeAutomationDeviceOccupacySensor)
 D | Zigbee:     Input clusters ( 1 )
 D | Zigbee:      - ZigbeeCluster(0x000f, BinaryInput, Server)
 D | Zigbee:     Output clusters ( 2 )
 D | Zigbee:      - ZigbeeCluster(0x000f, BinaryInput, Client)
 D | Zigbee:      - ZigbeeCluster(0x0004, Groups, Client)
```

closed 2022-11-27 by github-actions[bot]

mzanetti, 2022-11-27:
FWIW, this is still an issue. If the solution is "we don't care", then so be it, I can understand that.

volcanodave, 2023-01-10:
@mzanetti some people over [here](https://github.com/Koenkk/zigbee2mqtt/issues/6906) found a solution today

mzanetti, 2023-01-12:
Oh nice! Without having tried it, it does read exactly like what I'm seeing here, yes. Thank you.