Koenkk/zigbee-OTA #199: When the device OTA to the latest,  check for the new updates, it still show i can update

huanzaii, 2022-12-02:
I use zigbee2mqtt(version:1.28.4-1), and check for the new updates for a device, and successfully OTA from version v20 to version v21.

But when i click check for the new updates again, it still show i can update, i update again, it shows OTA success, but the version is still v21.  And click check for the new updates again, it still show there is a update

I search modelid in [https://github.com/Koenkk/zigbee-OTA/blob/master/index.json](url), it shows v21 is the latest firmware.

In log, when OTA complete, it shows : XXX was updated from 'null' to 'null'

So, why i have already OTA my device to the latest, click check for the new updates, it still show i can update?

weihuan1111, 2022-12-05:
@Koenkk

TheJulianJES, 2022-12-18:
What's the Zigbee model id + image type + manufacturer id, ...?

tomasdeml, 2022-12-21:
I have the same problem when updating Aqara Curtain Driver E1 (Rod Version). Relevant logs with OTA image file:

current
```json
{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":24} 
```

latest meta
```json
{"fileVersion":6168,"fileSize":288213,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota","sha512":"0d1bae25759420898d381b98e97858d2e47f48cd3495abd8754661d0e1957eb5b570a2ae7435acc435d5a18665409e07148abae6f1f4a190c9af786d94ec725d"}
```

huanzaii, 2022-12-23:
> What's the Zigbee model id + image type + manufacturer id, ...?

@TheJulianJES Zigbee model id is 3Rsb22BZ, image type is 54184, manufacturer id is 4659
[image]

closed 2023-01-31 by github-actions[bot]