Koenkk/zigbee-OTA #213: Document how and where firmware updates are extracted or obtained

deviantintegral, 2022-12-21:
I had an opportunity to play around with mitmproxy's [new wireguard mode](https://docs.mitmproxy.org/stable/concepts-modes/#wireguard-transparent-proxy) and figured mitm'ing the update process for my Hue lights was a good test since they support updates via the app with Bluetooth.

So, I went through that and ended up with one of the URLs over at https://github.com/Koenkk/zigbee-OTA/pull/212. I realized, from this repo it's not clear:

1. What types of firmware files are basically public and have indexes or easily guessable URLs.
2. What types of firmware can be obtained through an app, _without_ using the vendor's hub (like Hue).
3. What types of firmware require a vendor hub and Zigbee network (like Sengled devices).
4. Links to any scripts that use existing APIs to find new firmware.
 * https://github.com/Koenkk/zigbee-OTA/pull/20 mentions "a script" for Philips, but I can't find one anywhere. From what I can tell the firmware update API at `https://firmware.meethue.com/v1/checkUpdate?version=16784902&deviceTypeId=100b-11f` has no auth and only requires the device type ID. It seems like it'd be relatively straight forward to automate that.

I think it would be helpful if the README indicated what types of firmware users need to contribute and discover manually, especially if they require mitm'ing traffic.

TheJulianJES, 2022-12-22:
Regarding Philips Hue images:
The ``https://firmware.meethue.com/v1/checkUpdate`` API is the alternative Hue OTA API that only provides some images. This is the API that's used by the Hue app when connecting lights only via Bluetooth (and not Zigbee). This is also why it only provides a couple of OTA images (so newer models only).
The Hue Bridge "API" is more complicated and requires "bridge-specific data". I can't explain this further though.

You'll find more information here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4124#issuecomment-1097487059

deviantintegral, 2022-12-22:
> The Hue Bridge "API" is more complicated and requires "bridge-specific data". I can't explain this further though.

_nods_ so then, how are you getting firmware updates currently?

TheJulianJES, 2022-12-22:
While the old Hue Bridge API still provides updates, they're outdated at the moment.
I'm currently working on getting updates from the latest Hue Bridge API. (This requires physical access to a Hue Bridge.)

SgtBatten, 2022-12-31:
is there a way to create firmware and fix issues with a device where the manufacturer refuses to?

TheJulianJES, 2022-12-31:
Not really, no

deviantintegral, 2023-01-22:
Given that we don't have anything to write, I'm closing this. At least if someone else searches they'll hopefully find this issue. @TheJulianJES if you end up writing up what you find, please consider linking to it here. Thanks!

closed 2023-01-22 by deviantintegral