Koenkk/zigbee-OTA #230: Feature request: check.js

Kilill, 2023-01-04:
OTA files are supposed to be correct from the supplier but case in point the EcoDim addition was far from it.

As an addition to add.js , it would be nice if it was possible to do a sanity check on the ota file
before adding it. (with ample error messages....)

Also the esp32 chip is coming out with a zigbee/matter/.... version so it is to be expected that more DIY zigbee things are going to pop up...

Koenkk, 2023-01-05:
Checks are already being done: https://github.com/Koenkk/zigbee-OTA/blob/3a28aa12796e8c53b13bed9a840c22085c791644/lib/ota.js#L11, what more do you expect?

Kilill, 2023-01-05:
based on my experiences with the EcoDim , maybe a warning about non conformant version number, the recommended format is 4 bytes split as AppRelease AppBuild StackReleas StackBuild
the EcoDim file had just the lower byte set, which is ok according to the spec, but still a bit weird,
also not sure what instance complained when i tried to flash the file but it failed with: "_Image padding contains invalid bytes_" so there seems to be some checks done by the OTA flash routines themselves on the file. Would have been beneficial to catch those before the file gets added.

closed 2023-02-13 by github-actions[bot]