Koenkk/zigbee-OTA #263: [REQUEST] Full compatibility of index.json with zigpy's upcoming standard for Generic OTA providers?

Hedda, 2023-02-24:
@Koenkk please consider looking into if the index.json in your [zigbee-OTA](https://github.com/Koenkk/zigbee-OTA) repository matches the draft (work-in-progress) "Generic OTA providers" which [puddly](https://github.com/puddly) is working on for zigpy which could become a universal [open standard](https://en.wikipedia.org/wiki/Open_standard) and [open file format](https://en.wikipedia.org/wiki/Open_file_format) within the Zigbee open source scene that could possibly be reused by all open source Zigbee framework/gateway projects as a shared common standard:

https://github.com/zigpy/zigpy/pull/1165

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

For reference please check out puddly's draft PR here for zigpy which will add support for Generic OTA providers -> https://github.com/zigpy/zigpy/pull/1165

> Implements the format specified in https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers:
> 
> ```yaml
> zha:
>   zigpy_config:
>     ota:
>       remote_providers:
>         - url: "https://fw.zigbee.example.org/ota.json"
>           manufacturer_ids: [0x1234, 0x5678]
>         - url: "https://fw.zigbee.example.org/ota-beta.json"
> ```
> 
> This allows for new providers to be added to zigpy without extra code. This also allows for manufacturers to distribute test feed URLs to customers without deploying them globally.

puddly has also added a new feature to the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool that manufacturers can use to produce a skeleton JSON index from a given set of OTA files, and that way make sure manufacturers OTA files are correctly-structured so that they will be compatible with zigpy out-of-the-box. See the README.md for the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool:

https://github.com/zigpy/zigpy-cli/blob/dev/README.md#ota

This work by  [puddly](https://github.com/puddly) is a follow-up to these previous zigpy discussions about standardizing of a generic format for index.json support:

https://github.com/zigpy/zigpy/issues/535

https://github.com/zigpy/zigpy/discussions/654

PS: [zigpy](https://github.com/zigpy/zigpy) is a suite of open source libraries implementing a Zigbee stack that is used in several popular open-source home automation projects, including [zigpy](https://github.com/zigpy/zigpy) (Zigbee library used by [Home Assistant ZHA integration](https://www.home-assistant.io/integrations/zha), [Zigbee Plugin for Jeedom](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/), and [Zigbee Plugin for Domoticz](https://www.domoticz.com/wiki/ZigbeeForDomoticz), and more).

PPS: [puddly](https://github.com/puddly) has previously also co-designed the "[Open ZigBee Coordinator Backup Format](https://github.com/zigpy/open-coordinator-backup)" which is another universal [open standard](https://en.wikipedia.org/wiki/Open_standard) and [open file format](https://en.wikipedia.org/wiki/Open_file_format) that is already adopted by mentioned projects and more (including  [zigbee-herdsman](https://github.com/Koenkk/zigbee-herdsman) which is the Zigbee framework library used by [Zigbee2MQTT](https://www.zigbee2mqtt.io) and [IoBroker](https://www.iobroker.net/#de/adapters/adapterref/iobroker.zigbee/README.md) )-> https://github.com/zigpy/open-coordinator-backup

Koenkk, 2023-02-25:
The current `index.json` can already be re-used by all open source Zigbee framework/gateway projects, I'm not sure why a new standard has to be defined.

Hedda, 2023-02-27:
@Koenkk yes that is absolutely true, however, I still think it would be a very good idea for you to look into this and if possible try to get in on collaborating with your input early on in this effort on the documentation of it by at least giving your feedback on the concept and formatting, as chances are high that at least a few companies will sooner or later adopt their specific formatting of a universal [open standard](https://en.wikipedia.org/wiki/Open_standard) and common [open file format](https://en.wikipedia.org/wiki/Open_file_format).

While I do not know his motivation for sure I can make an educated guess as I know that puddly now works for [Nabu Casa](https://www.nabucasa.com/about/) together with a few other lead developers of Home Assistant, with him specifically very recently becoming employed there to work on the underlying zigpy framework libraries which Home Assistant's ZHA integration depends on, and I assume that it should now be their interest to try in the longterm try to improve standardization of some decentralized things in order to offer a better experience for end users, especially with such things like Zigbee firmware updates via OTA (Over-The-Air) being one of the major things that Zigbee device manufacturers are not collaborating on.

Therefore, I believe that from their point-of-view as well as other projects willing to maybe adopt it in the future this idea of having a documented universal [open standard](https://en.wikipedia.org/wiki/Open_standard) and common [open file format](https://en.wikipedia.org/wiki/Open_file_format) that can be used universally even by completing companies so they can simply point at as this documentation as a default reference which should directly be beneficial for all Zigbee projects that adopt that universal/common open standard as well as any companies that start to use this, and because Nabu Casa / Home Assistant and other projects that adopted this will have this documented universal/common open standard in one shared public location you can then not only share that standard publicly with multiple device hardware manufacturers but also try to convince all and any companies that they partner with to also adopt this universal/common open standard as then their devices can get out-of-the-box OTA support from many different projected without any further customizations being required.

For reference, Home Assistant / Nabu Casa recently started a partnership program which could indirectly allow them to put some pressure on those companies they partner with to adopt this new open standard if hardware manufacturers want to earn the right to use "**_Works with Home Assistant_**" badges for their devices once they tested compatibility. See:

https://www.home-assistant.io/blog/2022/07/12/partner-program/

https://partner.home-assistant.io

https://www.home-assistant.io/blog/2022/10/13/third-reality-partner/

https://www.home-assistant.io/blog/2022/07/27/leviton-partner/

closed 2023-04-07 by github-actions[bot]

Hedda, 2023-08-23:
FYI, ZHA and zigpy now support this new index format -> https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

That PR [zigpy/zigpy#1165](https://github.com/zigpy/zigpy/pull/1165) has been merged to zigpy (and upstream implementations like Home Assistant's ZHA integration), adding support for Generic OTA providers via this new standard.

Still wondering if it would be a good idea to generate a JSON index file with zigpy-cli (https://github.com/zigpy/zigpy-cli/blob/dev/README.md#generate-ota-index-files) for the OTA images that are hosted on GitHub in Koenkk's zigbee-OTA repository at https://github.com/Koenkk/zigbee-OTA/ and if so then maybe Koenkk could be convinced to host such an alternative index in the same GitHub repository if Zigbee2MQTT will not be adopting this new standard? That is, an additional index file to https://github.com/Koenkk/zigbee-OTA/blob/master/index.json that instead follow zigpy's new JSON index file format as documented here in zigpy's wiki -> https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

Hedda, 2023-08-23:
Oh, forgot to mention that this is the new standard that Home Assistant / Nabu Casa will ask its commercial partners to adopt:

If you are new aware, Home Assistant (and Nabu Casa) have introduced an "**_Works with Home Assistant_**" partner program with its matching "**_Works via Zigbee with Home Assistant_**" compatibility badges and for companies that joins that  partnership program.

https://www.home-assistant.io/blog/2022/07/12/partner-program/

There it says "**_Works with Home Assistant partners are expected to provide firmware updates via user-friendly services provided by Home Assistant_**" under **_Responsibilities_** in a section there called "**_Provide easy firmware updates_**".

https://partner.home-assistant.io/

"_**There are manufacturers that are creating products that integrate into Home Assistant using standards like Z-Wave, Zigbee, or Matter (soon). In these cases, the integration is maintained by the Home Assistant community and Nabu Casa. These companies can still become a member of the Works with Home Assistant program but are relieved from integration maintenance.**_"

"_**With Home Assistant we are always working on educating our users about preferring local control and open standards when acquiring new products. This is also reflected in the “Works with Home Assistant” badges.**_"

[image]

PS: So far they have only announced one other company which makes and sells Zigbee devices that joined their partner program:

https://www.home-assistant.io/blog/2022/10/13/third-reality-partner/