Koenkk/zigbee-OTA #264: New Danfoss Ally firmware

JornDL, 2023-02-25:
[ID438430853294-0301.zip](https://github.com/Koenkk/zigbee-OTA/files/10830409/ID438430853294-0301.zip)

Hi

There is a new Danfoss Ally firmware from the 6th february 2023:
1246-0100-01200120.0002_(2B60A723)

from this site:
https://www.danfoss.com/en/products/dhs/smart-heating/smart-heating/danfoss-ally/danfoss-ally-support/#tab-software

Could you please make this available ?

BR Jørn Dalgård Larsen

Koenkk, 2023-02-25:
Done!

closed 2023-02-25 by Koenkk

gecko17, 2023-03-01:
@Koenkk I am not sure if this change is correct. The new image comes with a new image type "288", so now the entry in the index.json is not correct anymore. Also I guess the update would fail (I didn't try). When you download the image from the referenced site, there is also a pdf document. If I understood correctly, the new image is only for the new microcontroller, which is probably why they changed the imageType. This also explains why my newest Popp Thermostat came with the same imageType 288.

Please correct me, if I'm wrong as I am not an expert in OTA.

JornDL, 2023-03-02:
I just updated two of them without any issues.

suaveolent, 2023-03-03:
Just opened this issue, and found this thread:

https://github.com/Koenkk/zigbee2mqtt/issues/16892

hotchkj, 2023-06-12:
As I commented over on #265, the OTA file doesn't look to be correctly set up by Danfoss. The actual file has an `imageType` of `256`. The PDF with it indicates the new G-prefixed Allys are `imageType` `288` but there doesn't seem to be any firmware posted for them. This is particularly frustrating, since the default firmware with mine (`00.20.0008 00.20`) doesn't have external sensors working.

mdeweerd, 2023-12-13:
To make sure which TRVs are compatible and which are not, I suggest to report the working references.

I have some old devices that show 014G2420 Next to "Danfoss Ally TM".
The update to 1.20 failed but I can not exclude 100% that this is not due to the network.

Basic Cluster/HW VERSION/0x0003 indicates 69 .
The TRVs I try to update have 1.08 in them - I successfully updated a few in the past to 1.18.
The Basic Cluster/sw_build_id/0x4000 is 01.08.0008 01.08 at this time.