Koenkk/zigbee-OTA #265: Can not find OTA image for Danfoss Ally TRV

gedemagt, 2023-02-25:
Hi,

I recently bought a Danfoss Ally TRV, which is nicely supported by Z2M, and there are OTA updates in this repo.
However, when I try to check for updates, it says that it could not find any images for imageType 288
[image]

I can see that in you index.json in this repo, the imageType for Danfoss is also 256. 

Can you help me with why this doesn't recognize the correct version and type? My device looks like this:
[image]

lasse-aagren, 2023-02-27:
I have same issue with newly bought ally's

red-island, 2023-03-05:
As I do no permission to create a branch/pull request here the solution for the coders:

This is the new model from Danfoss Ally => there is a new image id 0x0120
Hence, index.json needs to get a new entry.

Copy record with "manufacturerCode": 4678, "imageType": 256
and add it with new **"imageType": 288**

[image]

gedemagt, 2023-03-05:
@Koenkk I created a PR for the suggestion by red-island. Please review: https://github.com/Koenkk/zigbee-OTA/pull/274

gedemagt, 2023-03-06:
Hmm, I tried to download the latest and upload it locally, but I keep getting an "imageType mismatch" error. I downloaded the firmware from here: https://www.danfoss.com/en/products/dhs/smart-heating/smart-heating/danfoss-ally/danfoss-ally-support/#tab-software and the .pdf that comes along clearly states that is should be the new imageType of 288 (0x120). 

Is it z2m or the device that responds check the imageType?

closed 2023-04-14 by github-actions[bot]

hotchkj, 2023-06-12:
I'm unconvinced Danfoss have actually set this data up correctly. I'm in the same situation, and the debug logs for the OTA file indicate:

```
getNewImage for '0xa46dd4fffe37f642', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4678,"imageType":256,"fileVersion":276,"zigbeeStackVersion":2,"otaHeaderString":"13-12-2022 V.0114.0114\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":385719}
```
Note the imageType for the file is still listed as `256`, contrary to what their PDF does indeed say.