Koenkk/zigbee-OTA #266: Lots of firmware with zero size

jrbenito, 2023-02-27:
Hi,

I download all the firmware referenced in `index.json` and found that many of than are zero sized:

```
~/ota$ find ./ |wc -l
254
find ./ -size 0|wc -l
211
```

Is that expected?

Thanks

Koenkk, 2023-02-27:
Can you provide the list of firmwares that are zero sied?

jrbenito, 2023-02-27:
@Koenkk 

First thanks for quick reply.

The problem is something related to curl, when I download (batch) the firmware to my HassOS, I've listed all URL and passed then to a `curl URL -output filename`. Many came zeroed, for instance:

```
$ curl https://github.com/Koenkk/zigbee-OTA/raw/master/images/Insta/ZLL_WS_4f_J_Release_10.03.32.02.zigbee --output ZLL_WS_4f_J_Release_10.03.32.02.zigbee
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 
 $ ls -l
total 0
-rw-rw-r-- 1 jbenito jbenito 0 Feb 27 13:52 ZLL_WS_4f_J_Release_10.03.32.02.zigbee
```
However, before composing the list to you, since I have wget on my main machine, I tried with it:

```
$ wget https://github.com/Koenkk/zigbee-OTA/raw/master/images/Insta/ZLL_WS_4f_J_Release_10.03.32.02.zigbee
--2023-02-27 13:52:40--  https://github.com/Koenkk/zigbee-OTA/raw/master/images/Insta/ZLL_WS_4f_J_Release_10.03.32.02.zigbee
Resolving github.com (github.com)... 140.82.113.3
Connecting to github.com (github.com)|140.82.113.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/Insta/ZLL_WS_4f_J_Release_10.03.32.02.zigbee [following]
--2023-02-27 13:52:41--  https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/Insta/ZLL_WS_4f_J_Release_10.03.32.02.zigbee
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.111.133, 185.199.108.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 134487 (131K) [application/octet-stream]
Saving to: ‘ZLL_WS_4f_J_Release_10.03.32.02.zigbee.1’

ZLL_WS_4f_J_Release_10.03.32.02.zigbee 100%[===========================================================================>] 131.33K  --.-KB/s    in 0.03s   

2023-02-27 13:52:41 (3.94 MB/s) - ‘ZLL_WS_4f_J_Release_10.03.32.02.zigbee.1’ saved [134487/134487]

```

Sorry, I should have realized this before opening the issue, I tried in HassOS (there is no wget there) and retried (with curl) on my on machine and since no errors from `curl` I wrongly assumed the problem was with the files.

I am closing this.

Thanks again.
Jose

closed 2023-02-27 by jrbenito