Koenkk/zigbee-OTA #293: How to create and run add.js script?

KoalaLyon, 2023-04-16:
Hello all, 

I'm kinda new here and a newbie with all that stuff. I have Zigbee2mqtt installed on my RPi since a while now and I tried to update some of my devices.
It appears that I do not have any index.json file on my directory nor the add.js script at all. I tried to follow all your readme section first but I do not understand how to place the add.js script, where. Do I need to update the add.js file given in zigbee-OT/scripts folder? 
If you can give me some tips and steps to figure it out...

Thank you very much.

Koenkk, 2023-04-16:
First [fork the repo](https://github.com/Koenkk/zigbee-OTA/fork) then clone it (`git clone https://github.com/ KoalaLyon/zigbee-OTA`). Alternatively, you can post the OTA files here so I can add them.

KoalaLyon, 2023-04-23:
Okay, so it worked. I cloned the repo on my Pi. Tried to execute `node update all.js` but it failed :

`node:fs:601
  handleErrorFromBinding(ctx);
  ^

Error: ENOENT: no such file or directory, open 'index.json'
    at Object.openSync (node:fs:601:3)
    at Object.readFileSync (node:fs:469:35)
    at main (/opt/zigbee2mqtt/zigbee-OTA/scripts/updateall.js:6:37)
    at Object.<anonymous> (/opt/zigbee2mqtt/zigbee-OTA/scripts/updateall.js:15:8)
    at Module._compile (node:internal/modules/cjs/loader:1254:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1308:10)
    at Module.load (node:internal/modules/cjs/loader:1117:32)
    at Module._load (node:internal/modules/cjs/loader:958:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:81:12)
    at node:internal/main/run_main_module:23:47 {
  errno: -2,
  syscall: 'open',
  code: 'ENOENT',
  path: 'index.json'
}

Node.js v18.16.0
`

If you have an idea... the Index.json file is in Zigbee-OTA folder.

Koenkk, 2023-04-24:
You need to execute in the root directory, not in `scripts` (then execute `node scripts/add.js PATH_TO_OTA_FILE_OR_URL`)

closed 2023-06-01 by github-actions[bot]