Koenkk/zigbee-OTA #296: Not possible to upgrade Hue white and color ambiance E26/E27/E14

m0uH, 2023-04-23:
I'm using the recent version 1.30.3 of Zigbee2MQTT and am no longer able to update the firmware of the Hue White and Color Ambiance light bulb (https://www.zigbee2mqtt.io/devices/9290012573A.html#philips-9290012573a).

In the past I was able to update to version 1.101.2 but with some recently used bulbs on version 1.88.1 and 1.50.2 updating is not possible:

> debug 2023-04-23 11:50:46Is new image available for '', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16785664}', latest meta '{"fileVersion":16783874,"fileSize":267452,"url":"https://otau.meethue.com/storage/ZGB_100B_010C/2ef158a5-ffb4-43ac-9d59-3cb71078f6f7/100B-010C-01001A02-ConfLight-Lamps_0012.zigbee","sha512":"c4591fe155bef8500779c36c7792f3960c4f83dde9dd47aa367113229c5bd73161f14cc92e6d6a0960e807c54626ce2ab0ce0d18c76d0206770dcda3a4776862"}'

> debug 2023-04-23 11:54:59Is new image available for '', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":268,"fileVersion":16786688}', latest meta '{"fileVersion":16783874,"fileSize":267452,"url":"https://otau.meethue.com/storage/ZGB_100B_010C/2ef158a5-ffb4-43ac-9d59-3cb71078f6f7/100B-010C-01001A02-ConfLight-Lamps_0012.zigbee","sha512":"c4591fe155bef8500779c36c7792f3960c4f83dde9dd47aa367113229c5bd73161f14cc92e6d6a0960e807c54626ce2ab0ce0d18c76d0206770dcda3a4776862"}'

Might be caused by the change in https://github.com/Koenkk/zigbee-OTA/pull/288/files#diff-7aebb122a6ea8a2749d60cb05b7e103c9eae6e2e85e48d2d6cd9e20b63013975R152, as afterwards there is two times the same firmware listed in index.json.

Using these versions of Zigbee2MQTT via the Hassio Addon:
Zigbee2MQTT version
    1.30.3 commit: unknown

Coordinator type
    zStack3x0

Coordinator revision
    20220928

Coordinator IEEE Address
    0x00124b0021cc3a1f

Frontend version
    0.6.127

donbobka, 2023-04-30:
Faced same issue. Fixed by overriding ota index: https://github.com/Koenkk/zigbee-OTA/pull/299

m0uH, 2023-05-06:
Thanks @donbobka this did fix the problem and I could upgrade my light bulbs!

closed 2023-05-06 by m0uH