Koenkk/zigbee-OTA #316: ERR_BUFFER_OUT_OF_BOUNDS

Schnittcher, 2023-05-30:
Hi, I want to add the firmware for the Tuya TV02.
I tried this: node scripts/add.js https://images.tuyaeu.com/smart/firmware/upgrade/ay156144560900410bUk/164155152229641863dfc.bin

An get this error:
```

node:internal/errors:490
    ErrorCaptureStackTrace(err);
    ^

RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:399:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt16LE (node:internal/buffer:245:5)
    at Object.parseImage (C:\Users\Kai\Documents\Git\zigbee-OTA\lib\ota.js:17:34)
    at main (C:\Users\Kai\Documents\Git\zigbee-OTA\scripts\add.js:92:28)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5) {
  code: 'ERR_BUFFER_OUT_OF_BOUNDS'
}

Node.js v18.16.0

```

Can anyone help?

Koenkk, 2023-05-31:
This does not seem to be a valid Zigbee OTA file, the upgrade identifier bytes are missing (`0x1E, 0xF1, 0xEE, 0x0B`)

Schnittcher, 2023-05-31:
Do you have any idea what I can do?

Schnittcher, 2023-05-31:
Unfortunately not.
I can try to pair the device again.

closed 2023-07-09 by github-actions[bot]

tracetechnical, 2023-10-25:
@Schnittcher  any joy with this? I think this could fix a large number of TV02 users' issues.

Daniel-WW, 2023-11-05:
Hi, what can I do if it is not possible to perform a firmware update with zigbee2mqtt? Do I have to buy a Tuya gateway? - I have five TV02 thermostats, all of which do not respond correctly to temperature changes. - A current firmware update is my last hope. - Thank you for every hint and advice.