Koenkk/zigbee-OTA #328: Legrand 067775/741811 update invalid image failure

yashijoe, 2023-06-19:
**What happened?**

Interface tells me there is an update
I try to upgrade and get the message “invalid image”

**What did you expect to happen?**

Upgrade successfully

**How to reproduce it (minimal and precise)**

Press upgrade firmware
Get the message:
“ Error 2023-05-17 09:18:40Update of 'refrigerator_plug' failed (Update failed with reason: 'invalid image')”

**Zigbee2MQTT version**

1.30.4-1

**Adapter firmware version**

004a

**Adapter**

Legrand 067775/741811

I’ve also tried with dev version and updated index.json. Same result

thank you

SilentT-FR, 2023-06-19:
Same problem for me

The Firmware files was downloaded from Legrand Website
https://static.developer.legrand.com/files/2023/05/1021-0011-004e4203-NLP.zip

I have open a post on the forum of legrand
https://developer.legrand.com/forums/topic/connected-outlet-nlp-78-firmware/

i Cant sniffing my zigbee traffic to capture the OTA

i have tested to capture from ethernet network but contrary from this howto : https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/Corner_Retreiving-Legrand-Firmware.md from @pipiche38 

i dont see same url
3	0.072753	192.168.1.88	51.145.143.28	HTTP	248	GET /firmware/0/00047432a0ec/17/74/1686370197/tiaLUbj6BjvLZXCRfIPne-l11eJ7vPp0GozOAxfY4NQ HTTP/1.1 Host: fw.netatmo.net:80\r\n

SilentT-FR, 2023-06-19:
@benjR You have Capture the 78 OTA Files and it was the same issue with your file, you have an idea about my capture ? i have this [Gateway](https://www.legrand.fr/pro/catalogue/pack-de-demarrage-drivia-with-netatmo-pour-installation-connectee-1-module-control)

@essegi65 just for follow the issue and an user with same issue with bticino product

@rossowl just for follow the issue, you have 46 outlet no news with the new file from legrand, you have better experiance ? 

@vezyr same issue follow the issue

@novisys same issue follow the issue

@sekot same issue i thinks follow the issue

@essegi65 same issue follow the issue

benjR, 2023-06-19:
I used another [gateway](https://www.legrand.fr/catalogue/interrupteur/pack-de-demarrage-eclairage-connecte-celiane-with-netatmo-commande-sans-fils-prise-control-mobile-et-ampoule-dimmable-blanc), but here is my theory :

I think the firmware files are fine (I used the same method for the other Legrand device I pushed and managed to update through z2m locally) but there is an issue on how the files are matched to the devices, seems like it's trying to apply to a wide scope of devices instead of just the single corresponding device.

When I added the fw file I sniffed I used the node tool provided to do so, and instead of adding some new lines it replaced another fw for another device in the json file.

So I'm thinking both issue could be the same : the detection of the device is not working well on Legrand files or products (same info? same ids?) therefore when trying to update it is probably using the wrong file on the wrong device.

SilentT-FR, 2023-06-19:
> I used another [gateway](https://www.legrand.fr/catalogue/interrupteur/pack-de-demarrage-eclairage-connecte-celiane-with-netatmo-commande-sans-fils-prise-control-mobile-et-ampoule-dimmable-blanc), [...]

here is your PR
https://github.com/Koenkk/zigbee-OTA/commit/3b61e152fbdfefa00c2df4351a454b5b82734208

you have writen : Legrand-412171-78.fw

NLP Firmware is for this devices : https://static.developer.legrand.com/files/2023/02/Connected-outlet.xlsx

Cloud legrand firmware say NLPO but i thinks its use the same firmware

[image]

SilentT-FR, 2023-06-19:
@benjR Comparing your file and the legrand firmware website

Only 2 lines is not equal the first and the last

buts its not exploitable file in notepad++ i dont know software can read this

SilentT-FR, 2023-07-01:
Normaly new info next week
https://developer.legrand.com/forums/topic/nlc-firmware-version/

yashijoe, 2023-07-01:
🤞🏻

trollix, 2023-07-03:
Same bug for me:
OTA doesn't work.

[image]

[image]

SilentT-FR, 2023-07-12:
Hi all Legrand respond : 
https://developer.legrand.com/forums/topic/nlc-firmware-version/

If anyone of subscribed user with same problem can capture the traffic and identify the version from you start upgrade 

For me from v74 (        "installed_version": 4867071, ) and v62 ("installed_version": 4080131) not work
003e  > v62
004a > v74 
(hex to dec)

SilentT-FR, 2023-07-25:
Ehhhh

i have testing to simply reboot the socket by using my circuit breaker and i have clic update just after apear online
and is go and running to the end

i make more socket tommorow,,, you now WAF...

If you can test other... 

Cordially

SilentT-FR, 2023-07-26:
Not work for other socket...

SilentT-FR, 2023-07-26:
Its because it was in older version

tkintscher, 2023-07-26:
Hmm, interesting. I have two wall switches here (067771) and am trying to upgrade them.

Both have `"installed_version": 4604927` and `"latest_version": 4669955`.

The first one was not power cycled. This shows the "invalid image" error (see before_reboot.txt).

The second one I rebooted with the circuit breaker (for ~15 sec), then clicked update as soon as the switch appeared again. Here I see several `commandQueryNextImageRequest` messages. Then nothing for a few minutes. Finally I get `Error: Timeout: device did not request any image blocks`.

I attach the two logs from z2m, but probably it's the same that you folks see...
[after_reboot.txt](https://github.com/Koenkk/zigbee-OTA/files/12176696/after_reboot.txt)
[before_reboot.txt](https://github.com/Koenkk/zigbee-OTA/files/12176697/before_reboot.txt)

tkintscher, 2023-07-26:
I also do not understand in the case of NLF:

It says on developer.legrand.com that the version is _72_.
The filename is `1021-000e-00474203-NLF.zigbee` and
0x47 (hex) = _71_ (dec)

Now, 474203 is the actual version in the file header, but maybe there was a mistake when uploading?

essegi65, 2023-08-02:
I updated to version 1.32.2 but I still have problems updating the "bticino" K4531C devices (zigbee associates them as LEGRAND 067775/741811) currente firmware version 004A.
Makes "invalid image" error

SilentT-FR, 2023-08-03:
No news from legrand forum
i have upload a sniffing capture i thinks its analyse it

titou4307, 2023-08-14:
Same problem for me : impossible to update Legrand 412170

Firmware passed of 0025 to 004e on 1/3 Legrand 412170
Firmware not pass of 0025 to 004e on 2/3 Legrand 412170

Does it possible to force the falsh with the firmware file under Z2M

benjR, 2023-08-15:
Didn't have the time to look more at it recently, but now the OTA check fails on all my Legrand devices with this message, never had it before on any device, tried with my Legrand 067776 and 412171 devices.

[image]

benjR, 2023-08-15:
> Didn't have the time to look more at it recently, but now the OTA check fails on all my Legrand devices with this message, never had it before on any device, tried with my Legrand 067776 and 412171 devices.
> 
> [image]

Had to re-pair all my devices to get rid of it.

Koenkk, 2023-08-25:
@SilentT-FR I think we should disable OTA for now, agree?

titou4307, 2023-08-25:
I don't know what to do...
Many sujects opened

lamerjack, 2023-08-26:
Same issue with a Bticino F20T60A that is a Legrand 412015 say invalid image.
If i check ota on a updated unit say that the firmware in the unit is more recent.
I tested also with SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P latest firmware. and with a different brand dongle.

Koenkk, 2023-08-27:
@titou4307 does it work for anyone? If not, we should disable it.

pierrepinon, 2023-08-27:
@Koenkk it's impossible to upgrade legrand devices to latest firmware. So I think you can disable it !

benjR, 2023-08-27:
Agree @Koenkk we can disable it until we find out why the fw files do not work as intended.

titou4307, 2023-08-27:
Hello
I don't know how but another one of my Legrand DIN smart relay for light control (BTicino FC80RC )  was updated today...

I have at this time one who cannot updated

Xavinooo, 2023-08-27:
Last week, I've managed to successfully update a wired switch (Legrand 067771), but it took hours... (device not responding).
Now, it says "the firmware installed is newer".

But

In the state section, I have:
```
"update": {
        "installed_version": 4735999,
        "latest_version": 4669955,
        "state": "idle"
    }
```

4735999 converted to hex is 4843FF. Bringing it to 8 digits, it's 0048 43FF.
0048 is the version displayed on the about page of the device.

I also have a Legrand hub. 

There, it says the version is 72.

And 72 is 0x48 in hex.

Maybe, a first enhancement, is to convert the version to be more precise in z2m.

This trick works also for a connected outlet (Legrand 067775) which I updated with the Legrand hub.
I have a second outlet not updated available to test, if one day it works in z2m.

Sadly, the page https://developer.legrand.com/production-firmware-download/ is not up-to-date. It says the latest firmware for a wired switch is 63. I have the version 72 installed...
Last page update was on July 11.

But I don't know if any of that helps...

titou4307, 2023-08-27:
Hi
After many try, many manipulations I have succeed to pass the update of my three LEGRAND DIN smart relay for light control (BTicino FC80RC )

pierrepinon, 2023-08-28:
Can you explain the manipulation ?

titou4307, 2023-08-28:
@pierrepinon 

If you need the manipulation, it needeed money !  (I'm kidding so I goes out 10 minutes...)

I am trying to give a response : 

I have done the how to give by @rossowl  and it desactivate the process of the update
So I reactivate the update process and I don't know why but one of my two LEGRAND mount and done the update but not the second (my last LEGRAND....)
But this second LEGRAND has not the same firmware than other two (it was more recent) so my idea is the server Legrand may be has not put in priority this device
I am not sure of all this argument

I have trying the same process (with mod, re mod the state.json ...start container Z2M) and it is not function
The same error message came.....No respon OTA update or similar

So I have try other thing....

I remaining the module NODON not be updated all the time (only a during time after the power ON) so I have power off the circuit .....
In first time wait 1 or 2 minutes and after I rearm the power
I have waiting a time to the Z2M to refind the device and add into the "mesh network" and I have click on the Update button on panel Z2M
At this time I see the update process begin to 0.01% or 0.03 % and then fall (message error) but the news is the process begin a little...

I have redone the manipulation : 
power off the circuit .....
But time wait 5 minutes....
clik on the button on the LEGRAND to pair (for to discharge all electricity in the device)
after I rearm the power
 waiting a time to the Z2M to refind the device and add into the "mesh network" 
click on the Update button on panel Z2M

And at this time the process update begin and goes to the end fine....

My idea is may be if the process not pass the first time, the device keep a part of code (x % or firmware downloaded....) and block all new try ...
So it needeed to cut the power a time for to made the memory of device empty and try a new download update

I am not sure of my reflexion but it have succeed for me

Sorry for my english, i am frenchy so poor traductor I am

@pierrepinon  You are french so we can have message in french by MP for after complete here in english

Titou43 (from haute loire)

pierrepinon, 2023-08-28:
I will try today thx

sekot, 2023-08-29:
for info :
my new Legrand 067775/741811 with usine firmware version : 003e > v62.
no upgrade problem
they are now on : 004e > v78
only my older power sokets on firmware 004a > v74 (original firmware in 003e) are stuck with "invalide image"

chpego, 2023-08-29:
Hi,
My ZigBee device (Legrand - Cable Outlet - 064882) won't upgrade from firmware version 001e to the last available here (31) with the same error : `failed (Update failed with reason: 'invalid image')`

SilentT-FR, 2023-09-11:
```
debug 2023-09-11 17:27:02Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Salon Télé Prise Gauche","transaction":"n3vd2-4"}'
info 2023-09-11 17:27:03MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'Salon Télé Prise Gauche' to latest firmware","meta":{"device":"Salon Télé Prise Gauche","status":"update_in_progress"},"type":"ota_update"}'
debug 2023-09-11 17:27:03Received Zigbee message from 'Salon Télé Prise Gauche', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":5129215,"imageType":17,"manufacturerCode":4129}' from endpoint 1 with groupID 0
debug 2023-09-11 17:27:03Got OTA request '{"fieldControl":0,"manufacturerCode":4129,"imageType":17,"fileVersion":5129215}'
debug 2023-09-11 17:27:03ZigbeeOTA: downloaded main index
debug 2023-09-11 17:27:03ZigbeeOTA: Loading override index /config/zigbee2mqtt/last.json
debug 2023-09-11 17:27:03getNewImage for '0x00047400001161ef', meta {"fileVersion":5325315,"fileSize":251175,"url":"https://github.com/SilentT-FR/zigbee-OTA/raw/9de7f8a557b171b05ec9c4b772db3c7962307b65/images/Legrand/1021-0011-00514203-NLP-81.zigbee","sha512":"d16c29b0eccad02caa043e600a23d87b76055722a4abf35dbb3eb63d9b768de9bb630e34667327023afb0e0e18a3edf998f48f41da53e9218b5f772b7dfe7b2e"}
debug 2023-09-11 17:27:03ZigbeeOTA: downloading firmware image from https://github.com/SilentT-FR/zigbee-OTA/raw/9de7f8a557b171b05ec9c4b772db3c7962307b65/images/Legrand/1021-0011-00514203-NLP-81.zigbee
debug 2023-09-11 17:27:03OTA update checksum validation succeeded for '0x00047400001161ef'
debug 2023-09-11 17:27:03getNewImage for '0x00047400001161ef', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4129,"imageType":17,"fileVersion":5325315,"zigbeeStackVersion":2,"otaHeaderString":" ","totalImageSize":251175}
debug 2023-09-11 17:27:07Received Zigbee message from 'Salon Télé Prise Gauche', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":5129215,"imageType":17,"manufacturerCode":4129}' from endpoint 1 with groupID 0
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Salon Télé Prise Gauche' at 0.02%, ≈ 952 minutes remaining","meta":{"device":"Salon Télé Prise Gauche","progress":0.02,"status":"update_progress"},"type":"ota_update"}'
info 2023-09-11 17:27:15MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Salon Télé Prise Gauche' failed (Update failed with reason: 'invalid image')","meta":{"device":"Salon Télé Prise Gauche","status":"update_failed"},"type":"ota_update"}'
info 2023-09-11 17:27:15MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Salon Télé Prise Gauche"},"error":"Update of 'Salon Télé Prise Gauche' failed (Update failed with reason: 'invalid image')","status":"error","transaction":"n3vd2-4"}'
debug 2023-09-11 17:27:15Error: Update failed with reason: 'invalid image' at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:422:24
```

SilentT-FR, 2023-09-11:
@Koenkk I think that the old firmwares that we were previously captured by sniffing, made our device bricked but with the official hub its ok i can upgrade it, so i cant confirm

SilentT-FR, 2023-09-11:
@Koenkk and from out of box and only upgrade with firmware from legrand.com with z2m no problem

SilentT-FR, 2023-09-11:
@Koenkk is it possible that there is a parameter to send in ota which allows to force the upgrade on the object side

pierrepinon, 2023-09-11:
@SilentT-FR You can upgrade your devices with news firmware inside your PR in review ?

Koenkk, 2023-09-11:
@SilentT-FR we cannot fix an `invalid image` returned by the device from the z2m side (no forcing is possible here)

yashijoe, 2023-09-11:
> > @SilentT-FR You can upgrade your devices with news firmware inside your PR in review ?
> 
> Only with socket out of box only updated with firmware of legrand.com

Is the problem solved once updated with legrand gateway?
Anyway that is a mess, I’ve sockets behind kitchen forniture

SilentT-FR, 2023-09-11:
> > > @SilentT-FR You can upgrade your devices with news firmware inside your PR in review ?
> > 
> > 
> > Only with socket out of box only updated with firmware of legrand.com
> 
> Is the problem solved once updated with legrand gateway? Anyway that is a mess, I’ve sockets behind kitchen forniture

Not tested yet

SilentT-FR, 2023-09-11:
i think the legrand gateway do something different but i cant explain
im going to sniff i think but i need to find the key of the gateway

SilentT-FR, 2023-09-11:
This is the update failed with invalid image 
If anyone can analyse this @Koenkk ?
[Update_iageinvalid.zip](https://github.com/Koenkk/zigbee-OTA/files/12579307/Update_imageinvalid.zip)

Key : 01:03:05:07:09:0B:0D:0F:00:02:04:06:08:0A:0C:0D

Network address
    0x8071

SilentT-FR, 2023-09-11:
For the moment the Gateway Legrand abort the OTA request...

[OTA ABORT.zip](https://github.com/Koenkk/zigbee-OTA/files/12580199/OTA.ABORT.zip)

29:06:2A:5F:ED:79:DD:E5:F4:28:93:0B:CD:22:06:03

Koenkk, 2023-09-13:
> For the moment the Gateway Legrand abort the OTA request...
> 
> [OTA ABORT.zip](https://github.com/Koenkk/zigbee-OTA/files/12580199/OTA.ABORT.zip)
> 
> 29:06:2A:5F:ED:79:DD:E5:F4:28:93:0B:CD:22:06:03

Do I understand correctly that it also fails with the Legrand gateway?

SilentT-FR, 2023-09-16:
> > For the moment the Gateway Legrand abort the OTA request...
> > [OTA ABORT.zip](https://github.com/Koenkk/zigbee-OTA/files/12580199/OTA.ABORT.zip)
> > 29:06:2A:5F:ED:79:DD:E5:F4:28:93:0B:CD:22:06:03
> 
> Do I understand correctly that it also fails with the Legrand gateway?

No its because legrand provide ota update on the legrand gateway after 1 week of the available on the legrand developer website

zbee_nwk.addr == 0x4ce9 

[Uploading OTA begin.zip…]()

Key : 29:06:2A:5F:ED:79:DD:E5:F4:28:93:0B:CD:22:06:03
i upload the begenning of the OTA i have cut off to keep the socket not up to date

SilentT-FR, 2023-09-16:
as you can see the update begin and the socket ask more than two block

Koenkk, 2023-09-17:
> as you can see the update begin and the socket ask more than two block

I don't see any `ZCL OTA: Image Block Request`/`ZCL OTA: Image Block Response` in [OTA begin.zip](https://github.com/Koenkk/zigbee-OTA/files/12641179/OTA.begin.zip) , can you sniff for a bit longer?

SilentT-FR, 2023-09-17:
Yes i see, after 1600sec you can see the real begin
[OTA begin no response and real begin.zip](https://github.com/Koenkk/zigbee-OTA/files/12642704/OTA.begin.no.response.and.real.begin.zip)

SilentT-FR, 2023-09-17:
I see this difference : 

by Z2M

        File Version: 0x00514203
            0000 0000 .... .... .... .... .... .... = Application Release: 0
            .... .... 0101 0001 .... .... .... .... = Application Build: 81
            .... .... .... .... 0100 0010 .... .... = Stack Release: 66
            .... .... .... .... .... .... 0000 0011 = Stack Build: 3

By Legrand Gateway

        File Version: 0x00510000
            0000 0000 .... .... .... .... .... .... = Application Release: 0
            .... .... 0101 0001 .... .... .... .... = Application Build: 81
            .... .... .... .... 0000 0000 .... .... = Stack Release: 0
            .... .... .... .... .... .... 0000 0000 = Stack Build: 0

Legrand dev / Legrand from Gateway not use same image ?

SilentT-FR, 2023-09-17:
i double check
the downloaded file from legrand have same sha512 from my git folder and from a downloaded link from github

Koenkk, 2023-09-17:
Could it be that Legrand published the wrong OTA files on their site? The file version is parsed from the OTA file.

sekot, 2023-09-17:
the ota files work for new device.

SilentT-FR, 2023-09-20:
Legrand forum say

> We see that the product downloads the firmware starting the offstet 78 and not offstet 0. This case is hard to understand as it’s not reproduced on our side with our gateways and zigbee gateways tested

Legrand say 78 but i see 50 in the https://github.com/Koenkk/zigbee-OTA/files/12642704/OTA.begin.no.response.and.real.begin.zip packet number 6346

So is it not the Z2M or the file but the powersocket who asking not the beginning of the file offset

SilentT-FR, 2023-09-20:
So only method is it to poweroff for a long time period Legrand say

> Can you try the following :
> 
>     perform an electrical reboot of the device (and wait at least 30 seconds)
> 
> or
> 
>     make a factory reset, reinstall the device, then re-try to perform the update
>

SilentT-FR, 2023-09-20:
I have the chance where i can poweroff with the circuit breaker room by room but for those have circuit breaker for multiple room its hard !

SilentT-FR, 2023-09-20:
im going to test tonight

chpego, 2023-09-20:
> im going to test tonight

I will try too

chpego, 2023-09-21:
> > im going to test tonight
> 
> I will try too

Failed for me

FabianMangold, 2023-09-22:
Hi, sorry to jump in so late. Same issue for me, have a bunch of 067776 having the very same issue.
A long reset (power off) or factory reset doesn't help at all.

@Koenkk: Is there any chance the updater could be modified to send the offset requested by the device ?
If so, i am willing to potentially brick one of my switches to give this a try.

[75 more replies]

closed 2023-09-25 by Koenkk