Koenkk/zigbee-OTA #329: Some Ubisys devices moved to ota.zigbeeOTA and no longer pick up updates

sjorge, 2023-06-25:
Seems a while ago some of the Ubisys devices moved to the ota.zigbeeOTA provider... I don't understand why.

We have a zigbee ota.ubisys that is working fine, this weeks updates are not showing up for all devices using ota.zigbeeOTA but are showing up for those with ota.ubisys.

e.g.

[image]

sjorge, 2023-06-25:
| device  | provider | shows update with ota.ubisys | installs update with ota.ubisys |
| ------------- | ------------- | ------------- | ------------- |
| S1 | ota.ubisys | yes | untested |
| S1-R | ota.zigbeeOTA  | untested | untested |
| S2 + S2-R | ota.zigbeeOTA | untested | untested |
| D1 + D1-R | ota.zigbeeOTA | untested | untested |
| J1 + J1-R | ota.zigbeeOTA | untested | untested |
| C4 | ota.zigbeeOTA | untested | untested |
| H1 | ota.ubisys | yes | untested |
| R0 | ota.zigbeeOTA | untested | untested |

I'll update this table ones done testing

Koenkk, 2023-06-25:
See https://github.com/Koenkk/zigbee-OTA/pull/317 for the reason

sjorge, 2023-06-25:
> See #317 for the reason

That doesn't make much sense to me? If the ones fetched via ota.ubisys are the same ones as the ones manually available for download, there should be no difference (at the time the firmware as added, there is obviously a newer one available now)

We should at the very least I think, switch them all over to ota.zigbeeOTA then? I'll still round of my tests with switching them back to ota.ubisys though. I wonder if they user just got unlucky and he checked while ubisys was in the process of publishing them firmwares.

sjorge, 2023-06-25:
Something that also comes to mind, some devices need a 2 stage update as ubisys swapped firmware formats at some point, if you check the index used by ota.ubysis the older format one is listed as ota1.zigbee, AFAIK the still works when using ota.ubisys but it's probably broken with ota.zigbeeOTA as we cannot add multiple firmwares for a single devices AFAIK.

e.g.

10F2-7B01-0000-0006-01110206-spo-fmd.ota.zigbee 256748 5018ea0bf877e458ce78163e45f425dda7db16fb2190d3464dcb2fa496b0bdbc
10F2-7B01-0000-0006-0192020D-spo-fmd.ota1.zigbee 263324 facc5ac8cdd533b00559df80ea441d8bc8b1170504b4bac818307026b4783c16

sjorge, 2023-06-25:
@Koenkk 

Some more references as to why we also resole older images:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4466
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4491

I think the regex to pick the proper image is probably not 100% correct and somehow @WhistleMaster hit case where it returns the wrong image.

I do think we should figure out why it was not working for them and revert to ota.ubisys, as the current ota.zigbeeOTA approach will result in an incomplete upgrade path.

sjorge, 2023-06-25:
Updates one of each devices with ota.ubisys and they all still seem to work.

Koenkk, 2023-06-25:
Reverted it!

sjorge, 2023-06-25:
@WhistleMaster do you still happen to have one of those devices that you haven't updated yet?
If so would you mind adding some extra console.log() to see why the match is failing?

WhistleMaster, 2023-06-25:
Sure ! Where should I add those extra log ?

sjorge, 2023-06-25:
[Proba](https://github.com/Koenkk/zigbee-herdsman-converters/blob/3a30a5f5d1bad50ab985cffee653feeeeb97c168/src/lib/ota/ubisys.ts#L28)

Above that line try adding:
```
logger.debug(`FWMATCH - Device: ${JSON.stringify(device)}`);
logger.debug(`FWMATCH - ImageMatch 1-16: ${parseInt(imageMatch[1], 16)}, ImageMatch 2-16: ${parseInt(imageMatch[2], 16)}, ImageMatch 3-16: ${parseInt(imageMatch[3], 16))}, ImageMatch 4-16: ${parseInt(highestMatch[4], 16)}, ImageMatch hwVer: ${hardwareVersion}`);
```

It should snow up in the z2m log if you have debug level enabled, also make sure to switch to ota.ubisys while adding this.

There is also a chance it will just work TM now, given Ubisys pushed a new fw earlier this week.

sjorge, 2023-06-26:
OK that's interesting, 

You are currently on `2.0.0` which should be image type `0x7b33` (Going to try and find a way to verify this).

I did some digging and the major version changes were:
- `1.9.1`, image type goes to `0x7b23`
- `1.9.2`, image type goes to `0x7b33`

Assuming the device does indeed report `0x7b33` and a hwVer of 5 it should match one of these I think (`{manufacturerCode}-{imageType}-{hardwareVersionMin}-{hardwareVersionMax}-{fileVersion}-{?}-{?}`)
```
$ curl -s http://fwu.ubisys.de/smarthome/OTA/release/index | grep -i '7b33'
10F2-7B33-0000-0006-02120404-spo-fms2.ota.zigbee 144894 bdd07ed61f50767f72ab0825cae9c374a854957cd514c3c3646586d702978fca
10F2-7B33-0000-0006-02210412-spo-fms2.ota.zigbee 146942 2f99f07d83c71b703b5781fae617c6b7d5262dbe30ca7fae345024690a401a70
```

Can you replace L24 -> L35 with this instead:

```
    imageRegex.lastIndex = 0; // reset (global) regex for next exec to match from the beginning again
    let imageMatch = imageRegex.exec(firmwarePageHtml);
    let highestMatch = null;
    logger.debug(`OTA Ubisys: image type = ${current.imageType}`);
    while (imageMatch != null) {
        logger.debug(`OTA ubisys: image found: ${imageMatch[0]}`);
        if (parseInt(imageMatch[1], 16) === imageType &&
            parseInt(imageMatch[2], 16) <= hardwareVersion && hardwareVersion <= parseInt(imageMatch[3], 16)) {
            logger.debug(`OTA ubisys: potential match = ${imageMatch[0]}`);
            if (highestMatch === null || parseInt(highestMatch[4], 16) < parseInt(imageMatch[4], 16)) {
                logger.debug(`OTA ubisys: new highest match = ${imageMatch[0]}`);
                highestMatch = imageMatch;
            }
        }
        imageMatch = imageRegex.exec(firmwarePageHtml);
    }
```

And can you then past the debug log entry from when it's looping over the images, i would expect at least 2 loops to happen.

WhistleMaster, 2023-06-26:
I've made the changes, and I get the following in the log: [_zigbee2mqtt_logs.txt](https://github.com/Koenkk/zigbee-OTA/files/11870991/_zigbee2mqtt_logs.txt)

sjorge, 2023-06-26:
Ok this is weird, your current.imageType returns `0x7B23` and that one only matches 

```
10F2-7B23-0000-0006-0192020D-spo-fms2.ota.zigbee 129022 8cf2e80213f743980d778d10d4899d200862065a2cd29173b4105b51344bfd77
```

So it seems you somehow have an older bootloader (based on the change logs) with a newer firmware image.

The one you updated with the manually downloaded firmware already, what does that device report for imageType after the update?

WhistleMaster, 2023-06-26:
Well, I did not had the time to update with the manually downloaded firmware, so all my devices are like that as the PR was reverted.

Is there another way to update them then ?

sjorge, 2023-06-26:
I see, I was looking at the commit and it basically has the same imageType with the same file. So I'd be a bit surprised if that one matched while via ota.ubisys it didn't.

I guess we could test the following:
- offer the newest firmware ignoring the imageType match (seems like a bad idea)
- offer the older firmware matching the imageType and see if it wants to downgrade, then it should 🤞 update properly after.

@Koenkk I'm guessing if the image type doesn't match the device would reject it correct? That would rule out option 1.

WhistleMaster, 2023-06-26:
And by looking at the Firmware build date, all of them are **20191127-DE-FB1** but with Firmware version **2.0.0**

sjorge, 2023-06-26:
build_date never changes on any of mine when updating, I think it's like batch number when they were made.

I have 2x S2, both went through the 3 stage update process, one has date 20191128-DE-FB1 and one has 20200327-DE-FB1.

The weird thing is that yours seems to be reporting an older imageType but already on 2.0.0 fw

You can track the fw/image type changes via the changelog https://www.ubisys.de/en/support/firmware/changelog-s2/

0x7B23 and 2.0.0 never appears anywhere :/

WhistleMaster, 2023-06-26:
Indeed... that's really weird. Before, all my S2 were used with Deconz and updated with Deconz.

I hope I can find a way to update them with Z2M. If there were a way to "upload" a file directly to it, that could maybe help.

Koenkk, 2023-06-26:
> I don't think we can force a particular firmware, also a question for Koen. Even if we somehow do force send it a file, I think the device would just reject it.

This is possible with the local OTA index: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html

sjorge, 2023-06-26:
I guess @WhistleMaster can try and force downgrade using http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B23-0000-0006-0192020D-spo-fms2.ota.zigbee which is the only one of the correct image type. Once that it done, the regular OTA should work again. Assuming the device accepts that one.

If that works, would there be a way to have ota.ubisys somehow also have a force option? I think something probably went wrong when they were updated via deconz

WhistleMaster, 2023-06-26:
Thanks for the help. I've made a local index like that and add it to the Z2M configuration.

```
[
    {
        "fileVersion": 26345997,
        "fileSize": 129022,
        "manufacturerCode": 4338,
        "imageType": 31523,
        "sha512": "6b59596d69c0049ac8111ddec0edee16c7fe3353900e9d04eef7a3d42a7cda10152883e743b8114b2140957fc417fa64c62b15fc236a4d8d86a10cb786742cf2",
        "url": "http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B23-0000-0006-0192020D-spo-fms2.ota.zigbee",
        "force": true
    }
]
```

and changing `ota: ota.zigbeeOTA` for the S2 type in `src/devices/ubisys.ts`

Restarting Z2M and checking for firmware, I get the information that an update is available. I click on "Update firmware device" and after a bit I get:

`Update of 'switch_1' failed (Timeout: device did not request any image blocks)`

sjorge, 2023-06-26:
You still need to compile the .ts to a .js file... I'm not 100% sure on the steps but:
- checkout zigbee-herdsman-converter somewhere
- `npm run build` in that dir
- change to your zigbee2mqtt folder 
- `npm link /path/to/zigbee-herdsman-conveter

@Koenkk did I miss a step ?

WhistleMaster, 2023-06-26:
It looks like that it gets my local index as it says that a firmware is available, isn't it ? But I think that the device does not accept the firmware...

I've tried to change the local index to put the latest version available and I get:

`Failed to check if update available for 'switch_1' (No image available for imageType '31523')`

So I manually changed the imageType to 31523 instead of 31539 but I get:

`Update of 'switch_1' failed (Image type mismatch)`

sjorge, 2023-06-26:
Not sure it picked up the change, as http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B23-0000-0006-0192020D-spo-fms2.ota.zigbee is imageType 31523 / 0x7B23

WhistleMaster, 2023-06-26:
If I simply revert back to the original ubisys.ts, I get my old error that the firmware is newer and can't be updated even with my local index.

WhistleMaster, 2023-06-27:
I've written an email to Ubisys support to have some insights on the issue. 

By the way, how do you read the imageType on the Z2M web interface ? I've tried on the Dev Console, with Endpoint 232, Cluster OTA and Attribute imageTypeId but I get an error: Error: Cannot read properties of undefined (reading '0')

[image]

sjorge, 2023-06-27:
I'm not sure how to read it, that why I let you add the logging ealier.

Koenkk, 2023-06-27:
@sjorge those are the correct steps indeed!

WhistleMaster, 2023-06-27:
I’m using the Docker version, I guess that I can’t really follow the same steps, do I ?

What I don’t understand is why it seems that it takes my local index when modifying the ubisys.ts file through Docker console on my Z2M container ? I can’t “play” with the local index and it changes the error I get when I try to update.

WhistleMaster, 2023-06-28:
@sjorge, @Koenkk,

I got a reply from Ubisys:

```
The component already should have switched to image type 0x7b33 with the 1.9.2 firmware version. 
In fact without changing the OTA file it is not possible to update to 2.0.0 without having the correct image type.
```

I'm a bit lost as I'm not sure what should be done now. It seems not possible to have a different image type if I have the 2.0.0 firmware. I don't know what to do now and try to update the devices.

What about ignoring the imageType and force the latest firmware ?

It would be great to have a possibility to send a new firmware from the dashboard UI, by selecting it with an input file and a checkbox option to "force".

What are your thoughts on that one ?

sjorge, 2023-06-28:
So they basically confirm that the combination of requested imageType and firmware version 2.0.0 should be impossible.

If the device is willing to downgrade you could try and install the 1.9.2 firmware and see if it wants to upgrade later on.
But I think you tried that in https://github.com/Koenkk/zigbee-OTA/issues/329#issuecomment-1607977301 

I'm not really sure how to proceed.

WhistleMaster, 2023-06-28:
> So they basically confirm that the combination of requested imageType and firmware version 2.0.0 should be impossible.

Additional answer from Ubisys:

`as already mentioned, the image type should have already been changed to 0x7b33 with the update to 1.9.2, i.e. somehow your devices managed to update twice without having the correct image type. This has never been observed and should not be technically possible.`

> If the device is willing to downgrade you could try and install the 1.9.2 firmware and see if it wants to upgrade later on. But I think you tried that in [#329 (comment)](https://github.com/Koenkk/zigbee-OTA/issues/329#issuecomment-1607977301)

Yes indeed, this is what I tried first.

The problem is somewhere else but I don't know where 😅

closed 2023-08-05 by github-actions[bot]