Koenkk/zigbee-OTA #336: Nodon OTA doesn't work

dguihal, 2023-07-12:
> error 2023-07-12 17:43:45 Update of 'Volet bureau' failed (No image available for imageType '265')

Related to https://github.com/Koenkk/zigbee-OTA/pull/318 (looks like I'm not alone with the same error)

Z2M version 1.32.1
Zigbee module: Nodon SIN-4-RS-20

AlbanDurrheimer, 2023-08-04:
Same problem with:  
Z2M: 1.32.2  
Nodon module: Nodon SIN-4-RS-20 (x2)

stefa168, 2023-08-04:
Same problem here...
Zigbee2MQTT version: 1.32.2 (commit #1ec1e57)
Device: [SIN-4-2-20](https://www.zigbee2mqtt.io/devices/SIN-4-2-20.html#nodon-sin-4-2-20)

AlbanDurrheimer, 2023-09-04:
> Same problem with: Z2M: 1.32.2 Nodon module: Nodon SIN-4-RS-20 (x2)

Edit: the problem persists with Zigbee2MQTT version: 1.33.0-1

nanostra, 2023-09-08:
Hello, same problem with SIN-4-1-20 and SIN-4-2-20

dguihal, 2023-10-09:
keep alive

stefa168, 2023-10-09:
Up

AlbanDurrheimer, 2023-11-09:
keep alive

AlbanDurrheimer, 2024-01-03:
I still have the same problem with the Zigbee2MQTT 1.35.0 version.
Has anyone found a solution?

AlbanDurrheimer, 2024-01-03:
My last update log:
```
Zigbee2MQTT:debug 2024-01-03 10:37:56: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"shutter","transaction":"zxme3-1"}'
Zigbee2MQTT:info  2024-01-03 10:37:56: Updating 'shutter' to latest firmware
Zigbee2MQTT:debug 2024-01-03 10:37:57: Received Zigbee message from 'shutter', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2020"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-03 10:37:57: Updating to latest '0x9035eafffea1daa2' (SIN-4-RS-20� � ��T� T)
Zigbee2MQTT:debug 2024-01-03 10:37:57: Using endpoint '1'
Zigbee2MQTT:debug 2024-01-03 10:37:59: Received Zigbee message from 'shutter', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65794,"imageType":265,"manufacturerCode":4747}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-03 10:37:59: Got OTA request '{"fieldControl":1,"manufacturerCode":4747,"imageType":265,"fileVersion":65794}'
Zigbee2MQTT:debug 2024-01-03 10:37:59: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2024-01-03 10:37:59: getNewImage for '0x9035eafffea1daa2', meta {"fileVersion":65794,"fileSize":30206,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/NodOn/128b-0109-10102-700_nodon_sin_rs_fm_stm32_V10102.zigbee","sha512":"54f663c338f26aa89245edc79bbbdc911763897f2ca6ba92d72ce447f03f23d0d33a669af68e87ef84d95fa0df0231441f56fc45cb768a5e12ca2f1e77c2f1b9"}
Zigbee2MQTT:debug 2024-01-03 10:37:59: Update of 'shutter' failed (AssertionError [ERR_ASSERTION]: No new image available)
Zigbee2MQTT:info  2024-01-03 10:37:59: MQTT publish: topic 'zigbee2mqtt/shutter', payload '{"linkquality":232,"position":100,"state":"OPEN","tilt":0,"update":{"installed_version":523,"latest_version":543,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2024-01-03 10:37:59: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"shutter"},"error":"Update of 'shutter' failed (No new image available)","status":"error","transaction":"zxme3-1"}'
Zigbee2MQTT:error 2024-01-03 10:37:59: Update of 'shutter' failed (No new image available)
Zigbee2MQTT:debug 2024-01-03 10:37:59: AssertionError [ERR_ASSERTION]: No new image available
    at getNewImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:526:11)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:370:19)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:269:41)
Zigbee2MQTT:debug 2024-01-03 10:37:59: No converter available for '064882' with cluster 'manuSpecificLegrandDevices2' and type 'attributeReport' and data '{"0":5}'
Zigbee2MQTT:debug 2024-01-03 10:38:02: Received Zigbee message from 'shutter', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":523,"imageType":9,"manufacturerCode":4747}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-03 10:38:02: Device 'shutter' requested OTA
Zigbee2MQTT:debug 2024-01-03 10:38:03: Active device 'Bedroom Bulb 2' was last seen '123.19' minutes ago.
Zigbee2MQTT:debug 2024-01-03 10:38:05: Received Zigbee message from 'shutter', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65794,"imageType":265,"manufacturerCode":4747}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-03 10:38:05: Device 'shutter' requested OTA
Zigbee2MQTT:debug 2024-01-03 10:38:08: Received Zigbee message from 'shutter', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":523,"imageType":9,"manufacturerCode":4747}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-03 10:38:08: Device 'shutter' requested OTA
Zigbee2MQTT:debug 2024-01-03 10:38:11: Received Zigbee message from 'shutter', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65794,"imageType":265,"manufacturerCode":4747}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-03 10:38:11: Device 'shutter' requested OTA
```

closed 2024-08-01 by github-actions[bot]