Koenkk/zigbee-OTA #352: BTicino K4003 reports old firmware with installed_version greater than latest_version, can't update OTA

dadino, 2023-09-04:
**What happened?**

Installed firmware on BTicino K4003C is old (build date is 20190516 or 20190221 or 20190528), firmware version appears as unknown characters, but firmware installed_version is greater than latest_version. 
OTA update is not possible because it can't be forced.

Example json from one of my problematic switches:
```
{
    "last_seen": "2023-09-04T11:04:06.204Z",
    "led_if_on": "OFF",
    "led_in_dark": "ON",
    "linkquality": 43,
    "state": "OFF",
    "update": {
        "installed_version": 54527488,
        "latest_version": 3555843,
        "state": "idle"
    },
    "action": null,
    "update_available": null
}
```

**What did you expect to happen?**

Correct build numbers so that I can update the firmware, or a way to force OTA to latest_version

How to reproduce it (minimal and precise)

K4003C with old firmware installed, go to OTA page on Zigbee2Mqtt and "check for new updates", a notification with "No update available for 'switch_name'" will appear.

**Zigbee2MQTT version**

1.33.0-1

**Adapter firmware version**
Not sure how to get that

**Adapter**
tube-zb-gw-cc2652p2-v2

[image]

pierrepinon, 2023-09-13:
You can force update by updating ota json file with force field like below

[
    {
        "url": "HelloZigbee.ota",
        "force": true
    }
]

FabianMangold, 2023-10-03:
Hi @dadino,
This issue seems unrelated to OTA#328 at first glance. Issue 328 dealt with "invalid images", not with the firmware version, which is a related but different problem really.

I don't have any of those devices, so i really can't help. I assume you've tried the obvious power cycling / factory reset and so forth right ?
Have you tried to update to an "intermediate" version of the firmware instead ?

dadino, 2023-10-03:
No I haven't, I guess I'll search for a previous version.

FabianMangold, 2023-10-04:
Which Firmware gets downloaded on those devices ? Can you please check in the logs ?
Depending on that, we may try to find another version

dadino, 2023-10-05:
It's this entry in the index

```
{
        "fileVersion": 3556351,
        "fileSize": 255815,
        "manufacturerCode": 4129,
        "imageType": 15,
        "sha512": "e4d8ba85d002012227622e5a7ca685b98018c73788cb5c0c7287d4c9e39fcacc0fe6761ba1894f1818855099b02315e967f2438a589ba52530678d96dd490803",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-000f-003643FF-NLL.zigbee",
        "path": "images/Legrand/1021-000f-003643FF-NLL.zigbee"
    }
```

FabianMangold, 2023-10-05:
I was looking for an older one too, but i am afraid there is none available in the OTA repository. Sorry for that.
Do you have an official gateway at hand ?
What happens when you "force" the above firmware ? How does the device behave ? I have update 5 of mine with the latest FW without any issues really.

FabianMangold, 2023-10-05:
Okay, difficult to see what happens without additional details.
Once last thing (which may seem obvious), can you please power cycle the device (switch if off for at least a minute) and give it another shot ?
If that doesn't work, we may have to add some more debug output to the OTA process.

FabianMangold, 2023-10-09:
Hi @dadino,

Just to let you know that the DEV branch now features additional debug output.
In case the issue remains, could you please crank the log level up to debug and give the upgrade another try ?
Thanks

closed 2023-11-16 by github-actions[bot]

81ondo, 2024-12-30:
Still the same.