Koenkk/zigbee-OTA #361: looking for an older firmware version for a legrand switch

dimovnike, 2023-09-20:
hello,
is there a chance to find an olderfirmware version of this file https://github.com/Koenkk/zigbee-OTA/blob/77fe348e66f284354698640abcec16b0fb6e43a6/images/Legrand/1021-0024-00174203-NLIS.zigbee ? 

I heard the old version was working in the following way: press up for on and press down for off while current version does toggle for all presses. Also, interestingly enough, I found the same switch, exposed in a store that works like I would like. So any chance to find that firmware? Thanks!

FabianMangold, 2023-09-24:
Hi @dimovnike ,
Can you clone the zigbee-OTA repository to your machine, and then issue 
`git checkout 96fd0b3d58a00eb2be7834400194680783a5700d`
This will restore NLIS Firmware 1021-0024-00164203-NLIS-22.zigbee (version 22).

Does that help ?

dimovnike, 2023-09-25:
yes @FabianMangold , that worked, thanks!

dimovnike, 2023-09-25:
the checkout worked, but the firmware upgrade does not:

```
2023-09-25 15:48:39.761 DEBUG (MainThread) [zigpy.zcl] [0x25A3:1:0x0019] OTA query_next_image handler for ' Legrand  NLIS - Double light switch': field_control=0, manufacturer_id=4129, image_type=36, current_file_version=1524735, hardware_version=None, model=' NLIS - Double light switch'
2023-09-25 15:48:39.779 DEBUG (MainThread) [zigpy.zcl] [0x25A3:1:0x0019] OTA image version: 1524227, size: 253703. Update needed: True
2023-09-25 15:48:44.940 DEBUG (MainThread) [zigpy.zcl] [0x25A3:1:0x0019] OTA image_block handler for ' Legrand  NLIS - Double light switch': field_control=2, manufacturer_id=4129, image_type=36, file_version=1524227, file_offset=0, max_data_size=62, request_node_addr=None, block_request_delay=20
2023-09-25 15:48:44.940 DEBUG (MainThread) [zigpy.zcl] [0x25A3:1:0x0019] OTA upgrade progress: 0.0
2023-09-25 15:48:45.048 DEBUG (MainThread) [zigpy.zcl] [0x25A3:1:0x0019] OTA image_block handler for ' Legrand  NLIS - Double light switch': field_control=2, manufacturer_id=4129, image_type=36, file_version=1524227, file_offset=40, max_data_size=22, request_node_addr=None, block_request_delay=20
2023-09-25 15:48:45.049 DEBUG (MainThread) [zigpy.zcl] [0x25A3:1:0x0019] OTA upgrade progress: 0.0
2023-09-25 15:48:45.229 DEBUG (MainThread) [zigpy.zcl] [0x25A3:1:0x0019] OTA upgrade_end handler for ' Legrand  NLIS - Double light switch': status=Status.INVALID_IMAGE, manufacturer_id=4129, image_type=36, file_version=1524227
```

any ideas?

FabianMangold, 2023-09-25:
Hi @dimovnike,

We i can't help you with "zigpy" to be honnest, but ZHA has faced the same issue until recently.
You seem to be running a flawed Legrand Firmware on your device, which prevents the upgrade.
A workaround has been implemented on Z2M, so if possible, could you please try to downgrade your device using Z2M ?

FabianMangold, 2023-09-28:
Hi @dimovnike,
I just happened to have found https://github.com/zigpy/zigpy/issues/1259, you may want to follow-up on that instead.
Is that OK ?
Cheers

dimovnike, 2023-09-28:
yes, i will follow up on that one. Thanks!

closed 2023-09-28 by dimovnike