Koenkk/zigbee-OTA #365: Legrand 412170/412171 invalid image update failure

haade-administrator, 2023-09-24:
initial version: 5129215
new-version: 5325315

i have use script **app.js** and code into index.json ***is all good***, with files: https://static.developer.legrand.com/files/2023/09/1021-0011-00514203-NLP.zip:

```json
   {
        "fileVersion": 5325315,
        "fileSize": 251175,
        "manufacturerCode": 4129,
        "imageType": 17,
        "sha512": "d16c29b0eccad02caa043e600a23d87b76055722a4abf35dbb3eb63d9b768de9bb630e34667327023afb0e0e18a3edf998f48f41da53e9218b5f772b7dfe7b2e",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-00514203-NLP.zigbee",
        "path": "images/Legrand/1021-0011-00514203-NLP.zigbee"
    },
```

## debug

```
2023-09-24 18:17:22Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Chauffe eau","transaction":"r5b7e-3"}'
2023-09-24 18:17:22Received Zigbee message from 'Chauffe eau', type 'readResponse', cluster 'genBasic', data '{"dateCode":" \u000020201113\u0000\u0000\u0000\u0000\u0000","swBuildId":"004e"}' from endpoint 1 with groupID 0
2023-09-24 18:17:22Received Zigbee message from 'Chauffe eau', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":5129215,"imageType":17,"manufacturerCode":4129}' from endpoint 1 with groupID 0
2023-09-24 18:17:23Received Zigbee message from 'Chauffe eau', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":5129215,"imageType":17,"manufacturerCode":4129}' from endpoint 1 with groupID 0
2023-09-24 18:17:27Update of 'Chauffe eau' failed (Error: Update failed with reason: 'invalid image')
```

any idea ?

Koenkk, 2023-09-24:
should be fixed after https://github.com/Koenkk/zigbee-herdsman-converters/pull/6193

pierrepinon, 2023-09-24:
I try dev branch and now my devices can updating ! Thx all !

FabianMangold, 2023-10-03:
Hi @haade-administrator,
Have you had a chance to test the dev branch or the latest Z2M version. Has the update worked successfully ?

closed 2023-11-10 by github-actions[bot]

ilgio, 2023-11-24:
is there not a solution?

FabianMangold, 2023-11-24:
Hi @ilgio,
What is the issue ? Isn't it solved with the latest Z2M update ?
If not, can you please provide details like logs ? Thanks

ilgio, 2023-12-11:
Nothing?

ilgio, 2024-03-27:
but this problem How do I solve this problem?

FabianMangold, 2024-04-20:
Hi @ilgio,

It's best to open new issues and refer to this one in the future, this is particular issue was closed long ago and not actively monitored anymore.
Could you please try it once more with the latest DEV version please (1.36.1 at the time of writing).
You may have to launch the update several times in a row after having powercycled the device. This is somewhat unfortunate but generally does the trick.

Thanks a lot