Koenkk/zigbee-OTA #394: Bosch thermostat 2 new Firmware available ?

Dermayer, 2023-11-08:
I have updated 3 Bosch thermostat 2 with the OTA File from the Repo, works fine :-) 
I saw in the Bosch release notes that there was probably an October update?
https://www.bosch-smarthome.com/de/de/service/aktuelle-meldungen/release-infos/system-solutions/2023-10-24.html

Is anyone able to create the corresponding OTA file and upload it to the repro?

mmattel, 2023-11-09:
@DerDreschner fyi, many folks would be happy if you could assist 😄

DerDreschner, 2023-11-10:
@mmattel Thanks for mentioning me.

Yeah, the changelog may suggest that there is a new firmware available. But I checked it right after release and the bugfix only affects the bosch controller software. There is no new firmware available for the thermostat itself. And just fyi: according to [this post in the bosch community forum](https://community.bosch-smarthome.com/t5/funktionen-und-nutzung/hinweis-und-umfrage-zum-heizk%C3%B6rper-thermostat-ii/td-p/42026), there are no changes planed to the heating algorithm of the thermostat.

Dermayer, 2023-11-10:
@DerDreschner:

That's right, it actually looks like there is new firmware. I fell straight for it.
With this note I have now found my error. As soon as the Windows Open flag is set, the thermostat briefly sends a heatset point of 8°C. But if the current heatset point is also set when sending the Windows Open flag, Windows Open will be ignored.
This is probably exactly what has been improved in the home controller or the app.

It is not yet clear to me how the scheduler object has to be filled. Do you have any information for this?

The issue can therefore be closed. There is no new firmware. :-)

DerDreschner, 2023-11-29:
@Dermayer 

Which scheduler object do you mean? As far as I know there is no field for setting a heating schedule inside the thermostat itself. At least not exposed via Z2M at the moment.

And you can close the issue yourself as I'm not able to do it ("comment and close" should be visible to you). We're still able to comment on it afterwards. :)

Dermayer, 2023-12-05:
@DerDreschner: 
I use the Conbee2 with deconz in combination with node red. There is a object in the payload which is for the schedule.

closed 2023-12-05 by Dermayer