Koenkk/zigbee-OTA #400: Philips Hue light strips declining newest OTA

peterjuras, 2023-11-19:
Hi!

There appears to be an issue with the newest available OTA update for two of my Philips Hue light strps (LCL001). 

According to home assistant I have the 16785410 version installed and a 16785412 version is available. When trying to install the update, both light strips start the OTA update but the fail to update with a "No new image available" log message.

Logs:

```
Debug 2023-11-19 07:03:16Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Bedroom right light strip","transaction":"ok52c-1"}'
Info 2023-11-19 07:03:16Updating 'Bedroom right light strip' to latest firmware
Debug 2023-11-19 07:03:16Received Zigbee message from 'Bedroom right light strip', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20230403","swBuildId":"1.104.2"}' from endpoint 11 with groupID 0
Debug 2023-11-19 07:03:16Updating to latest '0x001788010946d214' (LCL001)
Debug 2023-11-19 07:03:16Using endpoint '11'
Debug 2023-11-19 07:03:16Received Zigbee message from 'Bedroom right light strip', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16785412,"imageType":287,"manufacturerCode":4107}' from endpoint 11 with groupID 0
Debug 2023-11-19 07:03:16Got OTA request '{"fieldControl":0,"manufacturerCode":4107,"imageType":287,"fileVersion":16785412}'
Debug 2023-11-19 07:03:16ZigbeeOTA: downloaded main index
Debug 2023-11-19 07:03:16getNewImage for '0x001788010946d214', meta {"fileVersion":16785412,"fileSize":423942,"url":"https://otau.meethue.com/storage/ZGB_100B_011F/38f5ed2b-119a-4d6c-9852-2e6470fecf0e/100B-011F-01002004-ConfLightBLE-ModuLumV3-EFR32MG21.zigbee","sha512":"68b70b7f152b246dc34086982c2961eb5bc2a4f4c30d3276619e8db9458549177b3ea391cccaa825821268c0f4be8f86d3a1dc66d55876aa3119755ea8aabde0"}
Debug 2023-11-19 07:03:16Update of 'Bedroom right light strip' failed (AssertionError [ERR_ASSERTION]: No new image available)
Info 2023-11-19 07:03:16MQTT publish: topic 'zigbee2mqtt/Bedroom right light strip', payload '{"brightness":254,"color":{"x":0.371,"y":0.239},"color_mode":"xy","color_temp":444,"linkquality":200,"power_on_behavior":"on","state":"OFF","update":{"installed_version":16785410,"latest_version":16785412,"state":"available"},"update_available":null}'
Info 2023-11-19 07:03:16MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Bedroom right light strip"},"error":"Update of 'Bedroom right light strip' failed (No new image available)","status":"error","transaction":"ok52c-1"}'
Error 2023-11-19 07:03:16Update of 'Bedroom right light strip' failed (No new image available)
Debug 2023-11-19 07:03:16AssertionError [ERR_ASSERTION]: No new image available at getNewImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:459:11) at runMicrotasks (<anonymous>) at processTicksAndRejections (node:internal/process/task_queues:96:5) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:317:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:268:41)
Debug 2023-11-19 07:03:20Received MQTT message on 'homeassistant/sensor/AS24-9VVK9Y3-satellite/availability' with data 'online'
Debug 2023-11-19 07:03:26Received Zigbee message from 'Bedroom right light strip', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16785412,"imageType":287,"manufacturerCode":4107}' from endpoint 11 with groupID 0
Debug 2023-11-19 07:03:26Device 'Bedroom right light strip' requested OTA
Debug 2023-11-19 07:03:26Check if update available for '0x001788010946d214' (LCL001)
Debug 2023-11-19 07:03:26ZigbeeOTA: downloaded main index
Debug 2023-11-19 07:03:26Is new image available for '0x001788010946d214', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":287,"fileVersion":16785412}', latest meta '{"fileVersion":16785412,"fileSize":423942,"url":"https://otau.meethue.com/storage/ZGB_100B_011F/38f5ed2b-119a-4d6c-9852-2e6470fecf0e/100B-011F-01002004-ConfLightBLE-ModuLumV3-EFR32MG21.zigbee","sha512":"68b70b7f152b246dc34086982c2961eb5bc2a4f4c30d3276619e8db9458549177b3ea391cccaa825821268c0f4be8f86d3a1dc66d55876aa3119755ea8aabde0"}'
Debug 2023-11-19 07:03:26Update available for '0x001788010946d214': NO
```

peterjuras, 2023-11-19:
I just pressed update again and at least the "Update available" message now also dissappeared on Zigbee2MQTT. Is there any information available on what went wrong here?

crocodilus, 2023-11-22:
I am experiencing the same issue and will be following this thread closely.  I tried editing the states file and restarting zigbee2mqtt, but it still persists.

mundschenk-at, 2023-12-01:
I have seen this as well.

posixx, 2023-12-04:
Yes, same here

Koenkk, 2023-12-04:
Could you provide the the debug log of both the check for update and update process?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

posixx, 2023-12-05:
> Could you provide the the debug log of both the check for update and update process?
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

On my installation the buttons keep on "update device firmware" even on multiple tries. So i'm unable to trigger a "Check for new updates" for the failing devices. Hopefully @peterjuras can provide you with the requested logs!

Koenkk, 2023-12-05:
@posixx try via [this](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#manually-check-if-an-update-is-available) MQTT message instead.

posixx, 2023-12-05:
Tnx that was very helpful. After triggering the update check for the lights which were reporting update available the status went back to no update available. For some, i first needed to try the update; that failed with "aborted by device" after that a manual update check through MQTT marked the OTA as no update available. So my problem is solved now!

peterjuras, 2024-06-03:
@posixx  I don't experience this issue anymore.

hardwareadictos, 2024-09-06:
Well i'm certainly experiencing this with my 2 hue ambiance lightstrips. Unable to retrieve OTAs, and ehen it works update times are too big:

[image]

https://www.zigbee2mqtt.io/devices/929002994901.html#philips-929002994901

closed 2025-04-06 by github-actions[bot]