Koenkk/zigbee-OTA #403: Aurora AU-A1ZBDSS - New Firmware OTA help

Stegooding88, 2023-11-24:
Hi All, 

I have a firmware update for Aurora AOne AU-A1ZBDSS Double Zigbee socket. i'd like to share this with others by adding to the OTA update for Z2M. I'm not so familiar with GitHub, would someone be kind enough to do this for me? I've attached the Zip containing OTA files. 
[aurora-aone-zigbee-double-socket-v18-firmware.zip](https://github.com/Koenkk/zigbee-OTA/files/13457759/aurora-aone-zigbee-double-socket-v18-firmware.zip)

details from Z2M:
Device type - Router
Zigbee Model - DoubleSocket50AU
Zigbee Manufacturer - Aurora
Description - Double smart socket UK
Manufacturer
 - [Aurora Lighting](https://www.zigbee2mqtt.io/supported-devices/#v=Aurora%20Lighting)
Model
 - [AU-A1ZBDSS](https://www.zigbee2mqtt.io/devices/AU-A1ZBDSS.html#aurora%2520lighting-au-a1zbdss)

Many Thanks

Blackduke77, 2024-01-10:
Hi, I have these switches and wanted to check and update the firmware, how can we get this into the main repository?

Koenkk, 2024-01-11:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-11 by Koenkk

Blackduke77, 2024-01-12:
hi @Koenkk @Stegooding88  maybe a silly question from me, I am a new user to Zigbee2MQTT. I can see my devices asking for OTA, but logs are returning no image available. Do I need to do anything, or has this not reached the branch that I would see it in?

[image]

Blackduke77, 2024-01-22:
@Koenkk @Stegooding88 I have tested these on the dev branch and successfully updated all my sockets.

Pel1can111, 2024-02-04:
Any specifics on what this firmware added/fixed or any changes?

Stegooding88, 2024-02-06:
The revision notes with this firmware as follows:

  *   Corrects an issue where users may see irregular power monitoring values
  *   General Bug Fixes

Pel1can111, 2024-02-06:
Thanks, I went ahead and updated by sockets and I think this might have fixed the occasional flickering of the LEDs that seemed present on all my sockets so that’s great.

andyblac, 2024-02-20:
one of my sockets lets have gone YELLOW and BLUE, after trying to update, but Z2M responded with failed, now the device is offline, what should I do wait, or is it safe to reboot the device by cutting power ?