Koenkk/zigbee-OTA #413: Legrand Dimmer switch w/o neutral 067771 cannot force update over OTA

wladkolc, 2023-12-15:
Hello, trying to update legrand switch over OTA with force parameter.

Version INFO:
Zigbee2MQTT version
1.34.0-dev commit: 25de4fd
Coordinator type
EZSP v12
Coordinator revision
7.3.2.0 build 212
Coordinator IEEE Address
0xbc026efffe1fd840
Frontend version
0.6.149
Zigbee-herdsman-converters version
16.9.0
Zigbee-herdsman version
0.26.1

Using homeassistant skyconnect dongle.

I have forced update over json file, after successful update seeing new version and idle state.

    "update": {
        "installed_version": 4801027,
        "latest_version": 4801027,
        "state": "idle"
    },

but after zigbee2mqtt restart shows me old version again, and i have chance to update device again..

"update": {
        "installed_version": 54532608,
        "latest_version": 4801027,
        "state": "available"
    }

FabianMangold, 2024-03-30:
Hi @wladkolc,

Is this issue still relevant ?
Meanwhile, the firmware versions changed, and there we some minor fixes implemented to the update process.
Could you please try it again, or close this issue should the issue be fixed ?

Thanks a lot

wladkolc, 2024-03-31:
Hi @FabianMangold, it was fixed by using sonoff stick to upgrade and power off dimmer for few minutes.

closed 2024-03-31 by wladkolc