Koenkk/zigbee-OTA #419: TS011F_plug_1 OTA update stopped working

HuffYk, 2023-12-28:
When updating these devices 9 days ago (19.12.), everything worked ok. Now after receiving new device I wanted to update it and this error is shown (on all devices, not just new one) when checking for firmware update: **"Failed to check if update available for '...' (No image available for imageType '54179')**

DataGhost, 2023-12-28:
Can confirm. Looks like the wrong image was removed (3088f6c9cd1ba439d0e0b69ad1ae8b52d83efdb1) in some kind of revert (of f69a2b1593a573cbd201cad2dc594f6877ac9c5f)? I have previously updated _TZ3000_gvn91tmx (TuYa TS011F_plug_1) and _TZ3000_ss98ec5d (Moes ZK-EU) with no issue but they all throw the same error now.

Koenkk, 2023-12-29:
Added it back!

closed 2023-12-29 by Koenkk

HuffYk, 2023-12-29:
thx for quick fix, works now!