Koenkk/zigbee-OTA #457: OTA update Impossible Inter roller shutter Legrand 067726A

XDECO, 2024-03-14:
Good morning

I am trying to update the FW of my roller shutter switch 067726A unfortunately it is impossible on Home Assistant

I don't know how to force the update

Here is the error I have on the Zigbee2MQTT

Zigbee2MQTT:error 2024-03-14 23:37:54: Update of 'Volet Roulant' failed (OTA: Timeout, device did not request any image blocks)

Can you help me

r0m1CH, 2024-03-23:
Hi @XDECO 
Did you read this issue: https://github.com/Koenkk/zigbee-OTA/issues/328#issuecomment-1994510285
It worked for me.

XDECO, 2024-03-23:
> Hi @XDECO <https://github.com/XDECO>
> Did you read this issue: #328 (comment)
> It worked for me.

Good morning

Yes it works now

THANKS

FabianMangold, 2024-03-30:
Hi @XDECO,

Would you mind closing this issue if the problem is fixed ?
P.S: We just submitted another PR to solve the version reporting in the latest NLL firmware. (Unfortunately, this problem is still open on Legrand's side).

Cheers

closed 2024-10-27 by github-actions[bot]