Koenkk/zigbee-OTA #463: Gledopto GL-C-009P

mrmaximas, 2024-03-25:
@Koenkk FYI After two month of email pinging about https://github.com/Koenkk/zigbee2mqtt/issues/18775 i have recieved new fw for Gledopto GL-C-009P from Gledopto support
https://drive.google.com/file/d/11SyBcznUSGgBQP__Iduzw6m3oKhVsK4P/view?usp=sharing
Please add this to repo. 
Thanks!

Koenkk, 2024-03-25:
done!

closed 2024-03-25 by Koenkk

wurff, 2024-03-26:
Be careful with this OTA. Looks like it killed my GL-C-009P. After update all my zigbee network became unresponsive until I swithed the dimmer off. Also now I'm unable to reset it.

mrmaximas, 2024-03-26:
> Be careful with this OTA. Looks like it killed my GL-C-009P. After update all my zigbee network became unresponsive until I swithed the dimmer off. Also now I'm unable to reset it.

Strange thing, I successfully updated 6 of these devices yesterday, but one of them for some reason is displaying the old version even though the status is new version.
[image]

[image]

wurff, 2024-03-26:
> > Be careful with this OTA. Looks like it killed my GL-C-009P. After update all my zigbee network became unresponsive until I swithed the dimmer off. Also now I'm unable to reset it.
> 
> @wurff Strange thing, I successfully updated 6 of these devices yesterday, but one of them for some reason is displaying the old version even though the status is new version. 

Do your devices have a reset button? Could it be that thay have different hardware revisions?

mrmaximas, 2024-03-26:
> > > Be careful with this OTA. Looks like it killed my GL-C-009P. After update all my zigbee network became unresponsive until I swithed the dimmer off. Also now I'm unable to reset it.
> > 
> > 
> > @wurff Strange thing, I successfully updated 6 of these devices yesterday, but one of them for some reason is displaying the old version even though the status is new version.
> 
> Do your devices have a reset button? Could it be that they have different hardware revisions?

yes, my model is GL-C-009P mini from this lot https://aliexpress.com/item/1005005363670021.html
[image]

mrmaximas, 2024-03-26:
> My is not mini. https://aliexpress.ru/item/1005001430323964.html?spm=a2g2w.orderdetail.0.0.53944aa6q2qBbj&sku_id=12000032916815205 Though also GL-C-009P. Could they have different firmwares?

I'm assuming the difference is in the firmwares. 
However, z2m doesn't differentiate between . What is your database entry?
My GL-C-009P Mini is:
{"id":19,"type":"Router","ieeeAddr":"0xa4c138afbc7d3237","nwkAddr":61520,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genLevelCtrl":{"attributes":{"currentLevel":76}},"genBasic":{"attributes":{"modelId":"GL-C-009P","zclVersion":3,"dateCode":"20240227","swBuildId":"20451203"}},"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":255}}},"binds":[{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711455964464}

Gerben321, 2024-03-26:
I'm updating my GL-C-009P (non-mini) right now. I hope it also fixes the horribly slow dimming (the transition it has built in). Thanks for providing the FW. They never wanted to send me any new version.

mrmaximas, 2024-03-26:
> Thanks I'll pull the plug, hoping it will stop and not continue afterwards. Is there anything we can do to prevent it appearing in Z2M? Sounds dangerous, I can imagine most people will just click update, like I did.

@Gerben321 we need to cancel the commit immediately and look for a solution. @Koenkk ???

mrmaximas, 2024-03-26:
@Gerben321 it is possible to post your database entry for this device here?

wurff, 2024-03-26:
> firmware's

Mine has this entry
`{"id":40,"type":"Router","ieeeAddr":"0xa4c138ffa6ddf4b9","nwkAddr":47689,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"modelId":"GL-C-009P"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":13}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab792d","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0022ab792d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711459875522}
`

Gerben321, 2024-03-26:
> @Gerben321 it is possible to post your database entry for this device here?

Sure, but where do I get that?

mrmaximas, 2024-03-26:
> Maybe i was just unlucky and it is not a common problem?

at first glance, there's a difference,  "epId": 1 from your device and ,"epId":11 from my ;(((

mrmaximas, 2024-03-26:
> > @Gerben321 it is possible to post your database entry for this device here?
> 
> Sure, but where do I get that?

depend of your OS in the z2m folder please find /data/ folder the database.db file

wurff, 2024-03-26:
> > Maybe i was just unlucky and it is not a common problem?
> 
> at first glance, there's a difference, "epId": 1 from your device and ,"epId":11 from my ;(((

This is from unflashed one (I have 2, one is still ok and unflashed). Probably it could change with new firmware.

Gerben321, 2024-03-26:
I have two of them, of which only one gave me a possible update. One older and one newer one.

`{"id":43,"type":"Router","ieeeAddr":"0x84fd27fffe8130e1","nwkAddr":37419,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":2,"dateCode":"20210706","swBuildId":"10476802"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":25}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"dateCode":"20210706","swBuildId":"10476802","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711459822579}`

`{"id":47,"type":"Router","ieeeAddr":"0xa4c138fbf714bd65","nwkAddr":21081,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"modelId":"GL-C-009P","dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":13}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f20e0","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f20e0","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711459774999}`

mrmaximas, 2024-03-26:
> Probably it could change with new firmware.
 you posted is still with the old firmware, maybe it will kill it too.

mrmaximas, 2024-03-26:
> I have two of them, of which only one gave me a possible update. One older and one newer one.
> 
> `{"id":43,"type":"Router","ieeeAddr":"0x84fd27fffe8130e1","nwkAddr":37419,"manufId":0,"manufName":

because "manufId":0

mrmaximas, 2024-03-26:
I feel like I owe $10 to half the universe now ;(((

mainmind83, 2024-03-27:
this update has left all my regulators offline

mrmaximas, 2024-03-27:
> this update has left all my regulators offline
> 
> `Publish 'set' 'state' to 'XXX_led' failed: 'Error: Command 0xa4c1384d88ebdbc8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44949 - 1 - 128 - 6 - 11 after 10000ms)'`
> 
> ![image](https://private-user-images.githubusercontent.com/18678468/317216802-d125101d-142c-4794-bb79-904516500b8b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE1NDg0MzgsIm5iZiI6MTcxMTU0ODEzOCwicGF0aCI6Ii8xODY3ODQ2OC8zMTcyMTY4MDItZDEyNTEwMWQtMTQyYy00Nzk0LWJiNzktOTA0NTE2NTAwYjhiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI3VDE0MDIxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM5MGJhODAxYmY4NWVmY2E1NDFiZTcxYzljNzJiNmFhOTQyM2U1MWE2NDBiZjZkYTFiZGNiZTI0OTk5OTNjZGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Q1M0MZYHLA8WIjBbASTYMeihEom1Ns1dB9ZfRGYwdfA)
> 
> Is it possible to download and recover the firmware of another device that is not updated? https://github.com/Koenkk/zigbee-OTA/tree/master/images/Gledopto
> 
> UPDATE:
> 
> 1. Delete the zigbee2mqtt device, force its deletion
> 2. Reset the device (turn off 5 times waiting 2 seconds)
> 3. Restart zigbee2mqtt
> 4. Pair again

@mainmind83 Thanks for this information. 
Does it follow that the firmware does not turn the regular Gledopto GL-C-009P into a brick? Has your device been updated? I've requested firmware for it too, but haven't heard back yet.

mainmind83, 2024-03-27:
> @mainmind83 Thanks for this information. Does it follow that the firmware does not turn the regular Gledopto GL-C-009P (not Mini) into a brick? Has your device been updated? I've requested firmware for it too, but haven't heard back yet.

I have 2 with 369176577 and 2 with 604057601 firnware.

After updating them, they appeared "offline" but with LQI value, without responding to any of the actions. Only after forcefully deleting them have I been able to recover the previous normal operation...

mrmaximas, 2024-03-27:
> I have 2 with 369176577 and 2 with 604057601 firnware.
> 

@mainmind83 It's mini or regular version of GL-C-009P updated with the 604057601 ??

mainmind83, 2024-03-27:
They are inside a false ceiling and I cannot confirm the model now, but the purchase was for the Zigbee LED Controller W (pro) model.

mrmaximas, 2024-03-27:
> They are inside a false ceiling and I cannot confirm the model now, but the purchase was for the Zigbee LED Controller W (pro) model.

can you post database entry for this device?

mainmind83, 2024-03-27:
What information do you need and how can I obtain it?

[image]

mrmaximas, 2024-03-27:
@mainmind83 depend of you OS in the z2m folder please find /data/ folder then database.db file, then find ieeee addr of this device.

mainmind83, 2024-03-27:
1 / 4:

`{"id":4,"type":"Router","ieeeAddr":"0xa4c1384d88XXXXXX","nwkAddr":44949,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,11,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":236}}},"binds":[],"configuredReportings":[],"meta":{}},"11":{"epId":11,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1711534180809}`

mrmaximas, 2024-03-27:
> [ID] 3, 4 updated
> 
> ```
> {"id":2,"type":"Router","ieeeAddr":"0xa4c138fa5bXXXXXX","nwkAddr":31903,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":183}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711534390617}
> {"id":3,"type":"Router","ieeeAddr":"0xa4c13857e9XXXXXX","nwkAddr":3418,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711534380979}
> {"id":4,"type":"Router","ieeeAddr":"0xa4c1384d88XXXXXX","nwkAddr":44949,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,11,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":236}}},"binds":[],"configuredReportings":[],"meta":{}},"11":{"epId":11,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1711534180809}
> {"id":5,"type":"Router","ieeeAddr":"0xa4c1380b00XXXXXX","nwkAddr":14626,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242,11],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1711534300537}
> ```
> 
> NOTE: Home Assistant OS on /homeassistant/zigbee2mqtt/database.db or download backup of z2m

all off your devices looks like old version > "dateCode":"20221212","swBuildId":"10651203"
new one is "dateCode":"20240227","swBuildId":"20451203

mainmind83, 2024-03-27:
Sorry, I have published the data from a previous backup:

```
{"id":2,"type":"Router","ieeeAddr":"0xa4c138fa5bXXXXXX","nwkAddr":31903,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":183}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711554010682}
{"id":3,"type":"Router","ieeeAddr":"0xa4c13857e9XXXXXX","nwkAddr":3418,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711554001055}
{"id":4,"type":"Router","ieeeAddr":"0xa4c1384d88XXXXXX","nwkAddr":44949,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711553470037}
{"id":5,"type":"Router","ieeeAddr":"0xa4c1380b00XXXXXX","nwkAddr":47603,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711553682787}

```
[ID 3, 4]

`{
    "linkquality": 149,
    "power_on_behavior": "on",
    "state": "OFF",
    "update": {
        "installed_version": 604057601,
        "latest_version": 604057601,
        "state": "idle"
    }
}`

[ID 1,2 ]

`{
    "brightness": 246,
    "linkquality": 65,
    "power_on_behavior": "on",
    "state": "OFF",
    "update": {
        "installed_version": 369176577,
        "latest_version": 604057601,
        "state": "available"
    }
}`

mrmaximas, 2024-03-28:
> Sorry, I have published the data from a previous backup:
> 
> ```
> {"id":2,"type":"Router","ieeeAddr":"0xa4c138fa5bXXXXXX","nwkAddr":31903,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":183}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711554010682}
> {"id":3,"type":"Router","ieeeAddr":"0xa4c13857e9XXXXXX","nwkAddr":3418,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":2,"dateCode":"20221212","swBuildId":"10651203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711554001055}
> {"id":4,"type":"Router","ieeeAddr":"0xa4c1384d88XXXXXX","nwkAddr":44949,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711553470037}
> {"id":5,"type":"Router","ieeeAddr":"0xa4c1380b00XXXXXX","nwkAddr":47603,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-009P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-009P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":0,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20240227","swBuildId":"20451203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1711553682787}
> ```
> 
> [ID 4, 5]
> 
> `{ "linkquality": 149, "power_on_behavior": "on", "state": "OFF", "update": { "installed_version": 604057601, "latest_version": 604057601, "state": "idle" } }`
> 
> [ID 2,3 ]
> 
> `{ "brightness": 246, "linkquality": 65, "power_on_behavior": "on", "state": "OFF", "update": { "installed_version": 369176577, "latest_version": 604057601, "state": "available" } }`

@mainmind83 4,5 looks like my mini, you bought all four at the same time?

mainmind83, 2024-03-28:
Yess, 4 same model...

mrmaximas, 2024-03-28:
> Yess, 4 same model Pro...

this is very interesting, the updated devices have changed endpoints, this is probably why the devices stopped working after the update and were added after a full reset.

mrmaximas, 2024-03-28:
@wurff were you able to reset your device and get it working again?

mainmind83, 2024-03-28:
> > Endpoint "1" to "11"... It seems more like a bug? difference between the pro, mini or normal model?
> 
> I'm assuming that other users have with 1, I have a mini with 11, we have a zoo of similar devices.. it's good that yours worked with the firmware from the mini....

It' possible revert firmware?

wurff, 2024-03-28:
> to reset your device an

Unfortunately no. It won't reset neither with a reset button nor switching on/off several times. The LED satys solid green. Also, if powered on, it will strangely affect the whole network - seems like it does something to coordinator. Sometimes coordinator is even unable to start on z2m initialization, sometimes it starts, but z2m cpu usage is high and everything is soooo slow that it's not able to communicate with other devices.

mrmaximas, 2024-03-28:
> > to reset your device an
> 
> Unfortunately no. It won't reset neither with a reset button nor switching on/off several times. 

the recipe above didn't work for you?

wurff, 2024-03-28:
>the recipe above didn't work for you?

No. I had to force delete it from z2m and use another controller.

mrmaximas, 2024-03-29:
> > > Endpoint "1" to "11"... It seems more like a bug? difference between the pro, mini or normal model?
> > 
> > 
> > I'm assuming that other users have with 1, I have a mini with 11, we have a zoo of similar devices.. it's good that yours worked with the firmware from the mini....
> 
> It' possible revert firmware?

i ask for GL-C-009P (Pro) fw from Gledopto

scotch666, 2024-03-31:
> UPDATE:
> 
>     1. Delete the zigbee2mqtt device, force its deletion
> 
>     2. Reset the device (turn off 5 times waiting 2 seconds)
> 
>     3. Restart zigbee2mqtt
> 
>     4. Pair again

This does not help with my 0009P. I can pair it correctly but cannot read any states or control the light.
Has anybody managed to repair the device?

mrmaximas, 2024-03-31:
@Koenkk please help to disable this update a.s.a.p.!!!

mrmaximas, 2024-04-01:
> > UPDATE:
> > ```
> > 1. Delete the zigbee2mqtt device, force its deletion
> > 
> > 2. Reset the device (turn off 5 times waiting 2 seconds)
> > 
> > 3. Restart zigbee2mqtt
> > 
> > 4. Pair again
> > ```
> 
> This does not help with my 009P. I can pair it correctly but cannot read any states or control the light. Has anybody managed to repair the device?
> 
> PS: everything seems to be fine now. Took a while after pairing. PS2: updated my second Gledopto 009P as well, because lowest dim level seems to be bit darker than with the old firmware.

You have updated the mini version or regular?

scotch666, 2024-04-01:
> You have updated the mini version or regular?

I think it is the regular version

mrmaximas, 2024-04-01:
> > You have updated the mini version or regular?
> 
> I think it is the regular version

it's possible to make it photo?

DisDis, 2024-04-02:
It helps me in my case:
> 
> 1. Delete the zigbee2mqtt device, force its deletion
> 2. Reset the device (turn off 5 times waiting 2 seconds)
> 3. Restart zigbee2mqtt
> 4. Pair again

My device:
GL-C-009P (with reset button)
Firmware creation date
20240227
Firmware version
20451203
[image]

mainmind83, 2024-04-02:
@scotch666 nice ;) https://github.com/Koenkk/zigbee-OTA/issues/463#issuecomment-2022354545

Gledopto support has just indicated the following:

`For this thread, there are many users approaching us about upgrading the wrong firmware and causing damage to the product, usually we don't recommend updating the firmware yourself. Because the product model is different in different period, or even if the product model is the same, the firmware of the product is not common usually due to the internal parts updating, we recommend not to share the firmware or use the firmware shared by others for upgrading, the wrong upgrading may lead to the product is damaged directly, and can't continue to use it!`

I have requested a copy of the firmware (in my case) of my 4 devices, it should not be updated without having control of the different internal hardware versions...

drount, 2024-04-02:
> It helps me in my case:
> 
> > 1. Delete the zigbee2mqtt device, force its deletion
> > 2. Reset the device (turn off 5 times waiting 2 seconds)
> > 3. Restart zigbee2mqtt
> > 4. Pair again
> 
> My device: GL-C-009P (with reset button) Firmware creation date 20240227 Firmware version 20451203 ![PXL_20240402_185622216](https://private-user-images.githubusercontent.com/87201/318929081-6576dc1a-cb23-421b-9566-d268847019d0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIwODkwNjQsIm5iZiI6MTcxMjA4ODc2NCwicGF0aCI6Ii84NzIwMS8zMTg5MjkwODEtNjU3NmRjMWEtY2IyMy00MjFiLTk1NjYtZDI2ODg0NzAxOWQwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAyVDIwMTI0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdmNDhjMWFmMzZhMjU4ODM0ZmJjZjgyM2QzOWY1Mjg1ODNjY2ZmMWFmYzZlYjllNDk3NDkwZjdmZDc4ODU2NWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wba6_4QjlGrCAgrvyjjnAbQdj7_EQJ0CiF9HWoasoZ0)

Any clue on how to revert the FW or update to the 20240227?

I updated to the faulty version and I have 20221212 -  10651203 (same as before)

DisDis, 2024-04-02:
> > It helps me in my case:
> > > 1. Delete the zigbee2mqtt device, force its deletion
> > > 2. Reset the device (turn off 5 times waiting 2 seconds)
> > > 3. Restart zigbee2mqtt
> > > 4. Pair again
> > 
> > 
> > My device: GL-C-009P (with reset button) Firmware creation date 20240227 Firmware version 20451203 !
> 
> Any clue on how to revert the FW or update to the 20240227?
> 
> I updated to the faulty version and I have 20221212 - 10651203 (same as before)

After I updated to the faulty version, I found this issue and used this steps(1..4). Before it my GL-C-009P didn't work. I tried to reset and join, it didn't help. It only works if you do it step by step as described.

drount, 2024-04-02:
> After I updated to the faulty version, I found this issue and used this steps(1..4). Before it my GL-C-009P didn't work. I tried to reset and join, it didn't help. It only works if you do it step by [...]

Confirming it works. I missed the "restart zigbee2mqqtt" step. I had to do it once for each controller. 

Disconnect all the faulty controllers and add one by one.

The version updated from 20221212-> 20240227.

Does anybody have any insight on why this produce works and how it updated the fw version?

sevrugin, 2025-10-22:
> [@Koenkk](https://github.com/Koenkk) FYI After two month of email pinging about [Koenkk/zigbee2mqtt#18775](https://github.com/Koenkk/zigbee2mqtt/issues/18775) i have recieved new fw for Gledopto GL-C-009P Mini from Gledopto support https://drive.google.com/file/d/11SyBcznUSGgBQP__Iduzw6m3oKhVsK4P/view?usp=sharing Please add this to repo. Thanks!

Can you please share the firmware one more time? I need it as well