Koenkk/zigbee-OTA #466: Legrand device 412015 impossible update

r0m1CH, 2024-03-30:
Hi,

I tried to update my last legrand device without any success (412015).
I know that update can be tricky with Legrand.
I used this solution to successfully update my others devices : https://github.com/Koenkk/zigbee-OTA/issues/328#issuecomment-1991903289

I followed the procedure. 

- Shutting down the device for few minutes
- Restarting Z2M
- Start the update in the first two minutes.

The upgrade process seems to go well under 20 minutes like my others devices but always ends with an error:
[image]

I tried the latest stable version with the common.ts file modified for legrand and the latest available dev version.
Both version ends with the same results.

Any idea on how I can update my last device ?

Thanks

FabianMangold, 2024-03-30:
Hi @r0m1CH,

Could you please tell us what kind of device it is (or which firmware is used ?)
If possible, lets stick to the DEV version for now.

Thanks

r0m1CH, 2024-03-30:
Hi @FabianMangold 

For sure. I went back to the latest dev version.

here the device info:
[image]

FabianMangold, 2024-03-30:
Thanks for that
Can you figure out which firmware the device tries to install ?
Just switch the logs to DEBUG, you should get the reference there. I wonder if you have to install an intermediate version first.

r0m1CH, 2024-03-30:
Is this what you need ? 
`"installed_version":3883523,"latest_version":5456895,"progress":0,"state":"updating"},"update_available":false}'`

r0m1CH, 2024-03-30:
here more info maybe @FabianMangold 

`30/03/2024
10:57:21
Zigbee2MQTT:debug 2024-03-30 10:57:21: OTA: Updating to latest '0x000474000123d607' ( DIN power consumption module)
30/03/2024
10:57:21
Zigbee2MQTT:debug 2024-03-30 10:57:21: OTA: Using endpoint '1'
30/03/2024
10:57:21
Zigbee2MQTT:debug 2024-03-30 10:57:21: Received Zigbee message from 'consommation_plaque_cuisson', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":3883523,"imageType":17,"manufacturerCode":4129}' from endpoint 1 with groupID 0
30/03/2024
10:57:21
Zigbee2MQTT:info  2024-03-30 10:57:21: MQTT publish: topic 'zigbee2mqtt/consommation_plaque_cuisson', payload '{"last_seen":"2024-03-30T10:57:21+01:00","linkquality":123,"power":1,"power_alarm":null,"power_alarm_active":false,"power_alarm_active_value":0,"power_alarm_enabled":0,"power_alarm_wh_threshold":0,"power_apparent":2,"update":{"installed_version":3883523,"latest_version":5456895,"state":"available"},"update_available":true}'
30/03/2024
10:57:21
Zigbee2MQTT:debug 2024-03-30 10:57:21: OTA: Got request '{"fieldControl":0,"manufacturerCode":4129,"imageType":17,"fileVersion":3883523}'
30/03/2024
10:57:21
Zigbee2MQTT:debug 2024-03-30 10:57:21: ZigbeeOTA: Getting image metadata for ' DIN power consumption module'
30/03/2024
10:57:21
Zigbee2MQTT:debug 2024-03-30 10:57:21: ZigbeeOTA: Downloaded main index
30/03/2024
10:57:21
Zigbee2MQTT:debug 2024-03-30 10:57:21: OTA: Getting new image for '0x000474000123d607' ( DIN power consumption module), latest meta {"fileVersion":5456895,"fileSize":250743,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-005343FF-NLP.zigbee","sha512":"07d6b136c6012315efbfff1de9fb6b70c0b187463b80e6ec34005b38695682da073aa1ae20ac9d6c2569bf2b562492bc18bf672c1de78bff2bd1098dd1c186fb"}
30/03/2024
10:57:21
Zigbee2MQTT:debug 2024-03-30 10:57:21: OTA: Downloading firmware image from 'https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-005343FF-NLP.zigbee' using the zigbeeOTA custom CA certificates
30/03/2024
10:57:22
Zigbee2MQTT:debug 2024-03-30 10:57:22: OTA: Update checksum validation succeeded for '0x000474000123d607' ( DIN power consumption module)
30/03/2024
10:57:22
Zigbee2MQTT:debug 2024-03-30 10:57:22: OTA: Get new image for '0x000474000123d607' ( DIN power consumption module), image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4129,"imageType":17,"fileVersion":5456895,"zigbeeStackVersion":2,"otaHeaderString":"                                ","totalImageSize":250743}
30/03/2024
10:57:22
Zigbee2MQTT:debug 2024-03-30 10:57:22: OTA: Got new image for '0x000474000123d607' ( DIN power consumption module)
30/03/2024
10:57:22
Zigbee2MQTT:debug 2024-03-30 10:57:22: OTA: Starting upgrade
30/03/2024
10:57:22
Zigbee2MQTT:debug 2024-03-30 10:57:22: Received Zigbee message from 'consommation_plaque_cuisson', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":3883523,"imageType":17,"manufacturerCode":4129}' from endpoint 1 with groupID 0
30/03/2024
10:57:22
Zigbee2MQTT:info  2024-03-30 10:57:22: MQTT publish: topic 'zigbee2mqtt/consommation_plaque_cuisson', payload '{"last_seen":"2024-03-30T10:57:22+01:00","linkquality":116,"power":1,"power_alarm":null,"power_alarm_active":false,"power_alarm_active_value":0,"power_alarm_enabled":0,"power_alarm_wh_threshold":0,"power_apparent":2,"update":{"installed_version":3883523,"latest_version":5456895,"state":"available"},"update_available":true}'
30/03/2024
10:57:22
Zigbee2MQTT:info  2024-03-30 10:57:22: MQTT publish: topic 'zigbee2mqtt/consommation_plaque_cuisson', payload '{"last_seen":"2024-03-30T10:57:22+01:00","linkquality":116,"power":1,"power_alarm":null,"power_alarm_active":false,"power_alarm_active_value":0,"power_alarm_enabled":0,"power_alarm_wh_threshold":0,"power_apparent":2,"update":{"installed_version":3883523,"latest_version":5456895,"state":"available"},"update_available":true}'
30/03/2024
10:57:22
Zigbee2MQTT:debug 2024-03-30 10:57:22: OTA: Request offsets: fileOffset=0 pageOffset=0 dataSize=62
30/03/2024
10:57:22
Zigbee2MQTT:debug 2024-03-30 10:57:22: OTA: Payload offsets: start=0 end=62 dataSize=62
30/03/2024
10:57:22
Zigbee2MQTT:info  2024-03-30 10:57:22: Update of 'consommation_plaque_cuisson' at 0.00%
30/03/2024
10:57:22
Zigbee2MQTT:info  2024-03-30 10:57:22: MQTT publish: topic 'zigbee2mqtt/consommation_plaque_cuisson', payload '{"last_seen":"2024-03-30T10:57:22+01:00","linkquality":116,"power":1,"power_alarm":null,"power_alarm_active":false,"power_alarm_active_value":0,"power_alarm_enabled":0,"power_alarm_wh_threshold":0,"power_apparent":2,"update":{"installed_version":3883523,"latest_version":5456895,"progress":0,"state":"updating"},"update_available":false}'`

FabianMangold, 2024-03-30:
That's perfect, thanks. Let me check on the Legrand site real quick, if an intermediate firmware is required.
I have a similar device (a connected socket using the same firmware), which worked just fine though.

Just out of curiosity, have you restarted the update ? The device seems to be in update mode, with progress 0. Is it "stuck" there or attempting another update ?

r0m1CH, 2024-03-30:
Yes. I just restart it one more time.
on a misunderstanding, you never know :)

FabianMangold, 2024-03-30:
According to Legrand, there is no need for an intermediate firmware, so this should work just fine.
Please give it another try, potentially powercycling the device, and restart Z2m.
If that doesn't help, will give it a try with an "older" firmware first.

r0m1CH, 2024-03-30:
I just finished my other tried.
It ends with this error:
`Debug 2024-03-30 11:25:01Update of 'consommation_plaque_cuisson' failed (Error: OTA: Timeout, device did not request any image blocks)
Info 2024-03-30 11:25:01MQTT publish: topic 'zigbee2mqtt/consommation_plaque_cuisson', payload '{"last_seen":"2024-03-30T11:19:01+01:00","linkquality":116,"power":1,"power_alarm":null,"power_alarm_active":false,"power_alarm_active_value":0,"power_alarm_enabled":0,"power_alarm_wh_threshold":0,"power_apparent":2,"update":{"installed_version":3883523,"latest_version":5456895,"state":"available"},"update_available":true}'
Info 2024-03-30 11:25:01MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'consommation_plaque_cuisson' failed (OTA: Timeout, device did not request any image blocks)","meta":{"device":"consommation_plaque_cuisson","status":"update_failed"},"type":"ota_update"}'
Info 2024-03-30 11:25:01MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"consommation_plaque_cuisson"},"error":"Update of 'consommation_plaque_cuisson' failed (OTA: Timeout, device did not request any image blocks)","status":"error","transaction":"2b6nt-1"}'
Error 2024-03-30 11:25:01Update of 'consommation_plaque_cuisson' failed (OTA: Timeout, device did not request any image blocks)`

It was stucked for few minutes on :
[image]
The rest of the update is working well. The time decrease correctly.

Where can I download old firmware ? Directly on legrand dev website ? 
Then I need to specify the path file in a custom ota file right?

FabianMangold, 2024-03-30:
Yes, you will need to download the file to your device / vm and use a custom OTA file, as shown [here](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files).

Regarding the firmware, can you clone the git OTA project locally like so:
`git clone https://github.com/Koenkk/zigbee-OTA.git`
Then switch to an old commit, to retrieve a fairly old firmware. The following should work:
`git checkout c1494e83a1614981c7909f23430f372798763dba`
You will then be able to retrieve the old NLP firmware file (_1021-0011-4a4203-NLP-74.zigbee)_ from the images/Legrand directory.

Let me know if that works for you.

r0m1CH, 2024-03-30:
Hmm you will not believe it....
It went successfully with my latest attempt.
when it came to 98.81% it was stuck. It threw me the OTA error timeout and restarted automatically by itself.
This time it went through and my device is correctly updated to the latest version available.
Firmware: 053

With legrand device update... you never know. 

Thanks for the help.

closed 2024-03-30 by r0m1CH