Koenkk/zigbee-OTA #483: GL-C-006P(normal) received GL-C-006P(mini) OTA from Z2M

marce14-git, 2024-04-23:
GL-C-006P (normal version) received the OTA for the MINI Version an is now unusable - all commands are received but light does not turn on etc.

GL-C-006P (normal) has now the GL-C-006P (mini) Firmware = Build Date: 20240227 Version: 20451203.

Manual OTA Update with the right firmware: `"url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Gledopto/GL-C-006P_10576802_20240412.ota"` just says no firmware update available

see also : [https://github.com/Koenkk/zigbee2mqtt/issues/22287](url)
[
[image]
](url)
[
[image]
](url)

mariustt, 2024-04-24:
I have exactly the same problem. Additionally, I received the latest OTA version of GL-C-006P (not the mini) from the supplier - https://drive.google.com/file/d/1hA7h_qHQbrTFgNz0aJVS8HNuxzao8MZa

marce14-git, 2024-04-27:
the root of the problem is that GL-C-006P and GL-C-006P(mini) are not distinguished as seperate entities in Z2M and a new FW for either one of the devices shows up for both.

marce14-git, 2024-04-27:
> I have exactly the same problem. Additionally, I received the latest OTA version of GL-C-006P (not the mini) from the supplier - [GL-C-006P_V20451203_KEY_20240227.zip](https://github.com/Koenkk/zigbee-OTA/files/15125580/GL-C-006P_V20451203_KEY_20240227.zip)

your firmware did not work because it was the same Version Number than the one the device had - i was able to resolve my problem by emailing Gledopto and getting a new FW see #22287

mariustt, 2024-04-28:
Thank you for the information, @marce14-git !
I have tried your OTA, the update was successful, but the lights still do not turn on/off. Didn't you do an additional factory reset?

I just want to point out that your OTA is not the latest. I have 9 led drivers. 6 of them I bought at the end of 2022, 3 at the end of 2023. Buy the 2022 version - 10651203, and buy the 2023 version - 20351203.

[image]

marce14-git, 2024-04-28:
> Thank you for the information, @marce14-git ! I have tried your OTA, the update was successful, but the lights still do not turn on/off. Didn't you do an additional factory reset?
>
> I just want to [...]

No i just did the steps as mentioned in the other Post and everything worked on the spot. Sorry to hear that it didnt for you

mariustt, 2024-06-19:
@marce14-git the problem was solved with your OTA - thank you! I updated my PR by changing my OTA to yours (https://github.com/Koenkk/zigbee-OTA/pull/486).

mariustt, 2024-06-20:
@Koenkk I think we can mark this one as closed because it is resolved with this PR - https://github.com/Koenkk/zigbee-OTA/pull/486

closed 2024-06-20 by Koenkk