Koenkk/zigbee-OTA #527: Bticino/Legrand NLL devices reporting onboard firmware being newer than latest version

nomc2, 2024-07-15:
My Bticino K4003C which are Legrand NLL devices are reporting onboard firmware being newer than the latest firmware online.
Below an example update request:
```
[2024-07-15 22:24:43] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Luce Cameretta","transaction":"seuxf-99"}'
[2024-07-15 22:24:43] debug: 	zhc:ota:common: Checking if an update is available for '0x0004740000b23680' ( Light switch with neutral)
[2024-07-15 22:24:43] debug: 	zhc:ota:common: Using endpoint '1'
[2024-07-15 22:24:43] debug: 	zh:controller:endpoint: CommandResponse 0x0004740000b23680/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false,"sendPolicy":"immediate"})
[2024-07-15 22:24:43] debug: 	z2m: Received Zigbee message from 'Luce Cameretta', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":4015615,"imageType":15,"manufacturerCode":4129}' from endpoint 1 with groupID 0
[2024-07-15 22:24:43] debug: 	zhc:ota:common: Got request '{"fieldControl":0,"manufacturerCode":4129,"imageType":15,"fileVersion":4015615}'
[2024-07-15 22:24:43] debug: 	zhc:ota:common: Is new image available for '0x0004740000b23680' ( Light switch with neutral), current '{"fieldControl":0,"manufacturerCode":4129,"imageType":15,"fileVersion":4015615}'
[2024-07-15 22:24:43] debug: 	zhc:ota: Getting image metadata for ' Light switch with neutral'
[2024-07-15 22:24:43] debug: 	zhc:ota: Downloaded main index
[2024-07-15 22:24:43] debug: 	zhc:ota:common: Is new image available for '0x0004740000b23680' ( Light switch with neutral), latest meta '{"fileVersion":2245123,"fileSize":251495,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-000f-00224203-NLL.zigbee","sha512":"853abd58cd50dcfe84ffc123c2f2066e834de31c87b7493d343368f1a945585910164ce789e5dbbf3280c5c398f65898879a67000dfc259f64be58467111676f"}'
[2024-07-15 22:24:43] debug: 	zhc:ota:common: Update available for '0x0004740000b23680' ( Light switch with neutral): NO
[2024-07-15 22:24:43] warning: 	zhc:ota:common: Firmware on '0x0004740000b23680' ( Light switch with neutral) is newer than latest firmware online.
[2024-07-15 22:24:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Luce Cameretta","updateAvailable":false},"status":"ok","transaction":"seuxf-99"}'
```

The latest firmware online appears to be version 2245123 while the onboard version is 4015615
This started since [commit](https://github.com/Koenkk/zigbee-OTA/commit/602d53a3905bab16f48a611cd470360cf8a19de2#diff-7aebb122a6ea8a2749d60cb05b7e103c9eae6e2e85e48d2d6cd9e20b63013975L2734) which appears to have reverted back the NLL version reporting fix in [commit 4d56747](https://github.com/Koenkk/zigbee-OTA/commit/4d56747d4565df12fbcac4d2ac32a1e239a5d8b4)

Additionally, [commit 30015ca](https://github.com/Koenkk/zigbee-OTA/commit/30015cade2eef136ebbe63bb4a46cd175cf1aa65#diff-7aebb122a6ea8a2749d60cb05b7e103c9eae6e2e85e48d2d6cd9e20b63013975) appears to be using an older firmware version (dated 11/2023) which according to the [Legrand site](https://developer.legrand.com/production-firmware-download/) appears to be an older intermediate version (version 34).
Can you guys please check and confirm if this is an issue or if something is wrong with my setup?

nomc2, 2024-07-15:
Looks like someone beat me to it, issue #526 appears to be the same

Koenkk, 2024-07-16:
From what I understand from https://github.com/Koenkk/zigbee-OTA/commit/4d56747d4565df12fbcac4d2ac32a1e239a5d8b4#commitcomment-140948316 this is expected, @FabianMangold could you confirm?

FabianMangold, 2024-07-16:
Hi @Koenkk,

Yes indeed. Unfortunately, this has been a recurrent issue with latest Legrand firmware versions
Several users have reported this to Legrand themselves via their French forum in the past, me included, and they have acknowledged and promised to have this fixed. In vain it seems !
Two options:
- Reverse engineer the firmware, which is a grey area and would like to avoid
- Patch the binary's OTA header, to match the reported firwamre version.

So far, i had provided patched binaries to workaround that issue.
Since i was out of town until recently, i haven't had a chance to update any of my own devices and provide patched versions.
I can take care of this this weekend it that helps, and in case we want to continue to carry on that way, until Legrand fixes that for good. I'll ping Legrand on their forum once more too, but i don't have high hopes for any fixes anytime soon on that end.
Cheers

FabianMangold, 2024-07-18:
Hi @nomc2,

This issue should be solved by now.
Can you please confirm this and close this issue in case it really is ?

Thanks a lot

Crystian182, 2024-07-18:
Hi @FabianMangold,
I can confirm I was able to update my Bticino K4027C to the latest version. Really thanks!

[image]

nomc2, 2024-07-18:
Yes, the issue appears to be resolved now.
Thanks

closed 2024-07-18 by nomc2