Koenkk/zigbee-OTA #547: Please add fresh TS011F_plug_1 OTA FW 1.1.0

mrmaximas, 2024-09-02:
Catched FW 1.1.0 for  TS011F_plug_1 (_TZ3000_w0qqde0g in my case) via SL app
Works on _TZ3000_dksbtrzs too

{"id":100,"type":"Router","ieeeAddr":"0xa4c138035941e495","nwkAddr":19075,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"genBasic":{"attributes":{"65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_w0qqde0g","zclVersion":3,"appVersion":80,"modelId":"TS011F","powerSource":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":235,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1725231324735}

Works on _TZ3000_dksbtrzs too but not offered via SL app for this model

{"id":89,"type":"Router","ieeeAddr":"0xa4c1380b31fbf03c","nwkAddr":761,"manufId":4417,"manufName":"_TZ3000_dksbtrzs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��g.\u0017��g.\u0013��g.\u0012\r\u0000\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_dksbtrzs","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"moesStartUpOnOff":2,"tuyaBacklightMode":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,3],"divisor":100,"multiplier":1}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":232,"rmsCurrent":23,"activePower":0,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":2619665,"53251":123,"53252":11944,"53253":1374,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1725266493051}

MITM catch

{
    "result": [
        {
            "autoSwitch": false,
            "canUpgrade": true,
            "controlType": 0,
            "currentVersion": "1.0.13",
            "desc": "FW upgrade",
            "devType": 0,
            "diffOta": false,
            "fileSize": "310402",
            "firmwareDeployTime": 1724827082,
            "firmwareKey": "keyrh3sh",
            "lastUpgradeTime": 0,
            "md5": "cfce46a24d4eeaf12d39e5007c0a5925",
            "timeout": 6000,
            "type": 3,
            "typeDesc": "ZigBee Module",
            "upgradeStatus": 1,
            "upgradeType": 0,
            "upgradingDesc": "Please keep the power of the device connected during the upgrade process, please be patient.",
            "url": "https://euimagesd2h2yqnfpu4gl5.cdn5th.com/smart/firmware/upgrade/20240613/1718263020-oem_zg_tl8258_plug_OTA_1.1.0.bin",
            "version": "1.1.0"
        }
    ],
    "status": "ok",
    "success": true,
    "t": 1725229115753
}

octa22, 2024-09-02:
@mrmaximas added just for the devices you mentioned (_TZ3000_w0qqde0g, _TZ3000_dksbtrzs), not sure if it is working for other Telink plugs too

mrmaximas, 2024-09-02:
> @mrmaximas added just for the devices you mentioned (_TZ3000_w0qqde0g, _TZ3000_dksbtrzs), not sure if it is working for other Telink plugs too

Thks!

RezzZ, 2024-09-16:
will this firmware prevent the randomly turnoffs?

merlinschumacher, 2024-10-05:
This firmware update also works for _TZ3000_hdopuwv6 plugs.

el-bakkali, 2024-10-30:
_TZ3000_cphmq0q7  is missing from index.json
 
 please can you add
 
 [image]
[image]

MnM001, 2024-11-03:
Please add 

_TZ3000_cehuw1lw

too.

Thanks

mrmaximas, 2025-03-13:
@Nerivec hi! FYI about TUYA fw numbering. If Read result of 'genBasic': {"appVersion":72} then TUYA FW version is 1.10 (110 in base 8 numeral system), if "appVersion": 68 (104 octal) then 1.04 etc.

closed 2025-04-02 by mrmaximas