Koenkk/zigbee-OTA #561: Legrand 067773/067774 didn't respond to OTA request

glpotatoe, 2024-09-22:
Hi !

I have several Legrand devices, and the OTA update for the ones with batteries doesn't work.
I have this error : z2m: Update of 'Commande double sans fil séjour' failed (Device didn't respond to OTA request)
I tried to keep the device alive by clicking on a button every second, I had the same error.

Here is a picture of the concerned devices :
[image]

I'm on z2m 1.40.1, and I have a SLZB-06M coordinator with ember adapter.

Update on wired legrand devices works well.

glpotatoe, 2025-03-08:
Hi,
I still have the error since the issue was declared. I'm now in version 2.1.3 for z2m.

daemon707, 2025-05-24:
I think I can help you with legrand wireless switches. Hold both On and Off buttons until you see violet or yellow light. The light is On for only few secs and during this time you need to press "Check for new updates" button. When the update is found repeat this procedure and click "Update device firmware" button. Make sure your switch isn't far away from coordinator or router and the battery is good. If everything is ok, the update will be installed in appx 30min.

glpotatoe, 2025-08-27:
Thanks for the advice, i tried several times on différent switches, this tip don't work for me :-/