Koenkk/zigbee-OTA #571: How do I use these files？

Aogu181, 2024-10-18:
How do I use these files？How can I use this firmware to upgrade my device?

Koenkk, 2024-10-18:
Via Zigbee2MQTT

Aogu181, 2024-10-19:
@Koenkk I used Z2M, and I followed the instructions on the website. the website is [https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files](url)
But I can't upgrade my equipment. My error log is as follows:
_**z2m: Update of 'GL-B' failed (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)**_

My steps are as follows on the website below:
[https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/](url)
**What should I do?**

Koenkk, 2024-10-19:
There is no need to setup a local OTA index, remove it and try again. If that doesn't work, provide the debug log of this.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Aogu181, 2024-10-21:
@Koenkk Doesn 't the local firmware need to set an OTA index? So how do I specify the device to upgrade？
The OTA in my Z2M interface only has the check update button, but there is no update device firmware button.
(So I tried to send the subject line: zigbee2mqtt/bridge/request/device/ota_update/update, payload is my device id, but it didn't work.)
**log:** 
`info 2024-10-21 09:40:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"GL-B-007P"},"error":"Failed to check if update available for 'GL-B-007P' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)","status":"error","transaction":"rvcch-1"}'

error 2024-10-21 09:40:52z2m: Failed to check if update available for 'GL-B-007P' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)`
**In the Z2M the log picture is as follows:**
[image]

Koenkk, 2024-10-21:
> The OTA in my Z2M interface only has the check update button, but there is no update device firmware button.

If you check for update and an update is available, it will change to "update device firmware":

[image]

> error 2024-10-21 09:40:52z2m: Failed to check if update available for 'GL-B-007P' (Cannot destructure property 'data' of '(intermediate value)' as it is undefined.)`

Is this with or without `zigbee_ota_override_index_location` in your `configuration.yaml`?

Aogu181, 2024-10-22:
without zigbee_ota_override_index_location.
Where should my firmware be placed?
Will "check update" changed to "update device firmware" I is sent via · MQTT zigbee2mqtt/bridge/request/device/ota update/update, if directly click the update will not change.

Aogu181, 2024-10-22:
@Koenkk
Without zigbee_ota_override_index_location.
Where should my firmware be placed?
Will "check update" changed to "update device firmware" I is sent via · MQTT zigbee2mqtt/bridge/request/device/ota update/update, if directly click the update will not change.

Aogu181, 2024-10-22:
[image]

Koenkk, 2024-10-22:
> Without zigbee_ota_override_index_location. Where should my firmware be placed?

You don't have to download it, it is taken directly from this repo.

Could you provide the debug log of this error?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Aogu181, 2024-10-23:
**Where is the repo?**
Where should my firmware be placed?
**log:**
[image]

Koenkk, 2024-10-23:
> Where should my firmware be placed?

Nowhere, it's automatically taken from https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

I don't see the "Cannot destructure" error in this log, please provide the log file with this in it.

Aogu181, 2024-10-24:
> > Where should my firmware be placed?
> 
> Nowhere, it's automatically taken from https://github.com/Koenkk/zigbee-OTA/blob/master/index.json
> 
> I don't see the "Cannot destructure" error in this log, please provide the log file with this in it.

[image]

Aogu181, 2024-10-24:
You mean if I need to update the firmware, must from https://github.com/Koenkk/zigbee-OTA/blob/master/index.json this website is the firmware update? If the website is not changed, do I need to upload it myself?
But my firmware is on my local, not on the site, what should I do?

Koenkk, 2024-10-24:
If the firmware is not in this repo, you indeed need to use the local index.

Aogu181, 2024-10-25:
> If the firmware is not in this repo, you indeed need to use the local index.

**1.** I wrote the following to **configural.yaml**:
`ota:`
 `    zigbee_ota_override_index_location: my_index.json`
**2**,In **my_index.json**, I wrote the following:
`[
    {
        "url": "GL-B-007P_V20451233_20240425.ota",
        "force": true
    }
]
`
**3.** And I uploaded the firmware to the same directory as my index.json
**4.** Restart Z2M
**5.** I clicked the update button and then the following error appeared
[image]
**So how should I operate?**

Koenkk, 2024-10-25:
I tries to access this OTA first, is `https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json` accessible from your machine?

Aogu181, 2024-10-28:
> I tries to access this OTA first, is `https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json` accessible from your machine?

yes

Aogu181, 2024-10-28:
> I tries to access this OTA first, is `https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json` accessible from your machine?

Yes,the local OTA index is not working!

Nerivec, 2024-10-28:
As Koenkk indicated, the error (in every screenshot) seems to happen **before** it reads the local OTA index. It works as follows:
1. read `https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json`
2. if local index is defined, read it (url or local file), and concatenate it with step 1

Since it fails at step 1, something must be going wrong with accessing that URL.

_PS: I've added this error as 'not very descriptive' in my TODO list for the OTA refactoring that's coming up._

Aogu181, 2024-10-28:
> As Koenkk indicated, the error (in every screenshot) seems to happen **before** it reads the local OTA index. It works as follows:正如 Koenkk 所指出的，错误（在每个屏幕截图中）似乎发生在读取本地 OTA 索引**之前**。它的工作原理如下：
> 
> 1. read `https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json` 读 `https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json`
> 2. if local index is defined, read it (url or local file), and concatenate it with step 1如果定义了本地索引，则读取它（URL 或本地文件），并将其与步骤 1 连接
> 
> Since it fails at step 1, something must be going wrong with accessing that URL.由于它在第 1 步失败，因此访问该 URL 时一定出了问题。
> 
> _PS: I've added this error as 'not very descriptive' in my TODO list for the OTA refactoring that's coming up.PS：我已将此错误添加到我的 TODO 列表中，用于即将到来的 OTA 重构。_

So how do I solve this? My computer can access **https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json**
I use the virtual machine installed Home Assistant.How do I know if I can access the URL?