Koenkk/zigbee-OTA #594: Possibly new Firmware for Bosch BTH-RA (Radiator thermostat II), 2024.05.14

mmattel, 2024-11-05:
@DerDreschner hope you are doing fine 😄 

I have seen in the [Bosch release notes](https://www.bosch-smarthome.com/at/de/service/aktuelle-meldungen/release-infos/system-solutions/2024-05-14.html) from `2024.05.14` that they have a new FW for the  BTH-RA (Radiator thermostat II).

`Für Raumthermostat II und Raumthermostat II 230 V ist eine neue Firmware verfügbar, die verschiedene kleine Bugfixes enthält.`

Would it be possible that you do your OTA magic like you did in https://github.com/Koenkk/zigbee-OTA/pull/412?

DerDreschner, 2024-11-10:
Hey @mmattel , the firmware upgrade you mention isn't for the `BTH-RA`, it's for the `BTH-RM`. These firmware updates were already added to the repository by @burmistrzak ( #508 ) and @Juan-ma ( #577 ). I keep my eyes open for any firmware upgrades for the BTH-RA as I'm also looking for any improvements in the heating algorithm.

DerDreschner, 2024-11-17:
@mmattel : You mind closing this issue? Or are there any open questions? 😄 I have no rights to do that.

closed 2024-11-17 by mmattel

DerDreschner, 2024-11-21:
@mmattel : I've sniffed the new firmware released yesterday. #607 will publish it. :smile: