Koenkk/zigbee-OTA #612: The issue with Lumi upgrades after updating zigbee-OTA.

z-x-f, 2024-11-25:
Currently, in index.json, there are two OTA files for the same hardware. How are they handled? After I added a new OTA file, it cannot detect the upgrade.

Koenkk, 2024-11-27:
It will pick the first matching one, but I think it would be good to remove the duplicate.

z-x-f, 2024-11-27:
After a zigbee-OTA update, is it expected that the new firmware added will automatically move the old firmware to the images1 folder?

Koenkk, 2024-11-27:
Yes! Those are used for downgrades

z-x-f, 2024-11-28:
Ah, what did I do wrong when I submitted the PR? Was it because I didn't attach the `modelID`? I saw that the old firmware was not automatically moved to `images1` and the `index.json` did not delete the old OTA upgrade information.

closed 2024-11-30 by z-x-f