Koenkk/zigbee-OTA #621: Discuss firmware contribution procedures and improvements to firmware usage procedures

jehos, 2024-12-03:
# Overview

As discussed in #620, there is a possibility that the wrong firmware will be accidentally flash to the wrong device.

0. **Humans must make mistakes because they are incomplete.**
1. There are many cheap devices with the same identifier in the real world (and very much sold around world by aliexpress)
2. The custom-firmware developer did not validate the device during firmware flashing.
3. When the contributor submit the firmware with `index.json`, they did not provide enough detailed metadata. So, Z2m raise an update notification on the wrong device.

The problem occurred because all the mistakes that occurred at each stage passed through the sieve.
need to improvement the sieve to prevent the problem from passing through.

jehos, 2024-12-03:
by @KipK at https://github.com/Koenkk/zigbee-OTA/issues/620#issuecomment-2511603574
> Installing custom fw should only be made manually. 

Totally agree. I had no idea if the firmware I was updating was custom firmware.
because, The information displayed was very simple...

[image]
(sorry, i know this UI from HA, not Z2m)

To display additional information here, contributor will need at least some additional information, such as `ModelId`, `Manufacturer{ID,Name}` and `Changelog`. This information has been optional so far, but should either change it to mandatory or modify it to show it on the UI to encourage the user to submit it. (Everyone doesn't want to see the empty blanks, right? ;-)

These information can also be used by `zigbee-OTA` to match and validate firmware and devices, but can also be used as a basis for users to determine whether updates are being performed. (At least user know if the firmware they're trying to update is official, custom, or who signed the firmware.) And the contributors who submit firmware themselves will also help them understand clearly which devices this firmware is using only for.

It's already functional enough on the Z2m, but people might not use it.
Or, there might be engineering and practical reasons for not being able to easily change the method.
Please understand my lack of understanding.

Koenkk, 2024-12-03:
> need additional metadata when submitting firmware. (ModelId, Manufacturer{ID,Name}, Changelog)

I think we should do this for the overloaded manufacturer codes (like the Tuya ones), for e.g. Philips it doesn't make any sense, the same OTA is also used by many devices which makes it very hard to do and maintain.

closed 2025-07-03 by github-actions[bot]