Koenkk/zigbee-OTA #631: Updating firmware for CCT5010-0001/550B1012 fails with bad checksum

poizan42, 2024-12-24:
I get the following in my log

```
info 2024-12-24 03:35:15z2m: Updating 'Kasper kontor ovenlys' to latest firmware
info 2024-12-24 03:35:17zhc:ota: [0x84b4dbfffec0c536 | PUCK/DIMMER/1] Getting latest image, meta: {"fileName":"16855062966c6d846535c.zigbee","fileVersion":34341631,"fileSize":357015,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Tuya/16855062966c6d846535c.zigbee","imageType":13,"manufacturerCode":4190,"sha512":"befd51ad8ebb145b7d41b5709ace1a6dfe2af9174c161bb7057db570698cd166f10807b9a418440051c03f53c09f41a1d19219be8d0bb95abee7068edc795773","otaHeaderString":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","originalUrl":"https://images.tuyaeu.com/smart/firmware/upgrade/ay1557480848074dO17I/16855062966c6d846535c.zigbee"}
info 2024-12-24 03:35:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kasper kontor ovenlys', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":254,"ballast_minimum_level":1,"ballast_physical_maximum_level":254,"ballast_physical_minimum_level":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":3,"dimmer_mode":"rl","effect":null,"level_config":{"current_level_startup":"minimum","execute_if_off":false,"on_level":254},"linkquality":109,"power_on_behavior":null,"state":"ON","update":{"installed_version":33888767,"latest_version":34341631,"state":"available"}}'
info 2024-12-24 03:35:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Kasper kontor ovenlys' failed (Image CRC-32 is invalid)","status":"error","transaction":"71kbm-3"}'
error 2024-12-24 03:35:18z2m: Update of 'Kasper kontor ovenlys' failed (Image CRC-32 is invalid)
```

allistermaguire, 2025-02-08:
Hey @poizan42, did you find a solution to this? and what version were you running?

I have this issue with my Wiser devices since updating to version 2.x.x of Zigbee2Mqtt. I had no issues updating the firmware with version 1.42. This looks to be similar issue #676

poizan42, 2025-02-10:
@allistermaguire No unfortunately not. It's version "1.42.0-dev commit: [eaafc7c2](https://github.com/Koenkk/zigbee2mqtt/commit/eaafc7c2)"

allistermaguire, 2025-02-10:
Hey @poizan42, that version has zigbee-herdsman-converters >= 21.0.0, which is when the change was merged. Main 1.42 is using 20.58.0.

Koenkk, 2025-02-10:
@allistermaguire what exactly did you revert, because the logic was also there before but just in another file: https://github.com/Koenkk/zigbee-herdsman-converters/blob/fb1f2f77cb80f9f7eca1ce08ec44d472b6635439/src/lib/ota/common.ts#L323

allistermaguire, 2025-02-10:
Hey @Koenkk, I reverted the following function in `ota.js`

```
function validateSilabsGbl(data: Buffer): void {
    assert(data.indexOf(GBL_HEADER_TAG) === 0, `Not a valid GBL image`);

    const gblEndTagIndex = data.indexOf(GBL_END_TAG);

    assert(gblEndTagIndex > 16, `Not a valid GBL image`); // after HEADER, just because...

    const gblEnd = gblEndTagIndex + 12; // tag + length + crc32 (4*3)
    // TODO: nodejs >= v20.15.0, remove dep buffer-crc32
    //       import {crc32} from 'zlib';
    //       const calculatedCrc32 = crc32(data.subarray(0, gblEnd));
    // ignore possible padding
    const calculatedCrc32 = crc32.unsigned(data.subarray(0, gblEnd));

    assert(calculatedCrc32 === VALID_SILABS_CRC, `Image CRC-32 is invalid`);
}
```
back to this
```
function validateSilabsGbl(data: Buffer): void {
    const dataLength = data.length;

    let position = 0;

    while (position + 8 <= dataLength) {
        const tag = data.readUInt32BE(position);
        const len = data.readUInt32LE(position + 4);

        position += 8 + len;

        if (tag !== GBL_TAG_END) {
            continue;
        }

        const calculatedCrc32 = crc32.unsigned(data.subarray(0, position));

        assert(calculatedCrc32 === VALID_SILABS_CRC, `Image CRC-32 is invalid`);

        return;
    }

    throw new Error(`Image is truncated, not long enough to contain a valid tag`);
}
```

poizan42, 2025-02-10:
I don't really know anything about that GBL file format, but the line

https://github.com/Koenkk/zigbee-herdsman-converters/blob/80e718abd30a73ed43a2284b313d6310817ecf18/src/lib/ota.ts#L317-L317

```TS
    const gblEndTagIndex = data.indexOf(GBL_END_TAG);
```

looks sus to me. It's just looking for the sequence `[0xfc, 0x04, 0x04, 0xfc]` anywhere in the firmware image? Wouldn't that be quite likely to appear inside the code if the firmware accepts images in that format? The old version of the code seems to actually be parsing through tag+len records rather than just a blind string search.

Koenkk, 2025-02-11:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Koenkk, 2025-02-18:
Forgot to push it, done now.

closed 2025-02-20 by Koenkk

allistermaguire, 2025-10-07:
zigbee2mqtt has OTA capabilities, it supports your device, not sure if someone has added updates for it. You can find the links https://github.com/Koenkk/zigbee-OTA/blob/master/index.json, you are after "manufacturerCode": 4190

poizan42, 2025-10-07:
@Bart-1992 The current firmware files were added by @rmcleay. He has explained how he found the urls on his blog at https://fearthecow.net/projects/clipsal-wiser-updating-without-hub-zigbee-home-bridge-home-assistant/#4