Koenkk/zigbee-OTA #636: How to identify new firmware update name file in aqara repository?

brainstormi, 2024-12-30:
Hi...
   Maybe a stupid question, but how can I list file firmware names in aqara repositories?. I mean, I saw there were aqara T2 relay firmware updates managed through zigbee2mqtt extracted from this cdn:
https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.switch.acn047/
Like for example the latest one available here:
https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.switch.acn047/20240111171722_OTA_lumi.switch.acn047__20240110_EC6C1F.ota
There is a new update 3 weeks ago of this same device (LLKZMK12LM), but I'm unable to browse/list that repository in order to generate a pull request. How do you do it?  @Otnow 
Regards,

brainstormi, 2024-12-30:
Creating an account in Aqara developer platform, I'm able to query the list of latest firmware updates for 'lumi.switch.acn047', confirming there is a new version. Unfortunately the APIs doesn't point to the filename to be downloades

  "result": [
    {
      "releaseTime": "1733710625000",
      "necessary": 0,
      "fileSize": 597122,
      "updateLog": "1.Fix known bugs",
      "state": 2,
      "firmwareVersion": "0.0.0_0032"
    },

Otnow, 2024-12-30:
Hi, @brainstormi.

You can get link to firmware via MITM using following instructions: https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

brainstormi, 2024-12-30:
@Otnow thanks for the info, i was getting crazy looking in the repository and didn't see it's documented . Now I need to buy a Xiaomi gateway for the future. In the meantime would you be able to punish this update? Thank you anyway for your support

closed 2024-12-30 by brainstormi

Otnow, 2024-12-31:
@brainstormi, of course, here you go:

```json
"createTime": 1725440408000,
"downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.switch.acn047/20240830115847_OTA_lumi.switch.acn047_0.0.0_0032_20240823_D8294E.ota",
"fileSize": 597122,
"firmwareCRC32": "4F605C7C",
"firmwareMD5": "9815d1a7d9b1adfddfa19a46b0d8294e",
"firmwareVersion": "0.0.0_0032",
"imageType": "exe",
"isForce": false,
"isNecessary": 0,
"part": "zigbee",
"releaseDate": 1733710678000,
"state": 3,
"updateLog": "1.Fix known bugs",
"updateTime": 1733710678000,
"upgradeTakesTime": 0
```

You are welcome )