Koenkk/zigbee-OTA #639: Z2M Not reporting new Danalock firmware

kebalto, 2025-01-03:


Nerivec, 2025-01-06:
Can you post the logs prior to the ones you posted? Specifically the line that gives the current metadata of your device.

karelfridrich, 2025-01-14:
Happens to me too:
`[2025-01-14 23:33:24] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"LockFrontDoor","transaction":"bmy4x-1"}'
[2025-01-14 23:33:24] debug:    zhc:ota:common: Checking if an update is available for '0x70ac08fffe846836' (V3-BTZBE)
[2025-01-14 23:33:24] debug:    zhc:ota:common: Using endpoint '1'
[2025-01-14 23:33:28] debug:    zhc:ota:common: Got request '{"fieldControl":0,"manufacturerCode":4444,"imageType":4,"fileVersion":285474816}'
[2025-01-14 23:33:28] debug:    zhc:ota:common: Is new image available for '0x70ac08fffe846836' (V3-BTZBE), current '{"fieldControl":0,"manufacturerCode":4444,"imageType":4,"fileVersion":285474816}'
[2025-01-14 23:33:28] debug:    zhc:ota: Getting image metadata for 'V3-BTZBE'
[2025-01-14 23:33:28] debug:    zhc:ota: Downloaded main index
[2025-01-14 23:33:28] debug:    zhc:ota:common: Is new image available for '0x70ac08fffe846836' (V3-BTZBE), latest meta '{"fileVersion":285474816,"fileSize":204794,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danalock/115C-0004-11040000-ZigbeeXM_101-029_E1_DanalockV3_17.4.0_20221213143911.ota","sha512":"44e421aecf5b0c5793a1ba836257f8601408e1dcc2832846b0093e7c2f2a4c100632b8a71ef346b9c611be1d15602bcc3b693002994830b8e857da6e931f6722"}'
[2025-01-14 23:33:28] debug:    zhc:ota:common: Update available for '0x70ac08fffe846836' (V3-BTZBE): NO
[2025-01-14 23:33:28] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"LockFrontDoor","updateAvailable":false},"status":"ok","transaction":"bmy4x-1"}'
`

If I exec bash in the container and get the metadata, both versions are there:
`45df7312-zigbee2mqtt:/app# curl -s https://raw.githubusercontent.com/Koenkk/zigbee-OTA/refs/heads/master/index.json | grep Dana -A4 -B2
  },
  {
    "fileName": "115C-0004-11040000-ZigbeeXM_101-029_E1_DanalockV3_17.4.0_20221213143911.ota",
    "fileVersion": 285474816,
    "fileSize": 204794,
    "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danalock/115C-0004-11040000-ZigbeeXM_101-029_E1_DanalockV3_17.4.0_20221213143911.ota",
    "imageType": 4,
    "manufacturerCode": 4444,
    "sha512": "44e421aecf5b0c5793a1ba836257f8601408e1dcc2832846b0093e7c2f2a4c100632b8a71ef346b9c611be1d15602bcc3b693002994830b8e857da6e931f6722",
    "otaHeaderString": "Danalock V3 BTZBE XM\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
    "modelId": "V3-BTZBE"
  },
  {
    "fileName": "PSoC4_1246-0100-01280128.0002_(4CA01CD1).ota",
--
  },
  {
    "fileName": "115C-0004-11070000-ZigbeeXM_101-029_E1_DanalockV3_17.7.0_20240201155008.ota",
    "fileVersion": 285671424,
    "fileSize": 204522,
    "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danalock/115C-0004-11070000-ZigbeeXM_101-029_E1_DanalockV3_17.7.0_20240201155008.ota",
    "imageType": 4,
    "manufacturerCode": 4444,
    "sha512": "9e45e56b0aef871e1280c678729fdce0a57628d40e569fbc3c26bca5acc2a5c4088a90cbd59c9b6ea378d5c450310e04766b133333b23e7b4520e5c80282e334",
    "otaHeaderString": "Danalock V3 BTZBE XM\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
  },
  {
    "fileName": "20240719174346_OTA_lumi.plug.macn01_0.0.0_0036_20240719_40C5BB.ota",
    "fileVersion": 36,
45df7312-zigbee2mqtt:/app# 
`

karelfridrich, 2025-01-14:
There's no model ID in the new version imho.

karelfridrich, 2025-01-15:
OK, figured it out. 'otaHeaderString' is different. It looks the same, but it is not. When I copied the header from the old version to the new version it found the update. (I used override URL)

Nerivec, 2025-01-15:
Override always takes precedence. _The ota header string is not actually used at the moment._
The issue is with the missing modelId.

@Koenkk according to linked source PR, this can be safely changed, not sure if you'd rather revert the PR and re-introduce it with proper metadata (and trigger auto-archiving) or move things manually?

karelfridrich, 2025-01-15:
Weird, I tried adding the ID first, but it did not work. It started working after I copied the header. Maybe there's some caching going on? Or I somehow screwed up :smile:

Koenkk, 2025-01-15:
Fixed it!

closed 2025-01-15 by Koenkk

kebalto, 2025-01-26:
Still not updating. Cannot see where z2m sends model-id.

[2025-01-26 15:34:27] info: 	z2m: Checking if update available for '0x588e81fffe43e16b'
[2025-01-26 15:34:27] debug: 	zhc:ota: [0x588e81fffe43e16b | V3-BTZBE] Checking if an update is available
[2025-01-26 15:34:27] debug: 	zhc:ota: [0x588e81fffe43e16b | V3-BTZBE] Using endpoint '1'
[2025-01-26 15:34:27] debug: 	zh:controller:endpoint: CommandResponse 0x588e81fffe43e16b/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"})
[2025-01-26 15:34:27] debug: 	zh:controller:endpoint: Request Queue (0x588e81fffe43e16b/1): send imageNotify request immediately (sendPolicy=immediate)
[2025-01-26 15:34:27] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x588e81fffe43e16b:350/1 (0,0,1)
[2025-01-26 15:34:27] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":350,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":62,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[25,31,0,0,100]}}
[2025-01-26 15:34:27] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,94,1,1,1,25,0,62,0,30,5,25,31,0,0,100,43]
[2025-01-26 15:34:27] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-01-26 15:34:27] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-01-26 15:34:27] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-01-26 15:34:27] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-01-26 15:34:27] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-26 15:34:30] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,62,248]
[2025-01-26 15:34:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,62,248]
[2025-01-26 15:34:30] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,62] - 248
[2025-01-26 15:34:30] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":62}
[2025-01-26 15:34:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-26 15:34:32] debug: 	zh:zstack:unpi:parser: <-- [254,32,68,129,0,0,25,0,94,1,1,13,0,127,0,107,242,142,0,0,12,1,27,1,0,92,17,4,0,0,0,4,17,94,1,29,206]
[2025-01-26 15:34:32] debug: 	zh:zstack:unpi:parser: --- parseNext [254,32,68,129,0,0,25,0,94,1,1,13,0,127,0,107,242,142,0,0,12,1,27,1,0,92,17,4,0,0,0,4,17,94,1,29,206]
[2025-01-26 15:34:32] debug: 	zh:zstack:unpi:parser: --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,94,1,1,13,0,127,0,107,242,142,0,0,12,1,27,1,0,92,17,4,0,0,0,4,17,94,1,29] - 206
[2025-01-26 15:34:32] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":350,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":127,"securityuse":0,"timestamp":9368171,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,27,1,0,92,17,4,0,0,0,4,17]}}
[2025-01-26 15:34:32] debug: 	zh:controller: Received payload: clusterID=25, address=350, groupID=0, endpoint=1, destinationEndpoint=13, wasBroadcast=false, linkQuality=127, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":27,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4444,"imageType":4,"fileVersion":285474816},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2025-01-26 15:34:32] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-26 15:34:32] debug: 	z2m: Received Zigbee message from '0x588e81fffe43e16b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":285474816,"imageType":4,"manufacturerCode":4444}' from endpoint 1 with groupID 0
[2025-01-26 15:34:32] debug: 	zhc:ota: [0x588e81fffe43e16b | V3-BTZBE] Got request '{"fieldControl":0,"manufacturerCode":4444,"imageType":4,"fileVersion":285474816}'
[2025-01-26 15:34:32] debug: 	zhc:ota: [0x588e81fffe43e16b | V3-BTZBE] Checking latest image availability, current: {"fieldControl":0,"manufacturerCode":4444,"imageType":4,"fileVersion":285474816}
[2025-01-26 15:34:32] debug: 	zhc:ota: [0x588e81fffe43e16b | V3-BTZBE] Getting image metadata...
[2025-01-26 15:34:32] debug: 	zhc:ota: Downloaded main index
[2025-01-26 15:34:32] debug: 	zhc:ota: [0x588e81fffe43e16b | V3-BTZBE] Result for latest image availability, meta: '{"fileName":"115C-0004-11040000-ZigbeeXM_101-029_E1_DanalockV3_17.4.0_20221213143911.ota","fileVersion":285474816,"fileSize":204794,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Danalock/115C-0004-11040000-ZigbeeXM_101-029_E1_DanalockV3_17.4.0_20221213143911.ota","imageType":4,"manufacturerCode":4444,"sha512":"44e421aecf5b0c5793a1ba836257f8601408e1dcc2832846b0093e7c2f2a4c100632b8a71ef346b9c611be1d15602bcc3b693002994830b8e857da6e931f6722","otaHeaderString":"Danalock V3 BTZBE XM\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","modelId":"V3-BTZBE"}'
[2025-01-26 15:34:32] debug: 	zhc:ota: [0x588e81fffe43e16b | V3-BTZBE] Upgrade available: NO
[2025-01-26 15:34:32] info: 	z2m: No update available for '0x588e81fffe43e16b'
[... cut ...] t"}}
[2025-01-26 15:34:44] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-26 15:34:44] debug: 	z2m: Received Zigbee message from '0x588e81fffe43e16b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":285474816,"imageType":4,"manufacturerCode":4444}' from endpoint 1 with groupID 0
[2025-01-26 15:34:44] debug: 	z2m: Device '0x588e81fffe43e16b' requested OTA
[2025-01-26 15:35:09] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
`

karelfridrich, 2025-07-17:
Still doesn't work for me either. Had to use local json with the older version deleted, then it finally updated.